lvgl/lvgl #5468: Very important function lv_label_set_recolor is missing in v9

xdzmkus, 2024-01-24:
### LVGL version

v9

### What happened?

Very important function lv_label_set_recolor is missing in v9

### How to reproduce?

#include "lv_label.h"
v_label_set_recolor(NULL, true);

kisvegabor, 2024-01-25:
It was removed it on favor of the `lv_span` widget which is way more feature rich.

See [this issue](https://github.com/lvgl/lvgl/issues/5352) about improving the span widget and simplifying its API.

closed 2024-02-16 by lvgl-bot

MProx, 2025-01-07:
It seems that this function is still used in the documentation example here:
https://docs.lvgl.io/master/details/widgets/label.html
See the example code in the "Line wrap, recoloring and scrolling" section.

vwheeler63, 2025-01-08:
I am confused, because that documentation was based on a working code example in v9.2....  And the function `lv_label_set_recolor()` is still present in the code and the exported API.  I just did an update from https://github.com/lvgl/lvgl `master` branch and the function is still there.

kisvegabor, 2025-01-09:
`lv_label_set_recolor()` is added again in `master`. See [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/label/lv_label.h#L138).

vwheeler63, 2025-01-09:
> `lv_label_set_recolor()` is added again in `master`. See [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/label/lv_label.h#L138).

I am SO GLAD!  That is a feature of Label Widgets that I use in our (HMQ's) Dashboards.