lvgl/lvgl #5494: Padding is not working in linux framebuffer v9

rmshub, 2024-01-26:
### Introduce the problem

Upgraded v8 code base to v9. Verified changes in linux SDL simulator and working as expected. However, same code is not working properly in hardware using linux framebuffer. Here looks like there is issue with padding. Could you please help us to resolve this issue.

**Linux SDL simulator**:
[image]
**Linux FrameBuffer**:
[image]

rmshub, 2024-01-26:
SDL lv_conf.h vs framebuffer lv_conf.h
[image]

kisvegabor, 2024-01-26:
Hi,

Could you print `lv_display_get_dpi(NULL)` in both cases?

rmshub, 2024-01-26:
Hi @kisvegabor 

**Linux Frame Buffer**: DPI -26008
```
[Info]  (95.621, +27)    lv_linux_fbdev_set_file: 1024x600, 32bpp lv_linux_fbdev.c:179
[Info]  (95.621, +0)     lv_linux_fbdev_set_file: The framebuffer device was mapped to memory successfully lv_linux_fbdev.c:194
[Info]  (95.622, +1)     lv_linux_fbdev_set_file: Resolution is set to 1024x600 at -26008dpi lv_linux_fbdev.c:238
```
**Linux SDL**: DPI 130

rmshub, 2024-01-26:
https://github.com/lvgl/lvgl/blob/master/src/drivers/display/fb/lv_linux_fbdev.c#L234
Calculating DPI for negative values also. I think, DPI should be calculated for positive width values.

With below changes linux framebuffer working as expected
```
 if(width > 0)
 {
      lv_display_set_dpi(disp, DIV_ROUND_UP(hor_res * 254, width * 10));
 }
```

kisvegabor, 2024-01-29:
Thank you, closing this issue.

closed 2024-01-29 by kisvegabor