lvgl/lvgl #5495: the button style in the list while scrolling is not user-friendly.

ustbshen, 2024-01-26:
### Introduce the problem

I create a list by `lv_list_create` and then add some buttons into it to make the list scrollable. when I click inside the list(also on a list button) and try to scroll the list, the button still shows the style of pressed while list is scrolling.
after I release the click, no button click event is not triggered.
I try in Android and some other touch UI, while scrolling, no button will be displayed as pressed.

Here is the example on webpage(https://docs.lvgl.io/master/examples.html#list), I scroll the list and keep mouse down, but the button is still in pressed style.
[image]

kisvegabor, 2024-01-29:
I was also thinking about it already. We can easily do it by adding `lv_obj_remove_state(indev->pointer.act_obj, LV_STATE_PRESSED);` [here](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev_scroll.c#L58).

cc @XuNeo

kisvegabor, 2024-02-15:
I've opened #5660

PR #5660 (2024-02-16): feat(indev): remove the PRESSED state on scroll
### Description of the feature or fix

fixes #5495

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `s [... cut]