lvgl/lvgl #5500: ESP32-S3 4.3inch Capacitive Touch Display Development Board, 800×480, 5-point Touch, 32-bit LX7 Dual-core Processor

Gert69, 2024-01-26:
### Introduce the problem

Good Day

I have been trying to use Square Line Studio to create GUI for the above screen only to realize I am not on that level yet and it wont work with my development board if you export the files and upload them or simply I don't know how it works.

The Reason its not working is  ESP32-S3 development board specifically designed with an onboard LCD. This type of development board typically routes the necessary pins for the LCD interface directly to the display, which means you wouldn't need to worry about manually connecting GPIO pins to an external LCD.

I have been trying to create sketches but encounter a lot of issues on of which "No such file or Directory"
simply because I am unsure where to paste the folders and route the sketch Directory.

I will add a working LVGL porting code that successfully loads LVGL v8 and will load one that's not working please take into consideration that I am a humbled Newbie to this and I learn as I go.

anyone willing to help I would truly appreciate it [https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3#demo1](url)

Gert69, 2024-01-26:
`/**
 * The example demonstrates how to port LVGL.
 *
 * ## How to Use
 *
 * To use this example, please firstly install `ESP32_Display_Panel` (including its dependent libraries) and
 * `lvgl` (v8.3.x) libraries, then follow the steps to configure them:
 *
 * 1. [Configure ESP32_Display_Panel](https://github.com/esp-arduino-libs/ESP32_Display_Panel#configure-esp32_display_panel)
 * 2. [Configure LVGL](https://github.com/esp-arduino-libs/ESP32_Display_Panel#configure-lvgl)
 * 3. [Configure Board](https://github.com/esp-arduino-libs/ESP32_Display_Panel#configure-board)
 *
 * ## Example Output
 *
 * ```bash
 * ...
 * Hello LVGL! V8.3.8
 * I am ESP32_Display_Panel
 * Starting LVGL task
 * Setup done
 * Loop
 * Loop
 * Loop
 * Loop
 * ...
 * ```
 */

#include <Arduino.h>
#include <lvgl.h>
#include <ESP_Panel_Library.h>
#include <ESP_IOExpander_Library.h>
#include "lv_conf.h"
#include <demos/lv_demos.h>

// Extend IO Pin define
#define TP_RST 1
#define LCD_BL 2
#define LCD_RST 3
#define SD_CS 4
#define USB_SEL 5

// I2C Pin define
#define I2C_MASTER_NUM 0
#define I2C_MASTER_SDA_IO 8
#define I2C_MASTER_SCL_IO 9

/**
/* To use the built-in examples and demos of LVGL uncomment the includes below respectively.
 * You also need to copy `lvgl/examples` to `lvgl/src/examples`. Similarly for the demos `lvgl/demos` to `lvgl/src/demos`.
 */
#include <demos/lv_demos.h>
#include <examples/lv_examples.h>

/* LVGL porting configurations */
#define LVGL_TICK_PERIOD_MS     (2)
#define LVGL_TASK_MAX_DELAY_MS  (500)
#define LVGL_TASK_MIN_DELAY_MS  (1)
#define LVGL_TASK_STACK_SIZE    (4 * 1024)
#define LVGL_TASK_PRIORITY      (2)
#define LVGL_BUF_SIZE           (ESP_PANEL_LCD_H_RES * 20)

ESP_Panel *panel = NULL;
SemaphoreHandle_t lvgl_mux = NULL;                  // LVGL mutex

#if ESP_PANEL_LCD_BUS_TYPE == ESP_PANEL_BUS_TYPE_RGB
/* Display flushing */
void lvgl_port_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
    panel->getLcd()->drawBitmap(area->x1, area->y1, area->x2 + 1, area->y2 + 1, color_p);
[... 152 lines cut ...]
     * source codes: https://github.com/lvgl/lvgl/tree/e7f88efa5853128bf871dde335c0ca8da9eb7731/examples
     */
    //  lv_example_btn_1();

    /**
     * Or try out a demo.
     * Don't forget to uncomment header and enable the demos in `lv_conf.h`. E.g. `LV_USE_DEMOS_WIDGETS`
     */
    lv_demo_widgets();
    // lv_demo_benchmark();
    // lv_demo_music();
    // lv_demo_stress();

    /* Release the mutex */
    lvgl_port_unlock();

    Serial.println("Setup done");
}

void loop()
{
    // Serial.println("Loop");
    sleep(1);
}
`

Gert69, 2024-01-26:
[image]
[image]
[image]

Gert69, 2024-01-26:
[image]

Gert69, 2024-01-26:
[image]

yanu727, 2024-01-27:
https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3#Library_Installation

[image]

Gert69, 2024-02-02:
Thanks Yanu

I Got this working if I use Square Line studio to export it exports as TFT eSPI and the board I am using runs on ST7262 through the ESP_Display_Panel.

I have no idea how to get the files that exports from Square Line studio as TFT eSPI files to run on the ST7262 driver in ESP_Display_Panel as TFTeSPI lib doesn'y have a ST7262 library.

kisvegabor, 2024-02-02:
I suggest asking around at https://forum.squareline.io/

Westcott1, 2024-02-19:
I'm using one of these Waveshare boards successfully with LVGL.
However, I use Arduino IDE and Arduino_GFX library.

Gert69, 2024-02-19:
@Westcott1  Thanks for the response appreciated however i have no clue how to integrate the GFX lib if I did I would've definitely gone that route if its successful I have been battling with this from day one and I need to get this project going time is not on my side at least I know there's other hope as well now .

Gert69, 2024-02-19:
@Westcott1  If I may what display driver are you using in Arduino_GFX Library?

Westcott1, 2024-02-19:
I've created an example for this board using the [Arduino_GFX library](https://github.com/Westcott1/Waveshare-ESP32-S3-Touch-LCD-4.3-and-Arduino/tree/main/Examples/Arduino_GFX)

To then use LVGL you need to integrate the above example with moononournation's [LvglHelloWorld.ino](https://github.com/moononournation/Arduino_GFX/tree/master/examples/LVGL/LvglHelloWorld) example.

Soon-ish I'll post the combined example.

Gert69, 2024-02-20:
@Westcott1 Thanks a million for your info and assistance much appreciated I will give it I go once you have posted the combined example in the meantime I'll do some more research on to get more familiar with the GFX library.

Gert69, 2024-02-20:
Managed to get a hello World example on this code @Westcott1 Still has a little screen interference but I'm sure with a little bit of tweaking one can get it going with no issues.

`#include <ESP_IOExpander_Library.h>
#include <Arduino_GFX_Library.h>
#include <bb_captouch.h>

ESP_IOExpander *expander;
Arduino_ESP32RGBPanel* rgbpanel;
Arduino_RGB_Display* gfx;
BBCapTouch bbct;

#define TP_RST 1
#define LCD_BL 2
#define LCD_RST 3
#define SD_CS 4
#define USB_SEL 5

#define I2C_MASTER_NUM 0
#define I2C_MASTER_SDA_IO 8
#define I2C_MASTER_SCL_IO 9

#define TOUCH_SDA 8
#define TOUCH_SCL 9
#define TOUCH_INT 4
#define TOUCH_RST 0

const char* szNames[] = { "Unknown", "FT6x36", "GT911", "CST820" };

void ExpanderInit() {
  Serial.println("IO expander init...");
  expander = new ESP_IOExpander_CH422G(I2C_MASTER_NUM, ESP_IO_EXPANDER_I2C_CH422G_ADDRESS_000);
  expander->init();
  expander->begin();
  expander->multiPinMode(TP_RST | LCD_BL | LCD_RST | SD_CS | USB_SEL, OUTPUT);
  BLset(LOW);
}

void BLset(byte state){
    expander->digitalWrite(LCD_BL, state);
}

void BLblink(){
  //Turn off backlight
  BLset(LOW);
  delay(100);
  BLset(HIGH);
  delay(100);
}

bool GFXinit() {
  Serial.println("GFX init...");

  if (!gfx->begin()) {
    Serial.println("gfx->begin() failed!");
    return false;
  }

  gfx->fillScreen(0x003030);
  BLset(HIGH);
  gfx->setTextColor(WHITE);
  gfx->setTextSize(4);
  gfx->setCursor(250, 200);
  gfx->println("Hello world");

  return true;
}

void TouchInit() {
  Serial.println("Touch init...");
  // Init touch device
  bbct.init(TOUCH_SDA, TOUCH_SCL, TOUCH_RST, TOUCH_INT);
  int iType = bbct.sensorType();
  Serial.printf("Touch sensor type = %s\n", szNames[iType]);
}

void TouchRead() {
  TOUCHINFO ti;

  if (bbct.getSamples(&ti)) {  // if touch event happened
    BLblink();
    Serial.printf("Touch x: %d y: %d size: %d\n", ti.x[0], ti.y[0], ti.area[0]);
  }
}

void setup() {
  Serial.begin(115200);
  while (!Serial)
    ;
  Serial.println("\n-- Init --");

  TouchInit();
  delay(200);
  ExpanderInit();

  rgbpanel = new Arduino_ESP32RGBPanel(
    5, /* DE */
    3, /* VSYNC */
    46, /* HSYNC */
    7, /* PCLK */
    1, 2, 42, 41, 40, /* R0-R4 */
    39, 0, 45, 48, 47, 21, /* G0-G5 */
    14, 38, 18, 17, 10, /* B0-B4 */
    0, 40, 48, 88, /* hsync_polarity, hsync_front_porch, hsync_pulse_width, hsync_back_porch */
    0, 13, 3, 32, /* vsync_polarity, vsync_front_porch, vsync_pulse_width, vsync_back_porch */
    1, 16000000 /* pclk_active_neg, prefer_speed */
  );

  gfx = new Arduino_RGB_Display(
    800, /* width */
    480, /* height */
    rgbpanel,
    0, /* rotation */
    true /* auto_flush */
  );

  if (GFXinit()) {
    Serial.println("GFX display initialised");
  }
}

void loop() {
  TouchRead();
  delay(5);
}
`

Westcott1, 2024-02-21:
Hi all,
I've now created an [LVGL example](https://github.com/Westcott1/Waveshare-ESP32-S3-Touch-LCD-4.3-and-Arduino/tree/main/Examples/Arduino_LVGL)
When I run it the ESP32 crashes every few minutes.

**This example only works on LVGL 8**

kisvegabor, 2024-02-24:
Could you post a log about the crash?

Westcott1, 2024-02-24:
It this what you mean?
It took 57 secs to crash this time. Other times quicker, sometimes several minutes.
The only thing happening in the code (that I can see) is gfx->flush() and lv_timer_handler(); being called in the loop()
Arduino IDE 2.3.2
ESP32 S3
LVGL v8.3.11

14:05:49.193 -> -- Init --
14:05:49.193 -> Touch init...
14:05:49.284 -> Touch sensor type = GT911
14:05:49.503 -> IO expander init...
14:05:49.503 -> GFX init...
14:05:49.549 -> GFX display OK
14:05:50.514 -> LVGL init...
14:05:50.645 -> LVGL OK
14:06:47.765 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
14:06:47.802 -> 
14:06:47.802 -> Core  1 register dump:
14:06:47.802 -> PC      : 0x42005ebf  PS      : 0x00060830  A0      : 0x8202050f  A1      : 0x3fcebea0  
14:06:47.832 -> A2      : 0x8202b468  A3      : 0x00000000  A4      : 0x3fc96d58  A5      : 0x3fc96d68  
14:06:47.832 -> A6      : 0x00000130  A7      : 0x3fcea0c0  A8      : 0x82005eb2  A9      : 0x3fcebe80  
14:06:47.832 -> A10     : 0x3fc96d54  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
14:06:47.832 -> A14     : 0x3fcec274  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
14:06:47.832 -> EXCVADDR: 0x8202b47c  LBEG    : 0x4201fe09  LEND    : 0x4201fe2b  LCOUNT  : 0x00000000  
14:06:47.874 -> 
14:06:47.874 -> Backtrace: 0x42005ebc:0x3fcebea0 0x4202050c:0x3fcebef0 0x4200252b:0x3fcebf20 0x4202b465:0x3fcebf40
14:06:47.874 -> 
14:06:47.874 -> ELF file SHA256: 7a8b214e5decea9f
14:06:47.874 -> 
14:06:47.958 -> Rebooting...
14:06:47.958 -> ESP-ROM:esp32s3-20210327
14:06:47.958 -> Build:Mar 27 2021
14:06:47.958 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
14:06:47.999 -> Saved PC:0x4205072e
14:06:47.999 -> SPIWP:0xee
14:06:47.999 -> mode:DIO, clock div:1
14:06:47.999 -> load:0x3fce3808,len:0x44c
14:06:47.999 -> load:0x403c9700,len:0xbe4
14:06:47.999 -> load:0x403cc700,len:0x2a68
14:06:47.999 -> entry 0x403c98d4

istvank, 2024-02-28:
Hi @ReacherS3 I managed to set up a small demo project using SquareLine Studio, you can find it here:
https://github.com/istvank/Waveshare-ESP32-S3-Touch-LCD-4.3

There are some problems, though, like flickering yellow on updates. In the meantime I even managed to rotate the screen, but only with a software rotation. I still need to check whether the driver has a rotation setting.

ReacherS3, 2024-02-28:
@istvank I will load it maybe I can assist with the flickering issue and get it going thanks a million!!

kisvegabor, 2024-03-03:
@Westcott1 Unfortunately, the log doesn't tell much :slightly_frowning_face: 
Please try using an empty `flush_cb`. Just call `lv_display_flush_ready(display)` and `return`.

Westcott1, 2024-03-04:
Regarding the screen interference, I think it might be when the underlying Arduino_GFX tries to draw something.

closed 2024-03-26 by lvgl-bot

hemangjoshi37a, 2024-05-19:
can anyone suggest me any working code for radial dial  and digital numeric display  for this rouch lvgl display ?

ReacherS3, 2024-05-19:
@hemangjoshi37a can you post a link to your hardware you are using currently I guess your using Arduino IDE?

01GOD, 2024-06-14:
Yikes. Bought a board with that chip this week, expecting that it's supported and easy to use for Squareline to LVGL. 
Hope it is!!!

ReacherS3, 2024-06-14:
@01GOD  Trust you are well definitely not a easy combination and setup but doable the libraries have updated alot since this conversation was opened.

If you have the basic Libraries installed I can post the links and guide you to the correct libraries depending on what you choose to do the best way I found was using the following libraries.

` #include <Arduino.h>
#include <lvgl.h>
#include <ESP_Panel_Library.h>
#include <ESP_IOExpander_Library.h>
#include "lv_conf.h"`

But keep in mind there is only certain version working.

And obviously this is using Arduino IDE.

panosGkrigkas, 2024-06-20:
I have followed the waveshare wiki instructions but after a week, I still have a black screen and esp crashes at best, when I don't have multiple fails with the libraries. If you could point me to a working guide/example to follow to get going, I would appreciate it!

ReacherS3, 2024-06-20:
@panosGkrigkas 

Try this first let me know if it works?If it does I will guide you through the setup and if it doesn't let me know what errors you are receiving.

`#include <Arduino.h>
#include <lvgl.h>
#include <ESP_Panel_Library.h>
#include <ESP_IOExpander_Library.h>
#include "lv_conf.h"

// Extend IO Pin define
#define TP_RST 1
#define LCD_BL 2
#define LCD_RST 3
#define SD_CS 4
#define USB_SEL 5

// I2C Pin define
#define I2C_MASTER_NUM 0
#define I2C_MASTER_SDA_IO 8
#define I2C_MASTER_SCL_IO 9

/**
 * To use the built-in examples and demos of LVGL uncomment the includes below respectively.
 * You also need to copy `lvgl/examples` to `lvgl/src/examples`. Similarly for the demos `lvgl/demos` to `lvgl/src/demos`.
 */
#include <demos/lv_demos.h>
#include <examples/lv_examples.h>

/* LVGL porting configurations */
#define LVGL_TICK_PERIOD_MS     (2)
#define LVGL_TASK_MAX_DELAY_MS  (500)
#define LVGL_TASK_MIN_DELAY_MS  (1)
#define LVGL_TASK_STACK_SIZE    (4 * 1024)
#define LVGL_TASK_PRIORITY      (2)
#define LVGL_BUF_SIZE           (ESP_PANEL_LCD_H_RES * 5)

ESP_Panel *panel = NULL;
SemaphoreHandle_t lvgl_mux = NULL;                  // LVGL mutex

#if ESP_PANEL_LCD_BUS_TYPE == ESP_PANEL_BUS_TYPE_RGB
/* Display flushing */
void lvgl_port_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
    panel->getLcd()->drawBitmap(area->x1, area->y1, area->x2 + 1, area->y2 + 1, color_p);
    lv_disp_flush_ready(disp);
}
#else
/* Display flushing */
void lvgl_port_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
    panel->getLcd()->drawBitmap(area->x1, area->y1, area->x2 + 1, area->y2 + 1, color_p);
}

bool notify_lvgl_flush_ready(void *user_ctx)
{
    lv_disp_drv_t *disp_driver = (lv_disp_drv_t *)user_ctx;
    lv_disp_flush_ready(disp_driver);
    return false;
}
#endif /* ESP_PANEL_LCD_BUS_TYPE */

#if ESP_PANEL_USE_LCD_TOUCH
/* Read the touchpad */
void lvgl_port_tp_read(lv_indev_drv_t * indev, lv_indev_data_t * data)
{
    panel->getLcdTouch()->readData();

    bool touched = panel->getLcdTouch()->getTouchState();
    if(!touched) {
[... 205 lines cut ...]
    lv_label_set_text(lv_obj_get_child(lv_scr_act(), 1), temp_str);

    snprintf(temp_str, sizeof(temp_str), "Battery: %.0f °C", temperature + 2.5);
    lv_label_set_text(lv_obj_get_child(lv_scr_act(), 2), temp_str);

    snprintf(temp_str, sizeof(temp_str), "Battery: %.0f °C", temperature - 4.0);
    lv_label_set_text(lv_obj_get_child(lv_scr_act(), 3), temp_str);

    snprintf(temp_str, sizeof(temp_str), "Battery: %.0f °C", temperature + 0.5);
    lv_label_set_text(lv_obj_get_child(lv_scr_act(), 4), temp_str);

    snprintf(temp_str, sizeof(temp_str), "Battery: %.0f °C", temperature + 2.5);
    lv_label_set_text(lv_obj_get_child(lv_scr_act(), 6), temp_str);

    snprintf(temp_str, sizeof(temp_str), "Battery: %.0f °C", temperature - 4.0);
    lv_label_set_text(lv_obj_get_child(lv_scr_act(), 7), temp_str);

    snprintf(temp_str, sizeof(temp_str), "Battery: %.0f °C", temperature + 0.5);
    lv_label_set_text(lv_obj_get_child(lv_scr_act(), 8), temp_str);

    lvgl_port_unlock();

    lv_task_handler(); // Add lv_task_handler to handle LVGL tasks
    vTaskDelay(pdMS_TO_TICKS(85)); // Delay for 1 second
}`

fellpower, 2024-08-02:
Im completely lost with this display. a clear question. is it possible, to get the UI, that is made with Squareline Studio, running on that display?

ReacherS3, 2024-08-02:
Yes you can do that.

01GOD, 2024-08-02:
> Yes you can do that.
> […](#)
> -- *Regards* Gert Coetser Cell: +27765782248 Email ***@***.*** Web: [www.seatitan.group](http://www.seatitan.group)

Any guidance or links or other info on how to use Squareline and LVGL with a JC4827W543 cap touch ESP32-S3 display board?

Thanks in advance!

fellpower, 2024-08-02:
Can i write a quick workaround, how to do that? im using squareline studio personal and arduino ide ^^

01GOD, 2024-08-02:
> Can i write a quick workaround, how to do that? im using squareline studio personal and arduino ide ^^

Yes. A tutorial on how to would be appreciated.

yaredna, 2024-10-09:
There are few issues with this board. I am using it with Squareline. it took some adaptation. Definitely use lvgl 8.3.11 (not 8.4 or 9.x). I found an issue where indev_proc_reset_query_handler() would often crash, just like your crash above. I used xtensa to backtrace the routine calls. Now I added in lvgl/src/core/indev.c in the indev_proc_reset_query_handler routine 
  if (indev != NULL) {
}
and this seems to be a bit more stable. I still have a crash every few hours...

Westcott1, 2024-10-11:
I have this board working correctly in the Arduino IDE using LovyanGFX, LVGL 9, and Expressif ESP32 v3.
Using the standard pins there is no backlight control.

yaredna, 2024-10-12:
> I have this board working correctly in the Arduino IDE using LovyanGFX, LVGL 9, and Expressif ESP32 v3. Using the standard pins there is no backlight control.

Is there a possibility to share more with us ? lv_conf.h ? Do you need a custom User-setup.h ? I would love to move to LVGL 9!

Westcott1, 2024-10-12:
Hi Yaredna,
I use the standard lv_conf.h that comes with LVGL 9.
User-setup.h is only needed for TFT_eSPI.
LovyanGFX doesn't need a config file,
I can't find a complete example, but the LovyanGFX example is [here](https://github.com/Westcott1/Waveshare-ESP32-S3-Touch-LCD-4.3-and-Arduino/tree/main/Examples/LovyanGFX).