lvgl/lvgl #5515: ESP-IDF LVGL 9 Direct rendering

silverchris, 2024-01-29:
### Introduce the problem

Hi All,
Hoping someone can help me out with this. 
I have direct rendering working in 8.3

Trying to set it up in 9 like

```C
    disp = lv_display_create(480, 480);
    lv_display_set_flush_cb(disp, flush_cb);
    lv_display_set_buffers(disp, fb0, fb1, 480*480*2, LV_DISPLAY_RENDER_MODE_FULL);
```

Which seems correct seems to get me to just straight crashing. Am I missing something obvious?

kisvegabor, 2024-01-29:
Hi,

Do you know where it crashes? 
Does it work with an empty `flush_cb` by calling only `lv_display_flush_ready()` in it?

I'm not sure it maters but now you have configured the display to "full refresh mode". For direct mode you should use `LV_DISPLAY_RENDER_MODE_DIRECT`.

Please also show how have you allocated fb0 and fb1.

silverchris, 2024-01-29:
I believe I tried a completely empty `flush_cb`, and still had the crash. Will try it with just `lv_display_flush_ready()`
I tried both `LV_DISPLAY_RENDER_MODE_DIRECT` and `LV_DISPLAY_RENDER_MODE_FULL`.

fb0 and fb1 are retrieved from the esp-idf rgb lcd code with `esp_lcd_rgb_panel_get_frame_buffer`

kisvegabor, 2024-01-29:
Are you sure that the buffers are screen-sized?

silverchris, 2024-01-29:
> Are you sure that the buffers are screen-sized? 

I believe they must be when dealing with an RGB interfaced screen. It also works with LVGL 8.3

silverchris, 2024-01-30:
@kisvegabor 
Here is a repository with an example of what I am trying to do https://github.com/silverchris/lvgl_test/blob/main/main/lvgl_test.c

In this test, I can get it to actually show something on the screen, but it's a rolling image, and the log output is spamming
```
[Info]  (198.204, +8)    lv_malloc: couldn't allocate memory (259203 bytes) lv_mem.c:73                                                                                                                                                                                                                                 
[Info]  (198.212, +8)    lv_malloc: used:   8448 ( 28 %), frag:   1 %, biggest free:  22596 lv_mem.c:77                                                                                                                                                                                                                 
[Warn]  (198.220, +8)    lv_draw_buf_create: No memory: 360x180, cf: 16, stride: 1440, 259200Byte,  lv_draw_buf.c:186                                                                                                                                                                                                   
[Warn]  (198.230, +10)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:376                                                                                                                                                                                                               
[Info]  (198.240, +10)   lv_malloc: couldn't allocate memory (259203 bytes) lv_mem.c:73                                                                                                                                                                                                                                 
[Info]  (198.248, +8)    lv_malloc: used:   8448 ( 28 %), frag:   1 %, biggest free:  22596 lv_mem.c:77                                                                                                                                                                                                                 
[Warn]  (198.256, +8)    lv_draw_buf_create: No memory: 360x180, cf: 16, stride: 1440, 259200Byte,  lv_draw_buf.c:186                                                                                                                                                                                                   
[Warn]  (198.266, +10)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:376                                                                                                                                                                                                               
[Info]  (198.276, +10)   lv_malloc: couldn't allocate memory (259203 bytes) lv_mem.c:73                                                                                                                                                                                                                                 
[Info]  (198.284, +8)    lv_malloc: used:   8448 ( 28 %), frag:   1 %, biggest free:  22596 lv_mem.c:77                                                                                                                                                                                                                 
[Warn]  (198.292, +8)    lv_draw_buf_create: No memory: 360x180, cf: 16, stride: 1440, 259200Byte,  lv_draw_buf.c:186                                                                                                                                                                                                   
[Warn]  (198.302, +10)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:376                                                                                                                                                                                                               
[Info]  (198.312, +10)   lv_malloc: couldn't allocate memory (259203 bytes) lv_mem.c:73                                                                                                                                                                                                                                 
[Info]  (198.318, +6)    lv_malloc: used:   8448 ( 28 %), frag:   1 %, biggest free:  22596 lv_mem.c:77                                                                                                                                                                                                                 
[Warn]  (198.328, +10)   lv_draw_buf_create: No memory: 360x180, cf: 16, stride: 1440, 259200Byte,  lv_draw_buf.c:186                                                                                                                                                                                                   
[Warn]  (198.338, +10)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:376                                                                                                                                                                                                               
[Info]  (198.346, +8)    lv_malloc: couldn't allocate memory (259203 bytes) lv_mem.c:73                                                                                                                                   [... cut ...]
[Info]  (198.390, +8)    lv_malloc: used:   8448 ( 28 %), frag:   1 %, biggest free:  22596 lv_mem.c:77                                                                                                                                                                                                                 
[Warn]  (198.398, +8)    lv_draw_buf_create: No memory: 360x180, cf: 16, stride: 1440, 259200Byte,  lv_draw_buf.c:186    
```

The configuration of lvgl should be pretty much the default from the esp-idf build. the sdkconfig is present in that repository as well

kisvegabor, 2024-01-30:
Does it work if you remove `lv_obj_set_style_clip_corner(obj, true)`?

silverchris, 2024-01-31:
Yup, removing that seems to fix the issue!
Now to figure out how to get the threaded renderer working. 
Any gotchas with that? I do need to stop and start the LVGL thread normally, as I only use LVGL for a menu at times, suspending the task and stopping the timer when it's not in use

kisvegabor, 2024-02-01:
I haven't tried it with ESP, but it should work by setting these:
- `LV_USE_OS LV_OS_FREERTOS`
- `LV_DRAW_SW_DRAW_UNIT_CNT 2`

And that's it. I hope it work well on ESP too.

silverchris, 2024-02-01:
> I haven't tried it with ESP, but it should work by setting these:
> - `LV_USE_OS LV_OS_FREERTOS`
> - `LV_DRAW_SW_DRAW_UNIT_CNT 2`
> 
> And that's it. I hope it work well on ESP too.

I tried it quickly yesterday, but couldn't get it to compile with `LV_USE_OS LV_OS_FREERTOS` I think esp-idf has a difference in `taskENTER_CRITICAL()` from - vanilla FreeRTos, as well as some header location differences. Ran out of time before I could get into it though

silverchris, 2024-02-02:
Modified it with the correct include paths, and changed this function to work with esp-idf, but it seems to result in it locking up when creating an object.
```C
portMUX_TYPE lvgl_portmux = portMUX_INITIALIZER_UNLOCKED;

static void prvCheckMutexInit(lv_mutex_t * pxMutex)
{
    /* Check if the mutex needs to be initialized. */
    if(pxMutex->xIsInitialized == pdFALSE) {
        /* Mutex initialization must be in a critical section to prevent two threads
         * from initializing it at the same time. */
        taskENTER_CRITICAL(&lvgl_portmux);

        /* Check again that the mutex is still uninitialized, i.e. it wasn't
         * initialized while this function was waiting to enter the critical
         * section. */
        if(pxMutex->xIsInitialized == pdFALSE) {
            prvMutexInit(pxMutex);
        }

        /* Exit the critical section. */
        taskEXIT_CRITICAL(&lvgl_portmux);
    }
}
```

kisvegabor, 2024-02-02:
It's interesting that it's locking during widget creation as in LVGL FreeRTOS is involved only during rendering.

Are you sure `lv_timer_handler` is not reached?

kisvegabor, 2024-02-02:
Increasing the stack size of the LVGL task is also worth a try.

silverchris, 2024-02-03:
This works if I set LV_USE_OS to LV_OS_NONE. It doesn't seem to be affected if I set draw threads to 1 as well.

I should note that the watchdog timer message only shows up with trace level logging on, so I believe that is just from the shear amount of logging LVGL is doing during startup.

[lvgl.log](https://github.com/lvgl/lvgl/files/14149465/lvgl.log)

silverchris, 2024-02-03:
Interestingly, if I reorder the display setup before starting the lv_timer_handler task, it starts faster and logs less.
It looks like it is hanging during some part of the lv_timer_handler task, but it is yielding, as the watchdog timeout isn't happening, I think?
[lvgl.log](https://github.com/lvgl/lvgl/files/14149504/lvgl.log)

kisvegabor, 2024-02-07:
I found only this in the log:
```
[Warn]  (0.526, +4)      _lv_display_refr_timer: No draw buffer lv_refr.c
```

Which indicates that you have already created an `lv_display` but haven't assigned a buffer with `lv_display_set_buffers()`.

C47D, 2024-02-07:
Maybe related to #5271

silverchris, 2024-02-08:
> I found only this in the log:
> 
> ```
> [Warn]  (0.526, +4)      _lv_display_refr_timer: No draw buffer lv_refr.c
> ```
> 
> Which indicates that you have already created an `lv_display` but haven't assigned a buffer with `lv_display_set_buffers()`.

That only appeared in the log before I changed the order of initialization, to create the display before before starting the lvgl task.

kisvegabor, 2024-02-10:
Could you share a longer initialization code to see how/when you call `lv_init`, create the display and call `lv_timer_hander`?

silverchris, 2024-02-11:
> Could you share a longer initialization code to see how/when you call `lv_init`, create the display and call `lv_timer_handler`?

Here is the changes I made to LVGL itself to get it to compile
https://github.com/silverchris/lvgl
And here is a repository with my test case, and config and such
https://github.com/silverchris/lvgl_test/blob/main/main/lvgl_test.c

kisvegabor, 2024-02-12:
> Here is the changes I made to LVGL itself to get it to compile

Thank you! 
@espzav Have you also tried to enable `LV_USE_OS LV_OS_FFREERTOS` and encountered the issues fixed [here](https://github.com/silverchris/lvgl/commit/3e1d5d840b96f8b76e680bf20619d356562c0bda)?

> And here is a repository with my test case, and config and such

Your `lvgl_test.c` looks good. Finally what is the current status? Is it working just there are some extra logs?

silverchris, 2024-02-12:
Sorry this is still not working, just the code I changed to get it to
compile. It is still hanging when trying to create an object

silverchris, 2024-02-15:
> Okay, I see. Let's investigate what happens. First we should find the offending line. Please add these:
> 
> ```c
>    printf("0\n");
>     lvgl_lock(-1);     printf("1\n");
>     lv_obj_t *label = lv_label_create(lv_scr_act());      printf("2\n");
>     lv_obj_set_pos(label, 240, 240);     printf("3\n");
>     lv_label_set_text(label, "It works?");     printf("4\n");
>     lvgl_unlock();     printf("5\n");
> ```
This is what I get, it seems that lv_timer_handler is hanging and therefor we never get the lock?
```
Build:Mar 27 2021                                                                                                                                                                                                                                                                                                       
rst:0x15 (USB_UART_CHIP_RESET),boot:0xb (SPI_FAST_FLASH_BOOT)                                                                                                                                                                                                                                                           
Saved PC:0x40378c67                                                                                                                                                                                                                                                                                                     
SPIWP:0xee                                                                                                                                                                                                                                                                                                              
mode:DOUT, clock div:1                                                                                                                                                                                                                                                                                                  
load:0x3fce3818,len:0x18e4                                                                                                                                                                                                                                                                                              
load:0x403c9700,len:0x4                                                                                                                                                                                                                                                                                                 
load:0x403c9704,len:0xda8                                                                                                                                                                                                                                                                                               
load:0x403cc700,len:0x30a4                                                                                                                                                                                                                                                                                              
entry 0x403c9934                                                                                                                                                                                                                                                                                                        
I (26) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader                                                                                                                                                                                                                                                                  
I (27) boot: compile time Feb  1 2024 19:31:00                                                                                                                                                                                                                                                                          
I (27) boot: Multicore bootloader                                                                                                                                                                                                                                                                                       
I (30) boot: chip revision: v0.1                                                                                                                                                                                                                                                                                        
I (34) qio_mode: Enabling default flash chip QIO                                                                                                                                                                                                                                                                        
I (39) boot.esp32s3: Boot SPI Speed : 80MHz                                                                                                                                                                                                                                                  [... cut ...]                                                                                                                
[Trace] (0.000, +0)      lv_realloc: reallocating 0x3fc9d374 with 30 size lv_mem.c:130                                                                                                                                                                                                                                  
[Trace] (0.000, +0)      lv_realloc: reallocated at 0x3fc9d374 lv_mem.c:146

kisvegabor, 2024-02-16:
> I (5713) Test: 0     

Is `printf("0\n");`?

> This is what I get, it seems that lv_timer_handler is hanging and therefor we never get the lock?

Does it change something if you don't create a task for lv_timer_handler()?
```c
    lv_obj_t *label = lv_label_create(lv_scr_act());      ESP_LOGI(TAG, "2\n");
    lv_obj_set_pos(label, 240, 240);     ESP_LOGI(TAG, "3\n");
    lv_label_set_text(label, "It works?");     ESP_LOGI(TAG, "4\n");
    while(true){
        lv_timer_handler()
        ESP_LOGI(TAG, "LOOPING %lli", esp_timer_get_time());
        vTaskDelay(1000/portTICK_PERIOD_MS);
    }
```

> event_send_core: Sending event 27 to 0x3fc9d4fc with 0x3fc9c8c8 param lv_obj_event.c:323

event 27 is `LV_EVENT_DRAW_POST_END` which is normally sent many times during rendering.

silverchris, 2024-02-18:
> Does it change something if you don't create a task for lv_timer_handler()?
>
> ```c
>     lv_obj_t *label = lv_label_create(lv_scr_act());      ESP_LOGI(TAG, "2\n");
>     lv_obj_set_pos(label, 240, [...]

It seems to be working if I call `lv_timer_handler` from main.
Guessing something must be causing a deadlock?

espzav, 2024-02-19:
Hi @kisvegabor I didn't try to use `LV_USE_OS`, when I enable it, I cannot compile my project. 
But it seems that I have simillar issue with new MIPI-DSI screen on ESP32-P4. I am trying it with our examples and when I use LVGL8, it is working. But when I change it to LVGL9, it show only the first flush (200 lines) and then it hangs. Seems, it waits to `lv_disp_flush_ready`. 

```
Print CPU 0 (current core) registers
Core  0 register dump:
Stack dump detected
MEPC    : 0x40010adc  RA      : 0x400110d0  SP      : 0x4ff233d0  GP      : 0x4ff11610
0x40010adc: wait_for_flushing at C:/WORK/GIT/esp-bsp/examples/display_rotation/managed_components/lvgl__lvgl/src/core/lv_refr.c:1062 (discriminator 1)
0x400110d0: draw_buf_flush at C:/WORK/GIT/esp-bsp/examples/display_rotation/managed_components/lvgl__lvgl/src/core/lv_refr.c:1009

TP      : 0x4ff05fa8  T0      : 0x4fc19a74  T1      : 0x0000000f  T2      : 0x00000000
0x4ff05fa8: rtc_clk_cal at C:/WORK/GIT/esp-idf_GitLab/components/esp_hw_support/port/esp32p4/rtc_time.c:193

S0/FP   : 0x4ff23a28  S1      : 0x4ff23d44  A0      : 0x4ff23a28  A1      : 0x4ff23d3c
A2      : 0x00000001  A3      : 0x1f000000  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00000001  A7      : 0xffffffc9  S2      : 0x00000000  S3      : 0x4ff23d5c
S4      : 0x0000027f  S5      : 0x00000140  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0xffff0879  T4      : 0x000001cb  T5      : 0x00000000  T6      : 0x0000012a
MSTATUS : 0x00000000  MTVEC   : 0x4ff23d44  MCAUSE  : 0x4ff23a28  MTVAL   : 0x400110d0
0x400110d0: draw_buf_flush at C:/WORK/GIT/esp-bsp/examples/display_rotation/managed_components/lvgl__lvgl/src/core/lv_refr.c:1009
```

I think, it can be related to the issue, what I told before. About the initialization of the LVGL timer and task, after that I add the screen. But I do not call `lv_timer_handler` till there is not initialized screen. I cannto find right solution for this :-(

kisvegabor, 2024-02-22:
> I think, it can be related to the issue, what I told before. About the initialization of the LVGL timer and task, after that I add the screen.

If there the display is not registered than no rendering happens and therefore the `flush_cb` is not called either. Are you sure that you call `lv_display_flush_ready`?

espzav, 2024-02-22:
> If the display is not registered then no rendering happens and therefore the `flush_cb` is not called either. Are you sure that you call `lv_display_flush_ready`?

Yes, it is working with LVGL8 (same initialization, same LCD driver, same callback that data flushed). I am using released version of LVGL9 now.

kisvegabor, 2024-02-24:
Please update to the latest master as a lot of initial issue are already fixed.

mr-sven, 2024-02-26:
Is it possible to include the changes done in https://github.com/silverchris/lvgl via macro? Something like `#if CONFIG_IDF_TARGET`?
So CONFIG_LV_USE_OS = LV_OS_FREERTOS can be used in ESP-IDF builds.

silverchris, 2024-03-03:
I just tested again with the latest master, and the issue still exists for me

mr-sven, 2024-03-12:
This is my Implementation with the FreeRTOS ESP fix https://github.com/mr-sven/esp32-8048S050C/tree/lvgl-test in the logs I see no render event message. The tick timer is running and the FreeRTOS Task but no render callback

mr-sven, 2024-03-13:
My log dump without memlog:
```
I (906) LVGL: lv_init
[Info]  (0.000, +0)      lv_init: begin lv_init.c:139
[Warn]  (0.000, +0)      lv_init: Log level is set to 'Trace' which makes LVGL much slower lv_init.c:244
[Trace] (0.000, +0)      lv_init: finished lv_init.c:312
I (925) LVGL: lv_display_create
[Info]  (0.000, +0)      lv_obj_create: begin lv_obj.c:100
[Trace] (0.000, +0)      lv_obj_class_create_obj: Creating object with 0x3c073fec class on 0x0 parent lv_obj_class.c:48
[Trace] (0.000, +0)      lv_obj_class_create_obj: creating a screen lv_obj_class.c:57
[Trace] (0.000, +0)      lv_obj_constructor: begin lv_obj.c:320
[Trace] (0.000, +0)      lv_obj_constructor: finished lv_obj.c:349
[Trace] (0.000, +0)      event_send_core: Sending event 44 to 0x3fc9a33c with 0x0 param lv_obj_event.c:330
[Trace] (0.000, +0)      event_send_core: Sending event 21 to 0x3fc9a33c with 0x3fcac3d0 param lv_obj_event.c:330
[Info]  (0.000, +0)      lv_obj_create: begin lv_obj.c:100
[Trace] (0.000, +0)      lv_obj_class_create_obj: Creating object with 0x3c073fec class on 0x0 parent lv_obj_class.c:48
[Trace] (0.000, +0)      lv_obj_class_create_obj: creating a screen lv_obj_class.c:57
[Trace] (0.000, +0)      lv_obj_constructor: begin lv_obj.c:320
[Trace] (0.000, +0)      lv_obj_constructor: finished lv_obj.c:349
[Trace] (0.000, +0)      event_send_core: Sending event 44 to 0x3fc9a398 with 0x0 param lv_obj_event.c:330
[Trace] (0.000, +0)      event_send_core: Sending event 21 to 0x3fc9a398 with 0x3fcac3d0 param lv_obj_event.c:330
[Info]  (0.000, +0)      lv_obj_create: begin lv_obj.c:100
[Trace] (0.000, +0)      lv_obj_class_create_obj: Creating object with 0x3c073fec class on 0x0 parent lv_obj_class.c:48
[Trace] (0.000, +0)      lv_obj_class_create_obj: creating a screen lv_obj_class.c:57
[Trace] (0.000, +0)      lv_obj_constructor: begin lv_obj.c:320
[Trace] (0.000, +0)      lv_obj_constructor: finished lv_obj.c:349
[Trace] (0.000, +0)      event_send_core: Sending event 44 to 0x3fc9a3e4 with 0x0 param lv_obj_event.c:330
[Trace] (0.000, +0)      event_send_core: Sending event 21 to 0x3fc9a3e4 with 0x3fcac3d0 param lv_obj_event.c:330
[Info]  (0.000, +0)      lv_obj_create: begin lv_obj.c:100
[Trace] (0.000, +0)      lv_obj_class_create_obj: Creating object with 0x3c073fec class on 0x0 parent lv_obj_class.c:48
[Trace] (0.000, +0)      lv_obj_class_create_obj: creating a screen lv_obj_class.c:57
[Trace] (0.000, +0)      lv_obj_constructor: begin lv_obj.c:320
[Trace] (0.000, +0)      lv_obj_constructor: finished lv_obj.c:349
[Trace] (0.000, +0)      event_send_core: Sending event 44 to 0x3fc9a430 with 0x0 param lv_obj_event.c:330
[Trace] (0.000, +0)      event_send_core: Sending event 21 to 0x3fc9a430 with 0x3fcac3d0 param lv_obj_event.c:330
[Trace] (0.000, +0)      event_send_core: Sending event 44 to 0x3fc9a33c with 0x0 param lv_obj_event.c:330
[Trace] (0.000, +0)      event_send_core: Sending event 21 to 0x3fc9a33c with 0x3fcac3c0 param lv_obj_event.c:330
[Trace] (0.000, +0)      event_send_core: Sending event 44 to 0x3fc9a3e4 with 0x0 param lv_obj_event.c:330
[Trace] (0.000, +0)      event_send_core: Sending event 21 to 0x3fc9a3e4 with 0x3fcac3c0 param lv_obj_event.c:330
[Trace] (0.000, +0)      event_send_core: Sending event 44 to 0x3fc9a430 with 0x0 param lv_obj_event.c:330
[Trace] (0.000, +0)      event_send_core: Sending event 21 to 0x3fc9a430 with 0x3fcac3c0 param lv_obj_event.c:330
I (1239) LVGL: lv_display_set_user_data
I (1239) LVGL: lv_display_set_flush_cb
I (1239) LVGL: lv_display_set_buffers
I (1241) LVGL: Install LVGL tick timer
I (1298) LVGL: lv_indev_create
I (1300) LVGL: lv_label_create
[Info]  (0.058, +58)     lv_label_create: begin lv_label.c:83
[Trace] (0.062, +4)      lv_obj_class_create_obj: Creating object with 0x3c073924 class on 0x3fc9a398 parent lv_obj_class.c:48
[Trace] (0.074, +12)     lv_obj_class_create_obj: creating normal object lv_obj_class.c:81
[Trace] (0.082, +8)      lv_obj_constructor: begin lv_obj.c:320
[Trace] (0.086, +4)      lv_obj_constructor: finished lv_obj.c:349
[Trace] (0.092, +6)      lv_label_constructor: begin lv_label.c:623
[Trace] (0.098, +6)      lv_label_constructor: finished lv_label.c:650
[Trace] (0.104, +6)      event_send_core: Sending event 44 to 0x3fc9a5f4 with 0x0 param lv_obj_event.c:330
[Trace] (0.112, +8)      event_send_core: Sending event 21 to 0x3fc9a5f4 with 0x3fcac410 param lv_obj_event.c:330
[Trace] (0.122, +10)     event_send_core: Sending event 36 to 0x3fc9a398 with 0x3fc9a5f4 param lv_obj_event.c:330
[Trace] (0.132, +10)     event_send_core: Sending event 37 to 0x3fc9a398 with 0x3fc9a5f4 param lv_obj_event.c:330
[Trace] (0.142, +10)     event_send_core: Sending event 44 to 0x3fc9a5f4 with 0x0 param lv_obj_event.c:330
[Trace] (0.150, +8)      event_send_core: Sending event 44 to 0x3fc9a5f4 with 0x0 param lv_obj_event.c:330

```

kisvegabor, 2024-03-13:
cc @zjanosy 
I think we should do some experiments with it too.

kisvegabor, 2024-04-08:
@mr-sven is [working on it](https://github.com/lvgl/lvgl/issues/6039#issue-2230523916).

mr-sven, 2024-04-09:
I've added now the timer handler : https://github.com/mr-sven/esp32-8048S050C/blob/lvgl-test/main/main.c 
Now the app is freezing on lock, because the timer is crashing:
```
I (901) LVGL: Turn on LCD backlight
I (907) LVGL: lv_init
[Info]  (0.000, +0)      lv_init: begin lv_init.c:139
[Warn]  (0.000, +0)      lv_init: Log level is set to 'Trace' which makes LVGL much slower lv_init.c:244
[Trace] (0.000, +0)      lv_init: finished lv_init.c:308
I (926) LVGL: lv_display_create
[Info]  (0.000, +0)      lv_obj_create: begin lv_obj.c:100
[Trace] (0.000, +0)      lv_obj_class_create_obj: Creating object with 0x3c07431c class on 0x0 parent lv_obj_class.c:48
[Trace] (0.000, +0)      lv_obj_class_create_obj: creating a screen lv_obj_class.c:57
[Trace] (0.000, +0)      lv_obj_constructor: begin lv_obj.c:320
[Trace] (0.000, +0)      lv_obj_constructor: finished lv_obj.c:349
[Trace] (0.000, +0)      event_send_core: Sending event 45 to 0x3fc9a360 with 0x0 param lv_obj_event.c:343
[Trace] (0.000, +0)      event_send_core: Sending event 22 to 0x3fc9a360 with 0x3fcac3c0 param lv_obj_event.c:343
[Info]  (0.000, +0)      lv_obj_create: begin lv_obj.c:100
[Trace] (0.000, +0)      lv_obj_class_create_obj: Creating object with 0x3c07431c class on 0x0 parent lv_obj_class.c:48
[Trace] (0.000, +0)      lv_obj_class_create_obj: creating a screen lv_obj_class.c:57
[Trace] (0.000, +0)      lv_obj_constructor: begin lv_obj.c:320
[Trace] (0.000, +0)      lv_obj_constructor: finished lv_obj.c:349
[Trace] (0.000, +0)      event_send_core: Sending event 45 to 0x3fc9a3bc with 0x0 param lv_obj_event.c:343
[Trace] (0.000, +0)      event_send_core: Sending event 22 to 0x3fc9a3bc with 0x3fcac3c0 param lv_obj_event.c:343
[Info]  (0.000, +0)      lv_obj_create: begin lv_obj.c:100
[Trace] (0.000, +0)      lv_obj_class_create_obj: Creating object with 0x3c07431c class on 0x0 parent lv_obj_class.c:48
[Trace] (0.000, +0)      lv_obj_class_create_obj: creating a screen lv_obj_class.c:57
[Trace] (0.000, +0)      lv_obj_constructor: begin lv_obj.c:320
[Trace] (0.000, +0)      lv_obj_constructor: finished lv_obj.c:349
[Trace] (0.000, +0)      event_send_core: Sending event 45 to 0x3fc9a408 with 0x0 param lv_obj_event.c:343
[Trace] (0.000, +0)      event_send_core: Sending event 22 to 0x3fc9a408 with 0x3fcac3c0 param lv_obj_event.c:343
[Info]  (0.000, +0)      lv_obj_create: begin lv_obj.c:100
[Trace] (0.000, +0)      lv_obj_class_create_obj: Creating object with 0x3c07431c class on 0x0 parent lv_obj_class.c:48
[Trace] (0.000, +0)      lv_obj_class_create_obj: creating a screen lv_obj_class.c:57
[Trace] (0.000, +0)      lv_obj_constructor: begin lv_obj.c:320
[Trace] (0.000, +0)      lv_obj_constructor: finished lv_obj.c:349
[Trace] (0.000, +0)      event_send_core: Sending event 45 to 0x3fc9a454 with 0x0 param lv_obj_event.c:343
[Trace] (0.000, +0)      event_send_core: Sending event 22 to 0x3fc9a454 with 0x3fcac3c0 param lv_obj_event.c:343
[Trace] (0.000, +0)      event_send_core: Sending event 45 to 0x3fc9a360 with 0x0 param lv_obj_event.c:343
[Trace] (0.000, +0)      event_send_core: Sending event 22 to 0x3fc9a360 with 0x3fcac3b0 param lv_obj_event.c:343
[Trace] (0.000, +0)      event_send_core: Sending event 45 to 0x3fc9a408 with 0x0 param lv_obj_event.c:343
[Trace] (0.000, +0)      event_send_core: Sending event 22 to 0x3fc9a408 with 0x3fcac3b0 param lv_obj_event.c:343
[Trace] (0.000, +0)      event_send_core: Sending event 45 to 0x3fc9a454 with 0x0 param lv_obj_event.c:343
[Trace] (0.000, +0)      event_send_core: Sending event 22 to 0x3fc9a454 with 0x3fcac3b0 param lv_obj_event.c:343
I (1192) LVGL: lvgl_lcd_init
I (1240) LVGL: lv_display_set_user_data
I (1240) LVGL: lv_display_set_flush_cb
I (1240) LVGL: lv_display_set_buffers
I (1242) LVGL: Install LVGL tick timer
I (1247) LVGL: lvgl_port_task
[Trace] (0.002, +2)      lv_timer_handler: begin lv_timer.c:65
[Trace] (0.008, +6)      lv_timer_exec: calling timer callback: 0x4200f314 lv_timer.c:298
0x4200f314: _lv_display_refr_timer at /home/sven/ESP32-8048S050C/build/../managed_components/lvgl__lvgl/src/core/lv_refr.c:325

[Trace] (0.016, +8)      _lv_display_refr_timer: begin lv_refr.c:327
[Trace] (0.022, +6)      lv_obj_update_layout: Layout update begin lv_obj_pos.c:301
[Trace] (0.028, +6)      lv_obj_update_layout: Layout update end lv_obj_pos.c:304
[Trace] (0.036, +8)      lv_obj_update_layout: Layout update begin lv_obj_pos.c:301
[Trace] (0.042, +6)      lv_obj_update_layout: Layout update end lv_obj_pos.c:304
[Trace] (0.048, +6)      lv_obj_update_layout: Layout update begin lv_obj_pos.c:301
[Trace] (0.056, +8)      lv_obj_update_layout: Layout update end lv_obj_pos.c:304
[Trace] (0.062, +6)      lv_obj_update_layout: Layout update begin lv_obj_pos.c:301
[Trace] (0.070, +8)      lv_obj_update_layout: Layout update end lv_obj_pos.c:304
[Trace] (0.076, +6)      event_send_core: Sending event 21 to 0x3fc9a3bc with 0x3fcbb4d0 param lv_obj_event.c:343
[Trace] (0.086, +10)     event_send_core: Sending event 23 to 0x3fc9a3bc with 0x3fc99818 param lv_obj_event.c:343
[Trace] (0.096, +10)     event_send_core: Sending event 24 to 0x3fc9a3bc with 0x3fc99818 param lv_obj_event.c:343
[Trace] (0.106, +10)     event_send_core: Sending event 25 to 0x3fc9a3bc with 0x3fc99818 param lv_obj_event.c:343
[Trace] (0.116, +10)     event_send_core: Sending event 26 to 0x3fc9a3bc with 0x3fc99818 param lv_obj_event.c:343
[... cut ...]
[Trace] (0.274, +8)      event_send_core: Sending event 28 to 0x3fc9a454 with 0x3fc99818 param lv_obj_event.c:343
I (1532) gpio: GPIO[19]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 
I (1542) gpio: GPIO[20]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 
I (1551) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1580) GT911: TouchPad_ID:0x39,0x31,0x31
I (1580) GT911: TouchPad_Config_Version:71
I (1580) LVGL: lv_indev_create
I (1582) LVGL: lvgl_lock
```

mr-sven, 2024-04-09:
Calling `lv_timer_handler()` once from app main will draw data, calling it from task don't draw.
Logs from main app:
```
0x4200f138: _lv_display_refr_timer at /home/sven/ESP32-8048S050C/build/../managed_components/lvgl__lvgl/src/core/lv_refr.c:325

[Trace] (0.136, +8)      _lv_display_refr_timer: begin lv_refr.c:327
[Trace] (0.142, +6)      lv_obj_update_layout: Layout update begin lv_obj_pos.c:301
...
[Trace] (0.600, +10)     event_send_core: Sending event 23 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.610, +10)     event_send_core: Sending event 24 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.620, +10)     event_send_core: Sending event 25 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.628, +8)      event_send_core: Sending event 26 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.638, +10)     event_send_core: Sending event 27 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.648, +10)     event_send_core: Sending event 28 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.658, +10)     wait_for_flushing: begin lv_refr.c:1076
[Trace] (0.664, +6)      wait_for_flushing: end lv_refr.c:1090
[Trace] (0.668, +4)      call_flush_cb: Calling flush_cb on (0;0)(799;479) area with 0x3c070b80 image pointer lv_refr.c:1056
I (1915) LVGL: flush_cb
[Trace] (0.684, +16)     wait_for_flushing: begin lv_refr.c:1076
[Trace] (0.688, +4)      wait_for_flushing: end lv_refr.c:1090
[Trace] (0.694, +6)      _lv_display_refr_timer: finished lv_refr.c:410
[Trace] (0.700, +6)      lv_timer_exec: timer callback 0x4200f138 finished lv_timer.c:303
0x4200f138: _lv_display_refr_timer at /home/sven/ESP32-8048S050C/build/../managed_components/lvgl__lvgl/src/core/lv_refr.c:325
```
Logs from task:
```
0x4200f1a8: _lv_display_refr_timer at /home/sven/ESP32-8048S050C/build/../managed_components/lvgl__lvgl/src/core/lv_refr.c:325

[Trace] (0.136, +8)      _lv_display_refr_timer: begin lv_refr.c:327
[Trace] (0.142, +6)      lv_obj_update_layout: Layout update begin lv_obj_pos.c:301
...
[Trace] (0.574, +8)      event_send_core: Sending event 23 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.584, +10)     event_send_core: Sending event 24 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.594, +10)     event_send_core: Sending event 25 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.604, +10)     event_send_core: Sending event 26 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.614, +10)     event_send_core: Sending event 27 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
[Trace] (0.624, +10)     event_send_core: Sending event 28 to 0x3fc99c50 with 0x3fc99014 param lv_obj_event.c:343
```
So it's not moving on to `wait_for_flushing`.
The Deadlock is happening here: https://github.com/lvgl/lvgl/blob/v9.1.0/src/core/lv_refr.c#L1020

It looks like a problem with the FreeRTOS task notify functionality, I've no experience with this, but maybe there is again some ESP IDF specific behavior like `taskENTER_CRITICAL`.

mr-sven, 2024-04-10:
@silverchris can you test the following on your side? I called the `lv_draw_init` function at task start again and it's drawing now:
```c
static void lvgl_port_task(void *arg)
{
    lv_draw_init();
    
    uint32_t task_delay_ms = LVGL_TASK_MAX_DELAY_MS;
    while (1)
    {
        // Lock the mutex due to the LVGL APIs are not thread-safe
        if (lvgl_lock(-1))
        {
            task_delay_ms = lv_timer_handler();
            // Release the mutex
            lvgl_unlock();
        }
        if (task_delay_ms > LVGL_TASK_MAX_DELAY_MS)
        {
            task_delay_ms = LVGL_TASK_MAX_DELAY_MS;
        }
        else if (task_delay_ms < LVGL_TASK_MIN_DELAY_MS)
        {
            task_delay_ms = LVGL_TASK_MIN_DELAY_MS;
        }
        vTaskDelay(pdMS_TO_TICKS(task_delay_ms));
    }
}
```

silverchris, 2024-04-11:
I will give it a shot this weekend!

kisvegabor, 2024-06-20:
I close this issue because there was no activity here for a while.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2024-06-20 by kisvegabor

lewisxhe, 2024-07-20:
The current test main branch also has the same problem, as long as the update will immediately assert
[image]