lvgl/lvgl #5519: SD card display image

joeyandyou, 2024-01-29:
### LVGL version

8.3

### What happened?

When I was using lvgl, I used the lvgl file system to display images. Due to limited memory, I saw that the underlying call of lvgl was to brush image data line by line. When I used it with the screen sliding switch window, the image data would be scrambled until the window switch was successful, and the image would only appear normal. However, when I used the internal c array image, this situation did not occur. How can I solve this problem?

kisvegabor, 2024-01-29:
cc @XuNeo 
What option do we have to read bin images from file system in v9?

XuNeo, 2024-01-29:
In v9, we still use same method for low memory device. If the image can't be read all to RAM like the C-array image, it cannot perform transformation.

kisvegabor, 2024-01-29:
I think it's not the transformations, but the image is read line by line  in v8 which is slow. I wonder if in v9 we have more option to manage this.

closed 2024-01-30 by joeyandyou

kisvegabor, 2024-01-30:
I just want to get a confirmation about in v9 `get_area_cb` reads the images line by line too. It seems this is the case in [lv_bin_decoder_get_area](https://github.com/lvgl/lvgl/blob/master/src/libs/bin_decoder/lv_bin_decoder.c#L371).