lvgl/lvgl #5522: Canvas memory leak

kisvegabor, 2024-01-29:
### LVGL version

v9.0.0

### What happened?

The canvas draw buffer is not freed when the canvas is deleted. 

### How to reproduce?

```c
  lv_demo_vector_graphic();
  lv_obj_clean(lv_screen_active());
```

It can be easily fixed by adding `lv_draw_buf_destroy(canvas->draw_buf);` in `lv_canvas_destructor` but I'm not sure it's the best way considering `lv_canvas_set_buffer()`.

cc @XuNeo

XuNeo, 2024-01-29:
Hi Gabor,
Since the draw buffer is provided by user, I thinks it's better to let user to free it. It's what we have done in internal project.

I know that provide automatic draw buffer create/destroy will save a lot, I'm not sure if it will make the canvas more complicated.

kisvegabor, 2024-01-29:
What do you think about adding `lv_canvas_set_auto_destroy_buf(canvas, true/false)`?

FASTSHIFT, 2024-01-29:
How about adding a reference count to draw_buf, since it is possible for a draw_buf to be referenced by multiple objects, draw_buf is released at the last dereference.

FASTSHIFT, 2024-01-29:
Or let the user choose canvas to automatically create and destroy draw_buf, and the life cycle of the canvas object is consistent with draw_buf.

kisvegabor, 2024-01-30:
I think reference counting is not a good idea, as you might want to keep a draw buffer even if it's not referenced temporarily.

> Or let the user choose canvas to automatically create and destroy draw_buf, and the life cycle of the canvas object is consistent with draw_buf.

Does it mean we set only the size and color format and canvas will allocate/deallocate the draw buf?

PGNetHun, 2024-01-30:
Making canvas (or adding a separate create function) to allocate draw buffer itself (by giving size and color format), and then destroying automatically it is better in my opinion… at the moment we do the same manually, and it’s an (annoying) needless step.

XuNeo, 2024-01-30:
> Does it mean we set only the size and color format and canvas will allocate/deallocate the draw buf?

Yes. We does have another canvas extension widget has this feature. It accept `w, h, cf` only, and automatically manage draw buffer. It won't have API to set external draw buffer, thus much clear.

kisvegabor, 2024-01-30:
Having a function like this seems really clean.

Any good name ideas?

closed 2024-02-14 by kisvegabor