lvgl/lvgl #5524: [Suggestion] Reconsider bar/slider and spinner/arc

hyansuper, 2024-01-30:
### Introduce the problem

The relation between bar and slider is similar to that of a "static_spinner"(or "knobless_arc") and an arc, but their set_value functions are inconsistent, bar and slider have built-in animation support, while arcs don't.

### Proposal

1. Move knob-related functions in arc's source code to `lv_arc_slider.c` (or some better name), now arc become knobless. 
Make lv_arc_slider_t extends from lv_arc_t, just like lv_slider_t extends from lv_bar_t in [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/slider/lv_slider.h#L45). 
Spinner can also extends from arc.

2. Rename [_lv_bar_anim_t](https://github.com/lvgl/lvgl/blob/master/src/widgets/bar/lv_bar.h#L48) to "_lv_progress_anim_t", so it can be internally used be lv_bar_t, lv_slider_t, lv_arc_slider_t, and lv_arc_t, and maybe even scale's needle. 
And we can bind these widgets to observers

3. Remove last parameter in `lv_bar_set_value()` and `lv_slider_set_value()`, and add `lv_xxx_enable_anim(bool en)` function. 
Currently observers bind these widgets with the parameter hard-coded to LV_ANIM_OFF in [here](https://github.com/lvgl/lvgl/blob/master/src/others/observer/lv_observer.c#L676)

kisvegabor, 2024-01-30:
> Move knob-related functions in arc's source code to lv_arc_slider.c (or some better name), now arc become knobless.

I also feel like "arc" is too complex now. We should have a very simple arc widget, which is really just an arc. Based on that "Arc slider" could have
- background arc
- indicator arc
- knob

My only problem with that it's a massive API break. :slightly_frowning_face: 

> Rename [_lv_bar_anim_t](https://github.com/lvgl/lvgl/blob/master/src/widgets/bar/lv_bar.h#L48) to "_lv_progress_anim_t", so it can be internally used be lv_bar_t, lv_slider_t, lv_arc_slider_t, and lv_arc_t, and maybe even scale's needle.
And we can bind these widgets to observers

Sounds good!

> Remove last parameter in lv_bar_set_value() and lv_slider_set_value(), and add lv_xxx_enable_anim(bool en) function.

You can set the animation time with the `anim_time` style property. I think it's important to have a both option on the API level. Setting without animations can be used e.g. during initialization.

hyansuper, 2024-01-31:
> My only problem with that it's a massive API break. 🙁

arc can keep its name, make another simpler widget lv_plain_arc or whatever 

> I think it's important to have a both option on the API level. Setting without animations can be used e.g. during initialization.

Sure. I just hope lv_arc_set_value and lv_bar_set_value have same parameters, and also be able to turn on/off animation when bined with observers

kisvegabor, 2024-02-01:
> arc can keep its name, make another simpler widget lv_plain_arc or whatever

I have no idea for "lv_plain_arc". Arc is so basic. It should be the name of the basic arc.

Anyway, it's an API change that we should keep for v10.

> Sure. I just hope lv_arc_set_value and lv_bar_set_value have same parameters, and also be able to turn on/off animation when bined with observers

:+1:

closed 2024-02-24 by lvgl-bot