lvgl/lvgl #5527: Linear gradient with richer features

xaowang96, 2024-01-30:
### Introduce the problem

### Feature
Currently, lvl only supports linear gradient in horizontal and vertical directions. We need to consider adding more features to it. For example, any angle, larger range than path bound, etc...

### Examples

45-degree

```
<svg width="800" height="800" viewBox="0 0 800 800" fill="none" xmlns="http://www.w3.org/2000/svg">
<rect x="0" y="45" width="100" height="100" fill="url(#paint1_linear_13691_50971)"/>
<defs>
<linearGradient id="paint1_linear_13691_50971" x1="0" y1="0" x2="100" y2="100" gradientUnits="userSpaceOnUse" >
    <stop offset="0" stop-color="red" stop-opacity="1"/>
    <stop offset="1" stop-color="green" stop-opacity="1"/>
</linearGradient>
</defs>
</svg>
```
[image]

Larger range

```
<svg width="800" height="800" viewBox="0 0 800 800" fill="none" xmlns="http://www.w3.org/2000/svg">
<rect x="10" y="10" width="200" height="200" fill="url(#paint0_linear_13691_50971)"/>
<rect x="212" y="10" width="200" height="200" fill="url(#paint0_linear_13691_50971)"/>
<rect x="10" y="212" width="200" height="200" fill="url(#paint0_linear_13691_50971)"/>
<rect x="212" y="212" width="200" height="200" fill="url(#paint0_linear_13691_50971)"/>
<defs>
<linearGradient id="paint0_linear_13691_50971" x1="10" y1="10" x2="400" y2="400" gradientUnits="userSpaceOnUse" >
<stop offset="0" stop-color="red" stop-opacity="1"/>
<stop offset="1" stop-color="green" stop-opacity="1"/>
</linearGradient>
</defs>
</svg>
```

[image]

I submit a [PR](https://github.com/lvgl/lvgl/pull/5516) to achieve these feature, but it is not good enough. Maybe we should discuss how to do it more perfectly. How do you think about that? @kisvegabor @FASTSHIFT

FASTSHIFT, 2024-01-30:
cc @onecoolx

kisvegabor, 2024-01-30:
Thank you for opening this issue.

`linearGradient id="paint0_linear_13691_50971"` uses absolute coordinates which is not the convenient in case of widgets. However, setting larger gradient areas can be really useful in some cases.

I'm not sure using 2 points is the best way to describe a gradient. What if there are 3 points? 

In CSS linear gradients can be described like this:
`linear-gradient(30deg, red 10% , blue 50%, green 80%)`
which seems easier to handle. 

Having this we can provide backward compatibility by:
```
LV_GRAD_DIR_HOR = 0
LV_GRAD_DIR_VER = 90
LV_GRAD_DIR_NONE = INT_MAX
```

Later grad dir can be extended to handle radial and conic gradients too.

xaowang96, 2024-01-31:
> Thank you for opening this issue.
> 
> `linearGradient id="paint0_linear_13691_50971"` uses absolute coordinates which is not the convenient in case of widgets. However, setting larger gradient areas can be really useful in some cases.

In svg, linear gradients have two units: `userSpaceOnUse` means absolute coordinate and `objectBoundingBox` means relative coordinate which relative to the path.

Like this, linear gradient from the top of the object to 50% of it.
```
<svg width="466" height="466" viewBox="0 0 466 466" fill="none" xmlns="http://www.w3.org/2000/svg">
<rect x="20" y="20" width="230" height="230" fill="url(#paint0_linear_13691_50971)"/>
<defs>
<linearGradient id="paint0_linear_13691_50971" x1="0" y1="0" x2="0" y2="0.5" gradientUnits="objectBoundingBox">
<stop offset="0" stop-color="red" stop-opacity="1"/>
<stop offset="1" stop-color="green" stop-opacity="1"/>
</linearGradient>
</defs>
</svg>
```

[image]

So, we can add a unit type into `lv_grad_dsc_t` to determine whether use absolute coordinate or relative coordinate. What do you think about that?

> I'm not sure using 2 points is the best way to describe a gradient. What if there are 3 points?

I want to know what do these 3 points mean.

> In CSS linear gradients can be described like this: `linear-gradient(30deg, red 10% , blue 50%, green 80%)` which seems easier to handle.

How does it describe the starting point of the gradient?

> Having this we can provide backward compatibility by:
> 
> ```
> LV_GRAD_DIR_HOR = 0
> LV_GRAD_DIR_VER = 90
> LV_GRAD_DIR_NONE = INT_MAX
> ```
> 
> Later grad dir can be extended to handle radial and conic gradients too.

It is a great idea to use `LV_GRAD_DIR_` to represent the angle of the gradient.

kisvegabor, 2024-02-01:
Do we know a use case for `absolute coordinate`?

>> In CSS linear gradients can be described like this: linear-gradient(30deg, red 10% , blue 50%, green 80%) which seems easier to handle.

> How does it describe the starting point of the gradient?

In CSS it's described differently:
[image]
https://cssgradient.io/

More info [here](https://developer.mozilla.org/en-US/docs/Web/CSS/gradient/linear-gradient#composition_of_a_linear_gradient):
[image]

xaowang96, 2024-02-02:
> Do we know a use case for `absolute coordinate`?

There are many cases using `absolute coordinate` in svg when desing. Here is a simple example:

[image]

And, This is the definition of linear gradient in svg: [here](https://www.w3.org/TR/2003/REC-SVG11-20030114/pservers.html#LinearGradientsl)

> More info [here](https://developer.mozilla.org/en-US/docs/Web/CSS/gradient/linear-gradient#composition_of_a_linear_gradient): 
![image](https://private-user-images.githubusercontent.com/7599318/301593498-0fa09ef5-3402-40cb-a0c4-e944b7bd8e3f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY4NTkwMDgsIm5iZiI6MTcwNjg1ODcwOCwicGF0aCI6Ii83NTk5MzE4LzMwMTU5MzQ5OC0wZmEwOWVmNS0zNDAyLTQwY2ItYTBjNC1lOTQ0YjdiZDhlM2YucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDJUMDcyNTA4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGMwM2Q2ODBjOTczZDFmMTQyMDVmNzNjMTYwYmQ3MTA5YTVhYWQ1ODc3YjI1ODc5MzY2NjQ2ZWQyMDdjZjhmZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.x_aXW0ownLD1A_XbWewusjOAnxY3yESMqoGfPbws8Pc)

I see. In CSS, gradients are considered to apply to the entire range of obj, which is somewhat different from svg standard.
@kisvegabor @onecoolx  What do you think need to be done to make linear gradients more useful?

onecoolx, 2024-02-02:
> There are many cases using `absolute coordinate` in svg when desing. Here is a simple example: [...]

I add 'lv_vector_path_get_bouding' function to get the bounding box of path. So you can adjust the ‘fill-matrix’ to achieve the effect you want.

xaowang96, 2024-03-05:
Not stale

closed 2024-03-31 by lvgl-bot