lvgl/lvgl #5529: ESP32-S3-WROOM-N8R8: Hangs when calling malloc()

ks75vl, 2024-01-30:
### LVGL version

v9.0.0

### What happened?

**Summary:**
When attempting to create a display using `lv_st7789_create` function, it hangs on indefinitely.Upon tracing the issue, I've founded the problem to the lv_st7789_create function, specifically within the malloc() call indicated in [link](https://github.com/lvgl/lvgl/blob/e4de2d265aa1b251038d18659835502e95d14eaf/src/display/lv_display.c#L75)

Despite attempting to switch between different types of malloc() caps, the problem persists, and the available heap memory remains sufficient.

### How to reproduce?

```c
static void lcd_send_cmd(lv_display_t *disp, const uint8_t *cmd, size_t cmd_size, const uint8_t *param, size_t param_size) { }
static void lcd_send_color(lv_display_t *disp, const uint8_t *cmd, size_t cmd_size, uint8_t *param, size_t param_size) { }

void app_main(void) {

    ESP_ERROR_CHECK(esp_event_loop_create_default());
    lv_display_t *driver = lv_st7789_create(320, 170, 0, lcd_send_cmd, lcd_send_color);
    // ...
}
```

ks75vl, 2024-01-30:
Trace logs output:
```
[Trace] (0.000, +0)      lv_malloc: allocating 8 bytes lv_mem.c:64
[Trace] (0.000, +0)      lv_malloc: allocated at 0x3c060900 lv_mem.c:88
[Trace] (0.000, +0)      lv_malloc_zeroed: allocating 60 bytes lv_mem.c:94
// Hang on
```

kisvegabor, 2024-01-30:
How have you set the [memory size of LVGL](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L38-L57)?

kisvegabor, 2024-01-30:
And have you called `lv_init()`?

ks75vl, 2024-01-31:
You're absolutely right. I overlooked calling the `lv_init` function. After calling the `lv_init` function, everything worked like a charm. Thanks for pointing it out!

[image]

closed 2024-01-31 by ks75vl

kisvegabor, 2024-01-31:
Cool 🙂