lvgl/lvgl #5533: LV_USE_DRAW_SDL currently broken

neuschaefer, 2024-01-30:
### LVGL version

v9.0.0-65-ge4de2d265

### What happened?

When I enable `LV_USE_DRAW_SDL`, compilation fails as follows:

```
[  6%] Building C object lvgl/CMakeFiles/lvgl.dir/src/draw/sdl/lv_draw_sdl.c.o
/.../lvgl/src/draw/sdl/lv_draw_sdl.c: In function ‘draw_to_texture’:
/.../lvgl/src/draw/sdl/lv_draw_sdl.c:177:15: error: ‘lv_layer_t’ {aka ‘struct _lv_layer_t’} has no member named ‘buf’
  177 |     dest_layer.buf = lv_draw_buf_align(sdl_render_buf, LV_COLOR_FORMAT_ARGB8888);
      |               ^
make[2]: *** [lvgl/CMakeFiles/lvgl.dir/build.make:888: lvgl/CMakeFiles/lvgl.dir/src/draw/sdl/lv_draw_sdl.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:132: lvgl/CMakeFiles/lvgl.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

This seems to be due to an API change, but I don't know how to fix it.

C47D, 2024-01-31:
Hi,

Can you try applying this patch to solve the complication error? It's based on this commit https://github.com/lvgl/lvgl/commit/8701dece6561870b052d1ec1380aad5157dd5663

I ran the chart demo but only got a black screen.

```patch
iff --git a/src/draw/sdl/lv_draw_sdl.c b/src/draw/sdl/lv_draw_sdl.c
index b7e41cdce..36b019bef 100644
--- a/src/draw/sdl/lv_draw_sdl.c
+++ b/src/draw/sdl/lv_draw_sdl.c
@@ -174,7 +174,7 @@ static bool draw_to_texture(lv_draw_sdl_unit_t * u, cache_data_t * data)
 
     lv_layer_t dest_layer;
     lv_memzero(&dest_layer, sizeof(dest_layer));
-    dest_layer.buf = lv_draw_buf_align(sdl_render_buf, LV_COLOR_FORMAT_ARGB8888);
+    dest_layer.draw_buf = lv_draw_buf_align(sdl_render_buf, LV_COLOR_FORMAT_ARGB8888);
     dest_layer.color_format = LV_COLOR_FORMAT_ARGB8888;
 
     lv_area_t a;
diff --git a/src/draw/sdl/lv_draw_sdl.h b/src/draw/sdl/lv_draw_sdl.h
index 3979965eb..b394ecca8 100644
--- a/src/draw/sdl/lv_draw_sdl.h
+++ b/src/draw/sdl/lv_draw_sdl.h
@@ -17,7 +17,7 @@ extern "C" {
 
 #if LV_USE_DRAW_SDL
 
-#include <src/misc/cache/lv_cache.h>
+#include "../../misc/cache/lv_cache.h"
 #include "../../misc/lv_area.h"
 #include "../../misc/lv_color.h"
 #include "../../display/lv_display.h"
```

neuschaefer, 2024-02-01:
Hello @C47D,

it does indeed compile, and get further, but then it gets stuck in `execute_drawing` -> `draw_from_cached_texture` -> `lv_draw_dispatch_wait_for_request`

<details>
<summary>backtrace of the main thread, with `LV_OS_PTHREAD`</summary>

```gdb
Thread 1 "demo" received signal SIGINT, Interrupt.
__futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x5555555f35c8 <lv_global+520>) at ./nptl/futex-internal.c:57
57      in ./nptl/futex-internal.c
(gdb) bt
#0  __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x5555555f35c8 <lv_global+520>) at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common (futex_word=futex_word@entry=0x5555555f35c8 <lv_global+520>, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0, cancel=cancel@entry=true)
    at ./nptl/futex-internal.c:87
#2  0x00007ffff7b6c1bb in __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x5555555f35c8 <lv_global+520>, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0)
    at ./nptl/futex-internal.c:139
#3  0x00007ffff7b6e818 in __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x5555555f3578 <lv_global+440>, cond=0x5555555f35a0 <lv_global+480>) at ./nptl/pthread_cond_wait.c:503
#4  ___pthread_cond_wait (cond=cond@entry=0x5555555f35a0 <lv_global+480>, mutex=mutex@entry=0x5555555f3578 <lv_global+440>) at ./nptl/pthread_cond_wait.c:618
#5  0x000055555557e17b in lv_thread_sync_wait (sync=0x5555555f3578 <lv_global+440>) at /.../lvgl/src/osal/lv_pthread.c:129
#6  0x000055555556d05c in lv_draw_dispatch_wait_for_request () at /.../lvgl/src/draw/lv_draw.c:268
#7  0x00005555555b0915 in draw_to_texture (u=<optimized out>, data=0x555555f3ea70) at /.../lvgl/src/draw/sdl/lv_draw_sdl.c:246
#8  0x00005555555bf56b in lv_cache_acquire_or_create (cache=0x555555e07070, key=key@entry=0x7fffffffd590, user_data=user_data@entry=0x555555e07020) at /.../lvgl/src/misc/cache/lv_cache.c:136
#9  0x00005555555b0b3e in draw_from_cached_texture (u=0x555555e07020) at /.../lvgl/src/draw/sdl/lv_draw_sdl.c:324
#10 execute_drawing (u=0x555555e07020) at /.../lvgl/src/draw/sdl/lv_draw_sdl.c:391
#11 dispatch (layer=0x555555e50060, draw_unit=0x555555e07020) at /.../lvgl/src/draw/sdl/lv_draw_sdl.c:150
#12 dispatch (draw_unit=0x555555e07020, layer=0x555555e50060) at /.../lvgl/src/draw/sdl/lv_draw_sdl.c:121
#13 0x000055555556cd61 in lv_draw_dispatch_layer (disp=disp@entry=0x555555e53c50, layer=layer@entry=0x555555e50060) at /.../lvgl/src/draw/lv_draw.c:255
#14 0x000055555556cf23 in lv_draw_dispatch () at /.../lvgl/src/draw/lv_draw.c:161
#15 0x000055555556cfe4 in lv_draw_finalize_task_creation (layer=layer@entry=0x555555e50060, t=t@entry=0x5555567f96e0) at /.../lvgl/src/draw/lv_draw.c:138
#16 0x000055555556fd32 in lv_draw_rect (layer=layer@entry=0x555555e50060, dsc=dsc@entry=0x7fffffffd730, coords=coords@entry=0x7fffffffd720) at /.../lvgl/src/draw/lv_draw_rect.c:187
#17 0x000055555555aa3a in lv_obj_draw (e=<optimized out>) at /.../lvgl/src/core/lv_obj.c:458
#18 lv_obj_event (class_p=<optimized out>, e=<optimized out>) at /.../lvgl/src/core/lv_obj.c:711
#19 0x000055555555c48d in lv_obj_event_base (class_p=class_p@entry=0x0, e=e@entry=0x7fffffffd820) at /.../lvgl/src/core/lv_obj_event.c:86
#20 0x000055555555c522 in event_send_core (e=e@entry=0x7fffffffd820) at /.../lvgl/src/core/lv_obj_event.c:339
#21 0x000055555555c6a8 in lv_obj_send_event (obj=obj@entry=0x555555e511b0, event_code=event_code@entry=LV_EVENT_DRAW_MAIN, param=param@entry=0x555555e50060) at /.../lvgl/src/core/lv_obj_event.c:64
#22 0x000055555556901a in lv_obj_redraw (layer=layer@entry=0x555555e50060, obj=obj@entry=0x555555e511b0) at /.../lvgl/src/core/lv_refr.c:110
#23 0x000055555556945f in refr_obj (layer=layer@entry=0x555555e50060, obj=obj@entry=0x555555e511b0) at /.../lvgl/src/core/lv_refr.c:887
#24 0x0000555555569a21 in refr_obj (obj=0x555555e511b0, layer=0x555555e50060) at /.../lvgl/src/core/lv_refr.c:820
#25 refr_obj_and_children (layer=layer@entry=0x555555e50060, top_obj=<optimized out>, top_obj@entry=0x555555e511b0) at /.../lvgl/src/core/lv_refr.c:784
#26 0x0000555555569b07 in refr_area_part (layer=layer@entry=0x555555e50060) at /.../lvgl/src/core/lv_refr.c:717
#27 0x000055555556a825 in refr_area (area_p=<optimized out>) at /.../lvgl/src/core/lv_refr.c:619
#28 refr_invalid_areas () at /.../lvgl/src/core/lv_refr.c:566
#29 _lv_display_refr_timer (tmr=<optimized out>) at /.../lvgl/src/core/lv_refr.c:374
#30 0x000055555557d81c in lv_timer_exec (timer=0x555555e4f820) at /.../lvgl/src/misc/lv_timer.c:300
#31 lv_timer_handler () at /.../lvgl/src/misc/lv_timer.c:105
#32 0x000055555557d9db in lv_timer_handler () at /.../lvgl/src/misc/lv_timer.c:77
#33 0x00005555555584e5 in main () at /.../sdl/main.c:27
```

</details>

<details>
<summary>the same but with LV_OS_NONE</summary>

```gdb
Thread 1 "demo" received signal SIGINT, Interrupt.                                                                                                                                                                                             
0x000055555556cf8a in lv_draw_dispatch_wait_for_request () at /.../lvgl/src/draw/lv_draw.c:270                                                                                                          
[... cut ...]
- `lv_draw_dispatch_layer` **first calls `lv_draw_dispatch_layer_request` to acquire the draw lock** (` _draw_info.sync` or `_draw_info.dispatch_req`), and then calls `dispatch_cb`, which is lv_draw_sdl.c's `dispatch`
- `dispatch` calls `execute_drawing`,
- which calls `draw_from_cached_texture`, 
- which calls `lv_cache_acquire_or_create`
- which calls `cache->ops.create_cb` which is `sdl_texture_cache_create_cb`
- `sdl_texture_cache_create_cb` is a thin wrapper around `draw_to_texture`
- **which calls `lv_draw_dispatch_wait_for_request` again, and we get a deadlock**

C47D, 2024-02-01:
@kisvegabor Who can help us with this issue?

kisvegabor, 2024-02-01:
Hi,

I can fix it tomrrow.

closed 2024-02-02 by kisvegabor

neuschaefer, 2024-02-03:
Thanks, much appreciated!

PR #5578 (2024-02-05): fix(sdl): use the new draw buffer structure in the layers
### Description of the feature or fix

fixes #5533

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig) [... cut]