lvgl/lvgl #5536: Arduino example in 9.0 not working on RP2040 or ESP32

kb1ujs, 2024-01-31:
### LVGL version

9.0.0

### What happened?

I have been trying to run the LVGL_Arduino sketch included with the library install of LVGL on both the Raspberry Pi Pico and an ESP32.  I'm using an ILI9341 display with both microcontrollers.  I have the TFT_eSPI library installed with tested working configurations using TFT_eSPI examples on both microcontrollers.  I have LV_COLOR_DEPTH set to 16 and LV_USE_TFT_ESPI set to 1 in my lv.conf.  When I compile the script I get no errors, but I also get nothing on the screen of either microcontroller when I upload the code.  It appears the display is initialized, but nothing ever gets drawn.  I also lose the serial port on my Pico and have to power off and use the BOOTSEL button to reprogram.  I added some code to flash the onboard LED on my Pico as each stage of the script is completed and found that lv_timer_handler never returns after being called the first time through the loop() function.  

### How to reproduce?

Compile the LVGL_Arduino sketch as provided with the LVGL 9.0 library on either Pico

kisvegabor, 2024-01-31:
There was a fix related to Arduino. Please use the latest version from [here](https://github.com/lvgl/lvgl/blob/master/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino).

kb1ujs, 2024-01-31:
Thank you, that solved the problem.

I assume this corrected example will be included in a future Arduino library release?

closed 2024-01-31 by kb1ujs

kisvegabor, 2024-02-01:
Yes, sure