lvgl/lvgl #5537: Image scaling doesn't work for RGB565A8 format

sirius506, 2024-01-31:
### LVGL version

v9.0.0

### What happened?

Music player demo uses ARGB8888 format album cover image. I have generated RGB585A8 format cover image file and rebuilt the demo. Now, cover image crashed when it is zoomed.

### How to reproduce?

```
cd demos/music/assets
LVGLImage.py --ofmt C --cf RGB565A8 png/272_png/cover_1.png
mv output/cover_1.c ../img_lv_demo_music_cover1.c
```

kisvegabor, 2024-02-01:
I can reproduce it and will look into it next week.

kisvegabor, 2024-02-16:
Sorry, I'll get back to this ASAP!

kisvegabor, 2024-02-24:
Now it works on my end with the latest LVGL from master.

```c
LV_IMAGE_DECLARE(cover_111);
lv_obj_t * img1 = lv_image_create(lv_screen_active());
lv_image_set_src(img1, &cover_111);
lv_obj_align(img1, LV_ALIGN_CENTER, 0, 0);
lv_image_set_scale(img1, 400);
```
 
[image]

[cover111.c.zip](https://github.com/lvgl/lvgl/files/14393986/cover111.c.zip)

sirius506, 2024-02-25:
It looks image scaling itself is working, but music demo does't.

I am testing music demo with lv_port_stm32f769_disco. With latest master and RGB565A8 image, I got this result.  (Sorry, my screen is dirty, please ignore black dashes on below screen part.)

[image]

Cover image still goes wrong when scaled and play button is no longer displayed correctly.  With v9.0.0 release code, play button has shown correctly.

sirius506, 2024-03-04:
OK, I confirmed #5782 fixed the image crash problem.
Now, demo screen looks like this.

[image]

The cover image is shown as square figure. I checked PNG image in assets and its background alpha value is 0xff. So, the fix sets correct alpha value.

To get same cover image effect as we saw in older releases, I think PNG cover images should be fixed to have transparent background. But, I wonder why and how it has been working correctly before...

PR #5782 (2024-03-07): fix(draw): fix non antialiased rgb565a8 transformation
fixes #5537

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and u [... cut]
closed 2024-03-07 by XuNeo