lvgl/lvgl #5539: lv_ffmpeg_player_set_src does not set image stride

josepho0918, 2024-01-31:
### LVGL version

v9.0.0

### What happened?

```
    player->imgdsc.header.w = width;
    player->imgdsc.header.h = height;
    player->imgdsc.data_size = data_size;
    player->imgdsc.header.cf = has_alpha ? LV_COLOR_FORMAT_ARGB8888 : LV_COLOR_FORMAT_NATIVE;
    player->imgdsc.data = ffmpeg_get_image_data(player->ffmpeg_ctx);
```
The variable `player->imgdsc.header.stride` is not set in the `lv_ffmpeg_player_set_src` function.

kisvegabor, 2024-02-01:
Hi,

Could test it by setting the stride to 
`player->imgdsc.header.stride = width * lv_color_format_get_size(player->imgdsc.header.cf);`
?

josepho0918, 2024-02-02:
Hi @kisvegabor,

It works great!🙂

kisvegabor, 2024-02-02:
Ah great! Could you send a Pull request with the fix?

josepho0918, 2024-02-03:
Sent.

PR #5580 (2024-02-04): fix(ffmpeg): add missing stride setting to ffmpeg image
### Description of the feature or fix

fixes #5539 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig). [... cut]
closed 2024-02-03 by josepho0918