lvgl/lvgl #5542: v8.3 unused DMA STM32 functions

faxe1008, 2024-01-31:
### Introduce the problem

There is a pending PR on our lvgl module which tries to improve the STM32 performance of LVGL. Everything works but CI is failing because there are 4 unused functions. 

```
_lv_gpu_stm32_dma2d_invalidate_cache
_lv_draw_stm32_dma2d_copy_buffer
lv_gpu_stm32_dma2d_wait_cb
lv_draw_stm32_dma2d_img
```

### Proposal

Either removal of the functions or introduction of an `LV_ATTRIBUTE_MAYBE_UNUSED`.

CharlesDias, 2024-01-31:
Dear, this is the comparison result.

https://github.com/lvgl/lvgl/assets/27859307/a08147a6-ef36-45ca-8bf2-5beab04dfa46

kisvegabor, 2024-02-01:
Hi,

@faxe1008 I've just opened #5561

@CharlesDias Very impressive. I hope we will have ChromeArt and NeoChrome in LVGL v9 too soon.

kisvegabor, 2024-02-10:
The PR is merged, I close this issue.

PR #5561 (2024-02-07): fix(st-dma2d): remove unused functions
### Description of the feature or fix

fixes #5542

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/co [... cut]
closed 2024-02-10 by kisvegabor