lvgl/lvgl #5544: LV_USE_OBJ_ID / LV_USE_OBJ_ID_BUILTIN problem

PercentageCorporation, 2024-01-31:
### LVGL version

v9.0.0

### What happened?

I set the following in lv_conf.h which causes build problems in lv_obj.c and lv_obj_tree.c

/* Add `id` field to `lv_obj_t` */
#define LV_USE_OBJ_ID           1
/* Use lvgl builtin method for obj ID */
#define LV_USE_OBJ_ID_BUILTIN   0

The #if LV_USE_OBJ_ID statements in these files should be #if LV_USE_OBJ_ID_BUILTIN, otherwise undefined references occur in the build.

kisvegabor, 2024-02-01:
Hi,

Thank you for the report. 

Could you send a PR?

XuNeo, 2024-02-01:
Disable `LV_USE_OBJ_ID_BUILTIN ` means using custom ID generator. 
Below APIs needed to be implemented and linked to lvgl.
```c
    void lv_obj_assign_id(const lv_obj_class_t * class_p, lv_obj_t * obj);
    void lv_obj_free_id(lv_obj_t * obj);
    const char * lv_obj_stringify_id(lv_obj_t * obj, char * buf, uint32_t len);

```
See https://docs.lvgl.io/master/others/obj_id.html#use-custom-id-generator

closed 2024-02-24 by lvgl-bot