lvgl/lvgl #5546: Rendering with offset broken with fbdev driver

Johennes, 2024-01-31:
### LVGL version

master

### What happened?

I'm updating [buffyboard](https://gitlab.com/postmarketOS/buffybox/-/tree/master/buffyboard?ref_type=heads) to the current tip of `master` and I believe the rendering to a display subsection using offsets has broken.

### How to reproduce?

Here's a minimal example where on my laptop screen (1920x1080), I'm trying to push the display down by 100px.

```
    /* Create display */
    lv_display_t *disp = lv_linux_fbdev_create();
    lv_linux_fbdev_set_file(disp, "/dev/fb0");

    /* Set the physical resolution */
    int32_t hor_res_phys = lv_display_get_horizontal_resolution(disp);
    int32_t ver_res_phys = lv_display_get_vertical_resolution(disp);
    lv_display_set_physical_resolution(disp, hor_res_phys, ver_res_phys);

    /* Cut off the first 100px of the display and only render in the remaining space */
    lv_display_set_resolution(disp, hor_res_phys, 980);
    lv_display_set_offset(disp, 0, 100);

    /* Add keyboard */
    keyboard = lv_keyboard_create(lv_scr_act());
    lv_obj_set_pos(keyboard, 0, 0);
    lv_obj_set_size(keyboard, LV_HOR_RES, LV_VER_RES);
```
This cuts off the first 100px of the keyboard which seems to be rendered at (0,0) in the physical coordinate system instead of the one shifted by the display offset.

[image]

When I try to move the keyboard itself down by 100px via `lv_obj_set_pos(keyboard, 0, 0);`, the keyboard is rendered at the right y but now the lower 100px are cut off.

[image]

What I would expect and what used to work in 8.3 is the keyboard rendering over the entire area below the 100px cut-off at the top.

I have a hunch the issue might be in the new fbdev driver because when I tried to copy the old code in, it rendered all garbage but at least the garbage filled the correct coordinates. I haven't been able to figure out the root cause in the pointer arithmetic though.

kisvegabor, 2024-02-01:
I couldn't try it out yet, but probably the offset needs to be considered [here](https://github.com/lvgl/lvgl/blob/master/src/drivers/display/fb/lv_linux_fbdev.c#L265).

Johennes, 2024-02-01:
Ok, so for the specific issue of vertical offset, the rendering works as expected when I subtract `disp->offset_y` from `color_pos` and re-add it to `fb_pos`.
 
```
--- a/src/drivers/display/fb/lv_linux_fbdev.c
+++ b/src/drivers/display/fb/lv_linux_fbdev.c
...
     int32_t w = lv_area_get_width(area);
     uint32_t px_size = lv_color_format_get_size(lv_display_get_color_format(disp));
-    uint32_t color_pos = (area->x1 + dsc->vinfo.xoffset) * px_size + area->y1 * dsc->finfo.line_length;
-    uint32_t fb_pos = color_pos + dsc->vinfo.yoffset * dsc->finfo.line_length;
+    uint32_t color_pos = (area->x1 + dsc->vinfo.xoffset) * px_size + area->y1 * dsc->finfo.line_length - disp->offset_y * dsc->finfo.line_length;
+    uint32_t fb_pos = color_pos + dsc->vinfo.yoffset * dsc->finfo.line_length + disp->offset_y * dsc->finfo.line_length;
 
     uint8_t * fbp = (uint8_t *)dsc->fbp;
     int32_t y;
```

However, what slightly confuses me is that I think this wasn't necessary in the old fbdev driver. If I'm not mistaken, this offset handling would also have to be applied in every other display driver, too. I'm wondering if this could be solved at the higher level before `flush_cb` is called? Possibly by shifting `color_p` according to the display's offset?

Johennes, 2024-02-01:
Something like this where the rect remains translated by `offset_x` / `offset_y` for the driver to render into but the `px_map` pointer is translated _back_ by `offset_x` / `offset_y` so that the pixels are read from the untranslated location.

```
--- a/src/core/lv_refr.c
+++ b/src/core/lv_refr.c
static void call_flush_cb(lv_display_t * disp, const lv_area_t * area, uint8_t * px_map)
{
    LV_PROFILER_BEGIN;
    printf("Calling flush_cb on (%d;%d)(%d;%d) area with %p image pointer\n",
                  (int)area->x1, (int)area->y1, (int)area->x2, (int)area->y2, (void *)px_map);

    lv_area_t offset_area = {
        .x1 = area->x1 + disp->offset_x,
        .y1 = area->y1 + disp->offset_y,
        .x2 = area->x2 + disp->offset_x,
        .y2 = area->y2 + disp->offset_y
    };
 
     lv_display_send_event(disp, LV_EVENT_FLUSH_START, &offset_area);
-    disp->flush_cb(disp, &offset_area, px_map);
+    uint32_t px_size = lv_color_format_get_size(lv_display_get_color_format(disp));
+    disp->flush_cb(disp, &offset_area, px_map - disp->offset_x * px_size - disp->offset_y * disp->hor_res * px_size);
     lv_display_send_event(disp, LV_EVENT_FLUSH_FINISH, &offset_area);
 
     LV_PROFILER_END;
```

I'm not sure if this is safe though?

kisvegabor, 2024-02-07:
I'm sorry for the delay. 

It seems in v8 there weren't any extra calculations either. See [here](https://github.com/lvgl/lvgl/blob/release/v8.3/src/core/lv_refr.c#L1315-L1328).

I'm trying to understand what the root of the issue is.

Do you use the driver in PARTIAL or DIRECT mode? (`LV_LINUX_FBDEV_RENDER_MODE`)

I think the current "translate the area by the offset" should work in PARTIAL mode naturally. The driver just needs to place the rendered image to different area. However in DIRECT mode, where we assume that the draw buffer is always screen sized an offset can be really confusing. It's the same in FULL mode.

Johennes, 2024-02-11:
Argh, indeed I was using direct rendering mode due to a copy/paste error? :facepalm: 

It works as expected in partial rendering mode. It would be nice if somehow we could also make this work in direct rendering mode but that's a different problem and partial rendering works fine for my current use case.

Thanks for the help!

closed 2024-02-11 by Johennes

kisvegabor, 2024-02-12:
Ahhh, okay  :smile: 

Let's keep this direct mode with offset issue in mind and see if it pop up for other too.