lvgl/lvgl #5547: Enabled system monitor results compilation error in MicroPython unix build

PGNetHun, 2024-01-31:
### LVGL version

v9.0.0

### What happened?

When system monitor is enabled in MicroPython binding, then the unix port build fails with following error:
```
CC build-standard/lvgl/lv_mpy.c
build-standard/lvgl/lv_mpy.c:21405:22: error: ‘lv_sysmon_class’ undeclared here (not in a function); did you mean ‘lv_sysmon_create’?
21405 |     .lv_obj_class = &lv_sysmon_class,
      |                      ^~~~~~~~~~~~~~~
      |                      lv_sysmon_create
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [../../py/mkrules.mk:83: build-standard/build-standard/lvgl/lv_mpy.o] Error 1
```

There was a refactor of system monitor, and since then there is no `lv_sysmon_class`: https://github.com/lvgl/lvgl/commit/a61d87f8403d7be17f238640b520ca0fd96c5c57

### How to reproduce?

1. Clone [lv_micropython](https://github.com/lvgl/lv_micropython) repository (`master` branch)

2. Enable system monitor in `lv_conf.h` file: 
```
#define LV_USE_SYSMON 1
#define LV_USE_PERF_MONITOR 1
```

3. Build unix port in `lv_micropython` repo directory:
```
make -C mpy-cross
cd ports/unix
make submodules
make
```

kisvegabor, 2024-02-01:
I wonder why the binding thinks that there is lv_sysmon_class? Just because lv_sysmon_create() looks like any widget create function?

PGNetHun, 2024-02-16:
Not stale, did not investigated yet.

kdschlosser, 2024-03-18:
It because of how the dynamics of the gen_mpy script work. It looks for functions that end with `_create` and the first parameter has a type of `lv_obj_t` and it makes an assumption that it's a widget.

kdschlosser, 2024-03-18:
so 2 choices to fix it. Make sysmon a widget or rename the function.

kisvegabor, 2024-03-18:
What distinguishes sysmon from widget from MP's point of view? Is it having an `lv_sysmon_class`?

kisvegabor, 2024-03-18:
Now I see it in #5893.

The third option is to make `lv_sysmon_create` `static`. It's very unlikely that anyone ever wants to use `lv_sysmon` on its own.

We could make it based on `lv_label` but then we should move it to the `widgets` folder which doesn't sound appealing.

kdschlosser, 2024-03-18:
I personally would just rename the function, if it looks like a widget it can be mistaken for one correct? It is not a widget and there for it should not carry the same API as a widget.

kdschlosser, 2024-03-18:
this line also needs to be removed from lv_sysmon.c

```
#define MY_CLASS (&lv_sysmon_class)

```

kdschlosser, 2024-03-18:
@kisvegabor 

You could rename it to `lv_start_sysmon` or `lv_show_sysmon`. The latter would be better if the sysmon is able to run without having to be shown and adding a function `lv_hide_sysmon` this way it being displayed can be turned on and off real time. so if a developer wanted to make into the firmware having ti available as a debugging tool it wouldn't be necessary to flash the firmware to turn it on or off. 

Just a suggestion...

kisvegabor, 2024-03-21:
`lv_sysmon_show` sounds good to me!

kdschlosser, 2024-03-22:
renaming the function should solve the problem even when sysmon is turned on.

kisvegabor, 2024-04-08:
Continue in #5936.

closed 2024-04-08 by kisvegabor