lvgl/lvgl #5555: Has LVGL V9 decided to no longer support STM32 DMA2D?

Tyroo, 2024-02-01:
### Introduce the problem

I couldn't find any implementation related to STM32 DMA2D GPU in LVGL V9. Do I plan to not support it in future versions?

karlp, 2024-02-08:
You didn't think this might be maybe worth mentioning in the release notes _at least_?

trollcop, 2024-02-08:
Yeah what, as someone who used (and contributed patches) to DMA2D code I find this pretty offensive. I guess I won't be recommending lvgl to my next customer.

kisvegabor, 2024-02-10:
I'm sorry for causing this bad feelings. Although it was mentioned in several issues, you are right, a more transparent communication on a more central place would have been more appropriate. I've added it in #5632

Besides that I would like to ensure you that your contribution is not lost:
1. It's available in v8
2. It will be a very good starting point when we add DMA2D support to v9

closed 2024-02-11 by FASTSHIFT

dolence, 2024-02-19:
Why to remove something that was previously working? ST official support would be nice, but I just discovered the project I was migrating from 8.3 to 9.0 will not work as expected because it will be hogging resources unnecessarily as I'm using a big display. This makes no sense.

trollcop, 2024-02-19:
> Why to remove something that was previously working? ST official support would be nice, but I just discovered the project I was migrating from 8.3 to 9.0 will not work as expected because it will be hogging resources unnecessarily as I'm using a big display. This makes no sense.

Better not use projects that dictate what features are present or not with money

dolence, 2024-02-19:
Unfortunately the RTOS in use doesn't support TouchGFX yet. I will stick to 8.3 by now but I'm looking for alternatives.

kisvegabor, 2024-02-19:
> Better not use projects that dictate what features are present or not with money.

Yes, it's a trade off. LVGL has grown so big that it requires some kind of revenue. I don't want to change the license or have any kind of  licencing/subscriptions. Instead, I'm making agreements with the vendors so you can get LVGL for free in the future too.

I also miss ST a lot. I know that they are an important player. I hope that their management realizes the value of LVGL and the size of our community and we can proceed with the cooperation soon.

dolence, 2024-02-19:
This is awful because you are spoiling a portion of the users to achieve some sort of gain with a big company. Isn't the same as if we never had this working. Even worst, community contribution is being taken away from the users to be used to bargain with huge companies. Can't you see why it is wrong? I think it's question of time until someone skilled enough fork this and restore that. At least is what I hope for.

XuNeo, 2024-02-19:
The current challenge LVGL v9 encounters with ST is due to LVGL's extensive optimization for GPU/Hardware accelerators, which involves the introduction of draw units. LVGL v9 has undergone a significant evolution to reach its present state, but there is a current shortage of community contributors to adapt ST to this latest architecture.

However, it's incorrect to say that our collective community contributions have been lost, as v8 remains available to all and continues to be utilized in our internal projects. We remain hopeful for official support from ST in the near future.

kisvegabor, 2024-02-19:
Migrating v8's DMA2D to v9 is not trivial. Besides in v9 not only ChromArt but NeoChrom should be supported in future.

LuisCRSousa, 2024-04-02:
It's just me or is this offensive to all lvgl contributors? So many devs that helped refine STM32 DMA2D and now you guys are taking advantage of this to your own reward? And removing this feature in v9 until stm pays you?
This looks really bad

kisvegabor, 2024-04-05:
There are two things here:
1. Removing code is a normal part of the development process. Many widgets, or features were just removed in past years for various reasons, e.g. because we found a better alternative, or they just don't fit to a new architecture anymore. The DMA2D implementation of LVGL v8 is not compatible with LVGL v9, and it significant effort were required to make it compatible with v9. 
2. Although I don't remember anyone volunteered to port the ST driver to v9, if someone did that, I needed to say no. Yes, it's really bad. I don't like it either to make tech compromises due to business reasons. However, as a company we do need income. Honestly, I had more money if I were working as software engineer for a multi national company. So it's not about that I would like to buy my fifth Ferrari, but just to make a free project sustainable and well supported. 

I understand your frustration. You have worked on it, and ST is a big and popular player in this market. Just like you, I really want this matter sorted out as soon as possible. Next week is Embedded World, where I hope we can push this case a little.

LuisCRSousa, 2024-04-05:
I completely understand your point of view, although I believe this could have been processed differently.

DMA2D is a big feature, if it was to be removed it could have happened in a more transparent way. It was removed without alerting anyone of it, it was even missing from the release notes.

Removing code is a normal part of the development process yes, but it wasn't removing it, it was archiving it temporarily until STM contributes $. V9 could have been delayed until the partnership was done. DMA2D was initially made without a partnership to begin with.

I'm not against STM supporting the project, exactly the opposite, I want the project to have a future and all of your work is to be appraised, you deserve it.

kisvegabor, 2024-04-06:
> DMA2D is a big feature, if it was to be removed it could have happened in a more transparent way.

You are perfectly right. Being more upfront and transparent would have been better. I've learnt the lesson and will act accordingly next time.

woniulive, 2024-07-31:
@kisvegabor 
I noticed that in https://lvgl.io/, ST is Official partners now. So it there any plan for DMA2D?

kisvegabor, 2024-08-01:
Yes, there is, @liamHowatt will work on it. it will be added probably next week.

woniulive, 2024-08-02:
@kisvegabor 
Hi, Do you know whether I can use DMA2D with SPI interface?
By the way, when will the version include DMA2D release?

kisvegabor, 2024-08-08:
Yes, you can use DMA2D with an SPI based display. DMA2D just renders into a buffer and it can be sent via SPI to the display.

Liam is already working on it. Probably it will require a couple of days.

LuisCRSousa, 2024-08-27:
Was DMA2D implemented in 9.2? Nothing on the changelog about that

liamHowatt, 2024-08-27:
Unfortunately I couldn't have it ready in time. Some reviewing and testing will be done and then it can be merged and backported to the v9.2 release branch.

#6691

KlemenDEV, 2024-10-03:
Is NeoChrom support eventually planned too?

liamHowatt, 2024-10-03:
A PR for it will open tomorrow :slightly_smiling_face:

KlemenDEV, 2024-10-03:
That is awesome! :D

kisvegabor, 2024-10-03:
According to our initial tests its 1.6 times faster than software rendering. 😎

liamHowatt, 2024-10-04:
PR open: https://github.com/lvgl/lvgl/pull/7002
Riverdi STM32U5 port PR: https://github.com/lvgl/lv_port_riverdi_stm32u5/pull/1

Amrith-H-N, 2025-01-31:
so does this mean dma2d is working ? which version should I use ? I am using stm32h7b3i-dk (pretty old)

PR #5632 (2024-02-11): docs(changelog): mention more features removed in v9
### Description of the feature or fix

fixes #5555, #5589

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfi [... cut]