lvgl/lvgl #5557: Performance Monitor: Inconsistency in measuring flush time

ks75vl, 2024-02-01:
### LVGL version

v9.0.0

### What happened?

While using LVGL with the ST7789 driver to create a display, there's an inconsistency in measuring flush time.

The flush time is currently measured between `LV_EVENT_FLUSH_START` and `LV_EVENT_FLUSH_FINISH` events. However, on devices where `DMA` is utilized for parallel data transfer to the display, this measurement becomes inaccurate. Although the total render and flush time remains correct, the flush time itself is misrepresented due to parallel processing. In my tests, the flush time merely accounts for the time taken to enqueue data to the DMA queue, typically around 2ms, which is trivial.

Implementing a `sync` flush in `flush_cb` or `send_color_cb` can correct the measurement but disrupts the purpose of DMA.

kisvegabor, 2024-02-01:
I've just opened a related Pull request: https://github.com/lvgl/lvgl/pull/5558

ks75vl, 2024-02-02:
In my case, the measured data look weird, I suspect that the measurement of `flush_not_in_render` is being overlapped by `flush_in_render` due to the refresh period too fast (10ms). Essentially, when `LV_EVENT_FLUSH_WAIT_FINISH` is emitted, a new render has already completed and is waiting the release of the draw buffer.
This inconsistency results in a mismatch between FPS and frame time (in my test showing around 59 PFS at 10ms/frame).

```js
render_elaps_sum=303, flush_in_render_elaps_sum=179, flush_not_in_render_elaps_sum=0
render_elaps_sum=301, flush_in_render_elaps_sum=178, flush_not_in_render_elaps_sum=0
render_elaps_sum=301, flush_in_render_elaps_sum=178, flush_not_in_render_elaps_sum=0
render_elaps_sum=302, flush_in_render_elaps_sum=179, flush_not_in_render_elaps_sum=0
render_elaps_sum=301, flush_in_render_elaps_sum=178, flush_not_in_render_elaps_sum=0
render_elaps_sum=302, flush_in_render_elaps_sum=179, flush_not_in_render_elaps_sum=0
render_elaps_sum=301, flush_in_render_elaps_sum=178, flush_not_in_render_elaps_sum=0
render_elaps_sum=302, flush_in_render_elaps_sum=179, flush_not_in_render_elaps_sum=0
render_elaps_sum=302, flush_in_render_elaps_sum=177, flush_not_in_render_elaps_sum=0
render_elaps_sum=302, flush_in_render_elaps_sum=178, flush_not_in_render_elaps_sum=0
render_elaps_sum=302, flush_in_render_elaps_sum=179, flush_not_in_render_elaps_sum=0
render_elaps_sum=302, flush_in_render_elaps_sum=177, flush_not_in_render_elaps_sum=0
render_elaps_sum=301, flush_in_render_elaps_sum=178, flush_not_in_render_elaps_sum=0
render_elaps_sum=300, flush_in_render_elaps_sum=176, flush_not_in_render_elaps_sum=0
render_elaps_sum=303, flush_in_render_elaps_sum=179, flush_not_in_render_elaps_sum=0
render_elaps_sum=301, flush_in_render_elaps_sum=178, flush_not_in_render_elaps_sum=0
render_elaps_sum=303, flush_in_render_elaps_sum=179, flush_not_in_render_elaps_sum=0
```

kisvegabor, 2024-02-02:
Thank you for the feedback! I've updated #5558. Could you test it again?

ks75vl, 2024-02-05:
I have just tested on my device, the frame time is now accurate.

https://github.com/lvgl/lvgl/assets/22657508/da8bb2c5-a322-4f08-ade9-d2a0b32b1f6b

closed 2024-02-06 by kisvegabor