lvgl/lvgl #5564: `lv_tileview_set_tile_by_index` doesn't work after resizing the screen

neuschaefer, 2024-02-01:
### LVGL version

v9.0.0-80-g56077516a

### What happened?

I create an `lv_tileview` and switch tiles by index. This works as long as I don't resize the screen, but when I resize the screen, it fails.

The problem becomes apparent in `lv_tileview_set_tile_by_index`:

```c
void lv_tileview_set_tile_by_index(lv_obj_t * tv, uint32_t col_id, uint32_t row_id, lv_anim_enable_t anim_en)
{
    lv_obj_update_layout(tv);

    int32_t w = lv_obj_get_content_width(tv);   // <-- This finds the new size
    int32_t h = lv_obj_get_content_height(tv);

    int32_t tx = col_id * w;
    int32_t ty = row_id * h;

    uint32_t i;
    for(i = 0; i < lv_obj_get_child_count(tv); i++) {
        lv_obj_t * tile_obj = lv_obj_get_child(tv, i);
        int32_t x = lv_obj_get_x(tile_obj);     // ... but these coordinates aren't updated
        int32_t y = lv_obj_get_y(tile_obj);     // so they use use original screen resolution
        if(x == tx && y == ty) {
            lv_tileview_set_tile(tv, tile_obj, anim_en);
            return;
        }
    }

    LV_LOG_WARN("No tile found with at (%d,%d) index", (int)col_id, (int)row_id);
}
```

### How to reproduce?

Example program:

```c
#include <lvgl.h>
#include <unistd.h>

void timer_cb(lv_timer_t *timer)
{
        static int index = 0;

        lv_obj_t *tv = timer->user_data;
        lv_tileview_set_tile_by_index(tv, index, 0, LV_ANIM_ON);

        index ^= 1;
}

int main()
{
        lv_init();
        lv_display_t *display = lv_sdl_window_create(800, 600);

        lv_group_t *g = lv_group_create();
        lv_group_set_default(g);
        lv_indev_t *mouse = lv_sdl_mouse_create();
        lv_indev_set_group(mouse, g);

        lv_obj_t *tv = lv_tileview_create(lv_screen_active());

        lv_obj_t *tile0 = lv_tileview_add_tile(tv, 0, 0, LV_DIR_RIGHT);
        lv_obj_t *label0 = lv_label_create(tile0);
        lv_label_set_text(label0, "tile 0");

        lv_obj_t *tile1 = lv_tileview_add_tile(tv, 1, 0, LV_DIR_LEFT | LV_DIR_RIGHT);
        lv_obj_t *label1 = lv_label_create(tile1);
        lv_label_set_text(label1, "tile 1");

        lv_timer_t *timer = lv_timer_create(timer_cb, 1000, tv);

        while (true) {
                uint32_t ms_delay = lv_timer_handler();
                usleep(ms_delay * 1000);
        }
}
```

kisvegabor, 2024-02-02:
Hi,

I've found a simple solution. Please try out #5577.

neuschaefer, 2024-02-04:
Thanks, #5577 works for me.

PR #5577 (2024-02-04): fix(tileview): on size change auto. update the pos. of the tiles
fixes #5564

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen [... cut]
closed 2024-02-04 by FASTSHIFT