lvgl/lvgl #5581: fix variable name in v9.0.0

huenrong, 2024-02-03:
### LVGL version

v9.0.0

### What happened?

There are the same variable names in file `lvgl/examples/assets/emoji/img_emoji_F617.c` and file `lvgl/tests/src/test_assets/test_img_emoji_F617.c`

- file `lvgl/examples/assets/emoji/img_emoji_F617.c`
```c
const LV_ATTRIBUTE_MEM_ALIGN LV_ATTRIBUTE_IMAGE_1F617 uint8_t emoji_F617_map[] = {
...
const lv_image_dsc_t emoji_F617 = {
    .header.cf = LV_COLOR_FORMAT_NATIVE,
    .header.w = 72,
    .header.h = 72,
    .data_size = 5184 * LV_COLOR_DEPTH / 8,
    .data = emoji_F617_map,
};
```
- file `lvgl/tests/src/test_assets/test_img_emoji_F617.c`
```c
const LV_ATTRIBUTE_MEM_ALIGN LV_ATTRIBUTE_IMAGE_1F617 uint8_t emoji_F617_map[] = {
...
const lv_image_dsc_t emoji_F617 = {
    .header.cf = LV_COLOR_FORMAT_NATIVE,
    .header.w = 72,
    .header.h = 72,
    .data_size = 5184 * LV_COLOR_DEPTH / 8,
    .data = emoji_F617_map,
};
```

but In file `lvgl/tests/src/test_assets/test_img_emoji_F617.c` there should be prefix `test_`, such as `test_emoji_F617_map`, `test_emoji_F617`

C47D, 2024-02-04:
Hi, 

Is #5559 fix included in your project?

huenrong, 2024-02-05:
> Hi,
> 
> Is #5559 fix included in your project?

@C47D 
thanks, i just used master branch, it`s ok.
But I think should modify this variable name, because if I enable `LV_BUILD_EXAMPLES`, an error will also be reported.

kisvegabor, 2024-02-05:
Sorry, my bad. There was a typo in #5559

It should with #5594

huenrong, 2024-02-05:
> Sorry, my bad. There was a typo in #5559
> 
> It should with #5594

@kisvegabor 
thanks, If this is the case, then the contents of the files in this directory are somewhat inconsistent.
some files use `LV_BUILD_EXAMPLES` macro(eg: `tests/src/test_assets/test_animimg001.c`)
some files use `test_xx` name(eg: `src/test_assets/test_img_cogwheel_rgb565.c`)
some files use `LV_BUILD_TEST` macro(eg: `tests/src/test_assets/test_img_emoji_F617.c`)

It seems that the rules are not uniform, maybe I didn’t understand it correctly in some places.

closed 2024-02-05 by kisvegabor

kisvegabor, 2024-02-07:
Oh, really there is a bigger mess here than I thought :sweat_smile:

I unified the guard in #5616. I'll update the filenames too.

PR #5594 (2024-02-05): fix(test): fix typo related to #5559
### Description of the feature or fix

In test files we should use `LV_BUILD_TEST` and not `LV_BUILD_EXAMPLES`.

fixes #5581

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconf [... cut]
PR #5616 (2024-02-12): ci(assets): add LV_BUILD_TEST guard
### Description of the feature or fix

@XuNeo The images in the `test_images` folder doesn't have the guards. I think it would be uses full to add a `--guard "FOO == 3"` parameter to the image convert. What do you think?

fixes #5581

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_ [... cut]