lvgl/lvgl #5582: V9 two minor issues

DC2047, 2024-02-04:
### LVGL version

V9.0

### What happened?

Problems found in V9:
1, lv_anim_path_bounce The result returned is abnormal.
2, lv_obj_set_style_arc_image_src(arc, &img_xx, LV_PART_MAIN);
lv_obj_set_style_arc_image_src(arc, &img_xxx, LV_PART_INDICATOR);
ARC can't add a background image.

XuNeo, 2024-02-04:
Could you post the minimal reproducible code? I recently have used arc_image_src, and it does work. It may be related to the image is too small.

DC2047, 2024-02-04:
lv_obj_t *arc = lv_arc_create(cont);
    // lv_obj_set_style_bg_opa(arc, LV_OPA_TRANSP, LV_PART_MAIN);
    lv_obj_set_style_pad_all(arc, 4, LV_PART_MAIN);

    lv_obj_set_style_arc_image_src(arc, &img_circle_dark, LV_PART_MAIN);
    lv_obj_set_style_arc_width(arc, 16, LV_PART_MAIN);

    lv_obj_set_style_arc_image_src(arc, &img_circle, LV_PART_INDICATOR);
    lv_obj_set_style_arc_width(arc, 16, LV_PART_INDICATOR);
    lv_obj_set_style_arc_color(arc, theme->home.out.text2, LV_PART_INDICATOR);
    // lv_obj_set_style_arc_opa(arc, LV_OPA_COVER, LV_PART_INDICATOR);
    lv_obj_set_size(arc, 62 + 6, 62 + 6);
    lv_arc_set_rotation(arc, 180);
    lv_arc_set_bg_angles(arc, 0, 180);
    lv_arc_set_range(arc, 0, 1000);
    lv_arc_set_value(arc, 0);

The above is the code of ARC,
In V8.3 it can be displayed normally.
Thank you!

kisvegabor, 2024-02-04:
Please also `img_circle_dark` as well.

DC2047, 2024-02-04:
> 也请附上。`img_circle_dark`

 
[image]

const lv_image_dsc_t img_circle_dark = {
  .header.cf = LV_COLOR_FORMAT_RGB565A8,
  .header.magic = LV_IMAGE_HEADER_MAGIC,
  .header.w = 63,
  .header.h = 63,
  .data_size = 3969 * 3,
  .data = img_circle_dark_map,
};

Thank you!

XuNeo, 2024-02-04:
There are two issues from lvgl's side. You can try image format LV_COLOR_FORMAT_ARGB8888 instead.

1. bpp of RGB565A8 should be 0 or 16, not 24. It's special format and take care of it explicitly.
2. lv_draw_sw_arc.c needs to take A8 as a mask into consideration. @kisvegabor Could you help on this?

kisvegabor, 2024-02-07:
> lv_draw_sw_arc.c needs to take A8 as a mask into consideration. @kisvegabor Could you help on this?

How do you mean it? 
Using the above image I get this. So if there is an image, it completely replaces the background.
[image]

XuNeo, 2024-02-07:
> How do you mean it?

I mean using RGB565A8 image, it's not working. Is the screenshot you posted is with RGB565A8?

DC2047, 2024-02-07:
My display content：
[image]

```

lv_obj_t* arc = lv_arc_create(lv_screen_active());
//lv_obj_remove_style_all(arc);
lv_obj_set_style_pad_all(arc, 2, LV_PART_MAIN);
lv_obj_set_style_radius(arc, LV_RADIUS_CIRCLE, LV_PART_MAIN);
lv_obj_set_style_border_color(arc, lv_color_hex(0x222222), LV_PART_MAIN); // 0x222222 0x008A00
lv_obj_set_style_border_opa(arc, LV_OPA_COVER, LV_PART_MAIN);
lv_obj_set_style_border_width(arc, 1, LV_PART_MAIN);

lv_obj_set_style_arc_image_src(arc, &img_circle_dark, LV_PART_MAIN);
lv_obj_set_style_arc_width(arc, 5, LV_PART_MAIN);

lv_obj_set_style_arc_image_src(arc, &img_circle, LV_PART_INDICATOR);
lv_obj_set_style_arc_width(arc, 5, LV_PART_INDICATOR);
// lv_obj_set_style_arc_color(arc, theme->home.out.text2, LV_PART_INDICATOR);
lv_obj_set_style_arc_opa(arc, LV_OPA_COVER, LV_PART_INDICATOR);
lv_obj_set_size(arc, 62 + 6, 62 + 6); // 多加是 边框 + 间隙
lv_arc_set_rotation(arc, 180);
lv_arc_set_bg_angles(arc, 0, 180);
lv_arc_set_range(arc, 0, 1000);
lv_arc_set_value(arc, 0);
lv_obj_align(arc, LV_ALIGN_TOP_MID, 0, 100);
```

kisvegabor, 2024-02-15:
Ah, it's really not working with RGB565A8. The reason is that the A8 channel is treated as a mask during SW rendering, but arc drawing also wants to mask out the arc from the images A8 channel and the two things are not working togother.

We have 2 options to resolve it:
- Add RGB565A8 as a soruce format for all SW blend functions including all ASM accelerators.  (I wouldn't like to to that :slightly_smiling_face: )
- Copy the A8 part of the image to RAM and make the arc masking work on that. We have a buffer anyway for the mask but it's [initialized to 0xff normally](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw_arc.c#L168-L169). The only difference is that now we need to initialize it to the related part of the image alpha channel. It's a local solution and not generic, but much simpler and less invasive. 

What do you think?

XuNeo, 2024-02-20:
I vote for second solution. It's much simpler.

kisvegabor, 2024-02-22:
I marked it a bug now, and I'll look into it soon.

DC2047, 2024-02-23:
Very efficient! 👍

kisvegabor, 2024-03-13:
> Very efficient! 👍
Sorry, not so effective, I still couldn't get back to this...

kisvegabor, 2024-04-03:
I'm sorry that it took so long. I've opened https://github.com/lvgl/lvgl/pull/6009 to fix it.

PR #6009 (2024-04-23): feat(arc): support RGB565A8 arc image
fixes #5582

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](h [... cut]
closed 2024-04-23 by kisvegabor