lvgl/lvgl #5587: Enabling 'LV_USE_FLOAT` breaks arc animation

hyansuper, 2024-02-04:
### LVGL version

v9.0.0

### What happened?

related: https://github.com/lvgl/lvgl/issues/3093

Enabling `LV_USE_FLOAT` stops my arc from animating, can you please varify?

Other issues:
* I enable `LV_USE_FLOAT` macro only to print float number. I think we should separate `LV_USE_FLOAT` into `LV_USE_FLOAT_IN_PRINTF` and `LV_USE_FLOAT_PRECISE_VALUE`
* `lv_arc_set_xxx_angle` and `lv_img_set_rotation` , `lv_obj_set_style_transform_rotation` should all choose between  0.1 deg integer or float, depending on `LV_USE_FLOAT`

### How to reproduce?

```
lv_anim_t a;
lv_anim_set_var(&a, arc);
lv_anim_set_exec_cb(&a, lv_arc_set_start_angle);
...
lv_anim_start(&a);
```
but if I cast the value to int, arcs can animate.
```
void func(lv_obj_t* obj, int val) {
    lv_arc_set_start_angle(obj, val);
}
lv_anim_t a;
lv_anim_set_var(&a, arc);
lv_anim_set_exec_cb(&a, func);
...
lv_anim_start(&a);

```

XuNeo, 2024-02-04:
You can use a custom `exec_cb` to convert int to float and then call `lv_arc_set_start_angle`. Animation doesn't support float number yet.

hyansuper, 2024-02-11:
I really think we should distinguish these two features 

> I enable LV_USE_FLOAT macro only to print float number. I think we should separate LV_USE_FLOAT into LV_USE_FLOAT_IN_PRINTF and LV_USE_FLOAT_PRECISE_VALUE

kisvegabor, 2024-02-12:
> I enable LV_USE_FLOAT macro only to print float number. I think we should separate LV_USE_FLOAT into LV_USE_FLOAT_IN_PRINTF and LV_USE_FLOAT_PRECISE_VALUE

It still wouldn't solve the issue  in general. If someone enables `LV_USE_FLOAT` the animations are not working as intended.

However, `lv_anim_set_exec_cb(&a, lv_arc_set_start_angle);` should give a warning because of the mismatch in function prototypes. Therefore the "good" way of doing that it really create a new function.

closed 2024-03-05 by lvgl-bot