lvgl/lvgl #5589: Is there a replacement for `LV_DITHER_ORDERED` in v9.0.0

s-hadinger, 2024-02-04:
### Introduce the problem

I see that `LV_DITHER_ORDERED` was removed in commit https://github.com/lvgl/lvgl/commit/3ff625483c00458ebcf54307fad16b1cc402ff32 but I don't see any hint of how to enable dithering again.

s-hadinger, 2024-02-07:
Thanks, that's just a nice to have when doing a gradient background on large screens.

C47D, 2024-02-07:
> There is no replacement at this moment. We needed to rework several things with gradients and runtime generated dithering was already very complicated and in my opinion rarely used.

Maybe this should be mentioned in the changelog. "Removed because of its complexity and working on the replacement"

kisvegabor, 2024-02-10:
@C47D good idea, added in #5632

@s-hadinger as a workaround e.g. a 10x240 dithered tiled image can be used.

closed 2024-02-25 by s-hadinger