lvgl/lvgl #5590: lvgl touchpad calback function for coordinates does not work

Chineko-Koizumi, 2024-02-05:
### LVGL version

v9.0.0

### What happened?

### LVGL version

v9.0.0

### What happened?
I'm using ESP32 S3 with TFT ILI9341, callback function for getting coordinates does not work. But calling tft->getTouch( &x, &y ) directly in loop function does work for me. I have no idea what happening. I did small changes in "lv_tft_espi.c" and "lgvl.h" which are indicated below.

### How to reproduce?

//lv_tft_espi.c

lv_display_t * lv_tft_espi_create(uint32_t hor_res, uint32_t ver_res, void * buf, uint32_t buf_size_bytes)
{
    lv_tft_espi_t * dsc = (lv_tft_espi_t *)lv_malloc_zeroed(sizeof(lv_tft_espi_t));
    LV_ASSERT_MALLOC(dsc);
    if(dsc == NULL) return NULL;

    lv_display_t * disp = lv_display_create(hor_res, ver_res);
    if(disp == NULL) {
        lv_free(dsc);
        return NULL;
    }

    dsc->tft = new TFT_eSPI(hor_res, ver_res);
    dsc->tft->begin();          /* TFT init */
    dsc->tft->setRotation(0);   /* Changed it to 0 for Vertical orientation */

    Serial.printf("Created pointer: %X\n", dsc->tft);

    lv_display_set_driver_data(disp, (void *)dsc);
    lv_display_set_flush_cb(disp, flush_cb);
    lv_display_set_buffers(disp, (void *)buf, NULL, buf_size_bytes, LV_DISPLAY_RENDER_MODE_PARTIAL);
    return disp;
}

//lvgl.h

//...
#include "src/display/lv_display_private.h" //needed for _lv_display_t in main
//... 

//main.cpp
#include <TFT_eSPI.h>
#include <lvgl.h>

SET_LOOP_TASK_STACK_SIZE(32*1024)

#define TFT_HOR_RES   240 
#define TFT_VER_RES   320 

#define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES * 2)

lv_indev_t * indev   = nullptr;
lv_obj_t *label      = nullptr;

uint8_t draw_buf[DRAW_BUF_SIZE];
lv_display_t * disp;

TFT_eSPI * tft = nullptr;

void touchpad_read( lv_indev_t * indev, lv_indev_data_t * data )
{
    static uint16_t x, y;

    if(tft->getTouch( &x, &y )) 
    {
        data->point.x = x;
        data->point.y = y;
        data->state = LV_INDEV_STATE_PRESSED;
        
        Serial.printf("%U, %U\n", x, y);
    } 
    else
    {
        data->state = LV_INDEV_STATE_RELEASED;
    }
}

void setup()
{
    Serial.begin( 115200 );
    Serial.printf( "Hello Arduino! Version: %d.%d.%d \n", lv_version_major(), lv_version_minor(), lv_version_patch());

    lv_init();

    disp = lv_tft_espi_create(TFT_HOR_RES, TFT_VER_RES, draw_buf, DRAW_BUF_SIZE);
    lv_display_set_default(disp);

    struct TFT_Init
    {
        TFT_eSPI * tft;
    };

    TFT_Init * temp = reinterpret_cast<TFT_Init *>(disp->driver_data);
    tft = temp->tft;

    uint16_t calData[5] = { 242, 3583, 325, 3617, 4 };
    tft->setTouch(calData);

    indev = lv_indev_create();
    lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER);
    lv_indev_set_mode(indev, LV_INDEV_MODE_TIMER);          //---+
    lv_indev_set_read_cb(indev, touchpad_read);                       //---+---does not work with this or without this functions
    lv_indev_enable(indev, true);                                               //---+

    label = lv_label_create( lv_scr_act() );
    lv_label_set_text( label, "Hello Arduino, I'm LVGL!" );
    lv_obj_align( label, LV_ALIGN_CENTER, 0, 0 );

    Serial.println( "Setup done" );
}

void loop()
{
    delay( lv_timer_handler() );
}

kisvegabor, 2024-02-07:
>         Serial.printf("%U, %U\n", x, y);

What does it log?

> delay( lv_timer_handler() );

Please try
```c
lv_timer_handler();
delay(1);
```

Chineko-Koizumi, 2024-02-07:
@kisvegabor 
> What does it log?
before suggested change: nothing
after suggested change: nothing

(as I said earlier calling tft->getTouch( &x, &y ) directly from loop does work )

I did enable TRACE_LOG and have confirmed from logs that timer is working
[image]

There is a clear issue that LVGL does not call callback function for input read and I have no idea why, because according to documentation I did correct initialization for lv_indev.

kisvegabor, 2024-02-07:
Does the UI (i.e. the rendering) work? Please call e.g. this `lv_example_anim_2();` to see an infinite animation.

Chineko-Koizumi, 2024-02-07:
I did
``` 
 ...
  lv_example_anim_2();
    Serial.println( "Setup done" );
}

void loop()
{
    lv_timer_handler();
    delay(1);
}
```
and 

```
...

void loop()
{
   lv_example_anim_2();
    lv_timer_handler();
    delay(1);
}
```

both are static and displaying only first frame of animation

kisvegabor, 2024-02-07:
It means lvgl stuck somewhere. Please add a longer trace level log in a file.

kisvegabor, 2024-02-07:
Oh, something just came my mind. Please add lv_tick_inc(1) to the loop.

Chineko-Koizumi, 2024-02-07:
@kisvegabor Yep it works now (animation and touch) thank you.

May I ask what is an explanation for that behavior?

kisvegabor, 2024-02-08:
You have to provide a tick for LVGL so that it know when to run what. See [here](https://docs.lvgl.io/master/porting/tick.html).

In Arduino you can do this as well `lv_tick_set_cb(Millis);`. After that LVGL keeps calling `Millis` to know current tick count.

closed 2024-02-13 by Chineko-Koizumi