lvgl/lvgl #5595: Error reported when compiling the latest version    ， lv_conf_internal.h:59:18: fatal error: ../../lv_conf.h: No such file or directory

zuoganggit, 2024-02-05:
### LVGL version

v9.0.0

### What happened?

cmake compiling error,  errinfo is lv_conf_internal.h:59:18: fatal error: ../../lv_conf.h: No such file or directory

### How to reproduce?

#define LV_CONF_SKIP
/*If lv_conf.h is not skipped include it*/
#ifndef LV_CONF_SKIP
    #ifdef LV_CONF_PATH                           /*If there is a path defined for lv_conf.h use it*/
        #define __LV_TO_STR_AUX(x) #x
        #define __LV_TO_STR(x) __LV_TO_STR_AUX(x)
        #include __LV_TO_STR(LV_CONF_PATH)
        #undef __LV_TO_STR_AUX
        #undef __LV_TO_STR
    #elif defined(LV_CONF_INCLUDE_SIMPLE)         /*Or simply include lv_conf.h is enabled*/
        #include "lv_conf.h"
    #else
        #include "../../lv_conf.h"                /*Else assume lv_conf.h is next to the lvgl folder*/
    #endif
    #if !defined(LV_CONF_H) && !defined(LV_CONF_SUPPRESS_DEFINE_CHECK)
        /* #include will sometimes silently fail when __has_include is used */
        /* https://gcc.gnu.org/bugzilla/show_bug.cgi?id=80753 */
        #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
    #endif
#endif

C47D, 2024-02-06:
Are you using an specific platform? MCU or simulator? I'm working with the latest master commit and I see no issues.

Ev1lT3rm1nal, 2024-02-08:
Same error, Windows 11, Arduino IDE 2.3.0, lvgl@9.0.0

C47D, 2024-02-08:
I'll install Arduino in my Windows 11 machine and let you know if I can replicate the issue.

C47D, 2024-02-08:
Are you following the instructions from [here](https://docs.lvgl.io/master/integration/framework/arduino.html)? I just did and got a successful compilation

[image]

closed 2024-03-03 by kisvegabor