lvgl/lvgl #5598: discussion keyboard and ime

huenrong, 2024-02-05:
### Introduce the problem

Add a button to the virtual keyboard to switch the input method (Chinese and English), just like the mobile phone virtual keyboard. This makes it more convenient to use
I find some demos
- [在 RT1050 LittleVgl GUI 中嵌入中文输入法框架](https://community.nxp.com/t5/NXP-Designs-Knowledge-Base/%E5%9C%A8-RT1050-LittleVgl-GUI-%E4%B8%AD%E5%B5%8C%E5%85%A5%E4%B8%AD%E6%96%87%E8%BE%93%E5%85%A5%E6%B3%95%E6%A1%86%E6%9E%B6/ta-p/1366562)
- [lvgl8.0.1_cn_keyboard](https://github.com/xiaohua4758/lvgl8.0.1_cn_keyboard)

Sorry, some demos are in Chinese

kisvegabor, 2024-02-06:
Hi,

Does it mean by clicking the Zh/En button the keyboard switches to Pinyin IME mode? And is it a demo or an extension to the keyboard?

huenrong, 2024-02-06:
> Hi,
> 
> Does it mean by clicking the Zh/En button the keyboard switches to Pinyin IME mode? And is it a demo or an extension to the keyboard?

Yes, this new button (Zh/En) toggles input of Chinese and English. This feature could be a new component like Qt's virtual keyboard(qtvirtualkeyboard)

kisvegabor, 2024-02-10:
It would be nice to integrate it with the  [`lv_keyboard`](https://docs.lvgl.io/master/widgets/keyboard.html) widget. If you see a solution for it, could you send a Pull request?

huenrong, 2024-02-19:
> It would be nice to integrate it with the [`lv_keyboard`](https://docs.lvgl.io/master/widgets/keyboard.html) widget. If you see a solution for it, could you send a Pull request?

I think the two demos mentioned earlier are not very complete, I will try to refer to them for modification. This may take a while

huenrong, 2024-03-04:
@kisvegabor Hi, I submitted a PR(#5779) for this issue

kisvegabor, 2024-03-21:
The discussion continued in https://github.com/lvgl/lvgl/pull/5779.

closed 2024-03-21 by kisvegabor