lvgl/lvgl #5599: lv_draw_sw_rotate() 90 and 270 rotation are swapped

bbkarim, 2024-02-05:
### LVGL version

v9.0.0

### What happened?

I tried to use the lv_draw_sw_rotate() to perform software 90° rotation of my framebuffer but the result was inappropriate.

Touchscreen points was correct, but the buffer was rotated in the other angle (270 instead of 90).
After swapping implementations of rotate90_rgbxxxx() and rotate270_rgbxxxx() I got the correct behavior inline with touchscreen points. The result is then the same as v8.x with disp_drv.sw_rotate set to '1'

kisvegabor, 2024-02-06:
Hi Karim,

So let's say you have a 320x240 screen by default and the touchpad works well.

You call `lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90);` so LVGL will render in 240x320 and will rotate the touch points too.

However your hardware can't handle the rotation, you call `lv_draw_sw_rotate(..., LV_DISPLAY_ROTATION_90,...)` but the touchpoints are not correct.

Could you confirm if you really do it this way?

kisvegabor, 2024-02-22:
Will be resolved here: #5669

closed 2024-02-22 by kisvegabor

vishvanathdoijad, 2024-08-12:
> ### LVGL version
> v9.0.0
>
> ### What happened?
> I tried to use the lv_draw_sw_rotate() to perform software 90° rotation of my framebuffer but the result was inappropriate.
>
> Touchscreen points was [...]

Hi Can you share me your fluch_cb() function?
I am pretty close to the success, Just missing smaller things.
Thanx