lvgl/lvgl #5605: Has LVGL V9 decided to no longer support LV_USE_SJPG?

Drifter2016, 2024-02-06:
### Introduce the problem

I noticed that in LVGL V9's lv_conf.h, the LV_USE_SJPG configuration is no longer available and has been replaced by LV_USE_TJPGD. Additionally, there is no sjpg in the lib directory. Does this mean that SJPG will no longer be supported in the future with V9?

### Proposal

In LVGL 8.3, I was able to quickly open JPG files from an SD card using SJPG. However, in LVGL 9, despite enabling LV_USE_TJPGD for decoding, I am unable to display the images.

Do you have any suggestions for me? Thank you.
[image]

kisvegabor, 2024-02-06:
Please send the image you use and an code snippet to reproduce the issue.

PGNetHun, 2024-02-06:
Hello!
Yes, the **Split JPEG** feature is removed in LVGL v9, because TJPGD decoder is now used in **partial decoding mode**.
It means, that part of the JPEG image can be decoded also, the case previous SJPG feature was created for.
So from now on you can add your original JPEG files (still [baseline JPEG format](https://web.dev/learn/images/jpeg#progressive_jpeg)), no need for SJPG trick anymore.

See partial decoding function: https://github.com/lvgl/lvgl/blob/master/src/libs/tjpgd/lv_tjpgd.c#L205

Drifter2016, 2024-02-06:
> Hello! Yes, the **Split JPEG** feature is removed in LVGL v9, because TJPGD decoder is now used in **partial decoding mode**. It means, that part of the JPEG image can be decoded also, the case previous SJPG feature was created for. So from now on you can add your original JPEG files (still [baseline JPEG format](https://web.dev/learn/images/jpeg#progressive_jpeg)), no need for SJPG trick anymore.
> 
> See partial decoding function: https://github.com/lvgl/lvgl/blob/master/src/libs/tjpgd/lv_tjpgd.c#L205
> 
> (JPEG decoding works for me - I use it for background image -, although I noticed that rendering is slower)

Thank you for your response, but in my environment, JPG images still cannot be displayed. Even if I refresh the screen immediately after displaying a PNG image, it is not displayed correctly. There is a line of the original background color that gets refreshed. Could this be due to the delay in reading from the SD card, resulting in abnormal caching?

Drifter2016, 2024-02-06:
Anyway, help me take a look at the code first, is there any problem with the settings in lv_conf.h that may be causing the trouble I am encountering now, thank you.

PGNetHun, 2024-02-06:
Please enable LVGL logging: `LV_USE_LOG`
... and then after `lv_init()` register print: `lv_log_register_print_cb()`
(or enable `LV_LOG_PRINTF`).
Let's see whether there are error or warning log messages.

Drifter2016, 2024-02-06:
> Please enable LVGL logging: `LV_USE_LOG` ... and then after `lv_init()` register print: `lv_log_register_print_cb()` (or enable `LV_LOG_PRINTF`). Let's see whether there are error or warning log messages.

Thank you for your attention. I have enabled logging and printed the LV_LOG_LEVEL_INFO content. The print result is as follows:
```
[322630][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 8 - ASSOC_LEAVE
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42113e1e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   265][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Initializing...
SD card success!
LVGL:  [Info]	(0.000, +0)	 lv_obj_create: begin lv_obj.c:100

LVGL:  [Info]	(0.000, +0)	 lv_obj_create: begin lv_obj.c:100

LVGL:  [Info]	(0.000, +0)	 lv_obj_create: begin lv_obj.c:100

LVGL:  [Info]	(0.000, +0)	 lv_obj_create: begin lv_obj.c:100

LVGL:  [Info]	(0.000, +0)	 lv_obj_create: begin lv_obj.c:100

lv_image_create img_1
LVGL:  [Info]	(0.000, +0)	 lv_image_create: begin lv_image.c:123

lv_image_create img_2
LVGL:  [Info]	(0.000, +0)	 lv_image_create: begin lv_image.c:123

Refresh the screen.
```

Drifter2016, 2024-02-06:
I tried [lv_image_decoder_open](https://docs.lvgl.io/master/overview/image.html#manually-use-an-image-decoder) and was able to read the header of the JPG, but why can't it be displayed?
```
lv_result_t res;
lv_image_decoder_dsc_t dsc;
lv_image_decoder_args_t args = {0}; 
res = lv_image_decoder_open(&dsc, "S:/images/wallpaper/img_lvgl_logo.jpg", &args);

if (res == LV_RESULT_OK)
{
    Serial.println("decode jpg ok.");
    Serial.print("w: ");
    Serial.print(dsc.header.w);
    Serial.print("  h: ");
    Serial.println(dsc.header.h);
    Serial.print("src_type: ");
    Serial.println(dsc.src_type);
}
```
**print:**
decode jpg ok.
w: 105  h: 33
src_type: 1

kisvegabor, 2024-02-07:
I've tested on PC with the images, code and settings you have provided but it works well on my end. :slightly_smiling_face: 

A few things to try:
- Set `LV_CACHE_DEF_SIZE 0`
- Show only the jpg image
- Try reproducing the issue in [GitHub Codepaces](https://blog.lvgl.io/2023-04-13/monthly-newsletter)

Drifter2016, 2024-02-07:
> I've tested on PC with the images, code and settings you have provided but it works well on my end. 🙂
> 
> A few things to try:
> 
> * Set `LV_CACHE_DEF_SIZE 0`
> * Show only the jpg image
> * Try reproducing the issue in [GitHub Codepaces](https://blog.lvgl.io/2023-04-13/monthly-newsletter)

My friend, when I set LV_CACHE_DEF_SIZE to 0, none of the images are displayed, even if it's just a single PNG image. 

I'm using the ESP32-S3-WROOM-1-N16R8 module, Arduino framework, and PlatformIO platform for development.

A few hours ago, I tried bodmer/JPEGDecoder@^2.0.0, and with the example provided by this library, I was able to parse and display JPEG images.

 I will print and check the execution of the following function later.

lv_tjpgd.c
```
void lv_tjpgd_init(void)
{
    lv_image_decoder_t * dec = lv_image_decoder_create();
    lv_image_decoder_set_info_cb(dec, decoder_info);
    lv_image_decoder_set_open_cb(dec, decoder_open);
    lv_image_decoder_set_get_area_cb(dec, decoder_get_area);
    lv_image_decoder_set_close_cb(dec, decoder_close);
}

```

Also, is the library in the lvgl/src/libs/tjpgd directory an older version? I noticed that the latest one has some changes.

Drifter2016, 2024-02-07:
Is this a bug?
> lvgl/src/libs/tjpgd/

**At the end of the function decoder_get_area in lv_tjpgd.c:**
```
    /* Output the MCU (YCbCr to RGB, scaling and output) */
    rc = jd_mcu_output(jd, NULL, decoded_area->x1, decoded_area->y1);
    if(rc != JDR_OK) return rc;

    return LV_RESULT_OK;
```
**At the end of the function jd_mcu_output in tjpgd.c."**
```
    /* Output the rectangular */
    if(outfunc) return  outfunc(jd, jd->workbuf, &rect) ? JDR_OK : JDR_INTR;
    return 0;
```

kisvegabor, 2024-02-10:
I suggest using a [Simulator on PC ](https://docs.lvgl.io/master/integration/ide/pc-simulator.html) and load the `lv_example_tjpgd_1()` example. At least it filters out the issues with the SD card driver. 

> The second parameter, outfunc, is passed as NULL.

It just saves an extra operation. [Here](https://github.com/lvgl/lvgl/blob/master/src/libs/tjpgd/lv_tjpgd.c#L228) we use tjpegd's workbuf as the decoded data. Therefore the decoded data don't need to be output here because `decoded->data` will be blended by LVGL [here](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_image.c#L243-L267).

Drifter2016, 2024-02-13:
Thank you for your response.
I will give it another try, maybe it is due to the hardware environment.

closed 2024-03-07 by lvgl-bot