lvgl/lvgl #5606: Bug of label with DOT mode

seadog-www, 2024-02-06:
### LVGL version

V8.2

### What happened?

When the content of label is conbined with chinese char and number, there will be a problem, only showing “…”. As the following picture shown

[image]

### How to reproduce?

#!/opt/bin/lv_micropython -i

import lvgl as lv
import display_driver
import fs_driver

fs_drv = lv.fs_drv_t()
fs_driver.fs_register(fs_drv, 'S')
font_simsun_16_cjk = lv.font_load("S:../../assets/font/lv_font_simsun_16_cjk.fnt")

label1 = lv.label(lv.scr_act())
label = label1
label.set_style_text_font(font_simsun_16_cjk, 0)
label.set_long_mode(lv.label.LONG.DOT)     
label.set_text("机123456")
label.set_size(70, 20)                
label.align(lv.ALIGN.TOP_MID, 0, 20)
label.set_style_bg_color(lv.color_hex(0xFF00FF), 0)
label.set_style_bg_opa(100, 0)

label2 = lv.label(lv.scr_act())
label = label2
label.set_style_text_font(font_simsun_16_cjk, 0)
label.set_long_mode(lv.label.LONG.DOT)     
label.set_text("机1234567")
label.set_size(70, 20)                
label.align(lv.ALIGN.TOP_MID, 0, 60)
label.set_style_bg_color(lv.color_hex(0x00FFFF), 0)
label.set_style_bg_opa(100, 0)

kisvegabor, 2024-02-07:
It's known issue which is not resolved yet even in v9. :slightly_frowning_face:

seadog-www, 2024-02-18:
> It's known issue which is not resolved yet even in v9. 🙁

Alright, it seems like I'll have to find a temporary solution on my own for now.

kisvegabor, 2024-02-19:
I've just found a forgotten ancient feature which solves this issue. Just set `LV_TXT_LINE_BREAK_LONG_LEN 1`

Context: You see the issue  because LVGL assumes that long texts can be wrapped    on any Chinese character. And in your case the text is wrapped automatically like
```
机
123456
``` 

But the second line is clipped, so what remains is some dots in the first line.

`LV_TXT_LINE_BREAK_LONG_LEN 1` tells LVGL that it can wrap even after 1 character, so the text will become:
```
机123
456
```

seadog-www, 2024-02-20:
> I've just found a forgotten ancient feature which solves this issue. Just set `LV_TXT_LINE_BREAK_LONG_LEN 1`
>
> Context: You see the issue because LVGL assumes that long texts can be wrapped on any [...]

Yes, it does work.
However, after setting LV_TXT_LINE_BREAK_LONG_LEN 1, a complete word may be broken in middle in LV_LABEL_LONG_WRAP mode, which may be unacceptable for someone.
Perhaps these two modes(LV_LABEL_LONG_WRAP and LV_LABEL_LONG_DOT) should not share the same parameter(LV_TXT_LINE_BREAK_LONG_LEN )

kisvegabor, 2024-02-22:
I think the real solution would be to not wrap the text in the last line in case of `LV_LABEL_LONG_DOT`.

As a workaround you can just comment out this [line](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_text.h#L182).

niklasf, 2024-06-12:
Tried to implement this in #6362.

PR #6362 (2024-07-15): fix(label): do not break last line for LV_LABEL_LONG_DOT (#5606)
Fixes #5606 by implementing and using a new flag `LV_TEXT_FLAG_BREAK_ALL` (like [`word-break: break-all` in CSS](https://developer.mozilla.org/en-US/docs/Web/CSS/word-break)). The private `lv_text_get_next_word()` already had a `force` parameter, which is replaced by the new public flag.

Testing: Covered by existing reference images. Note how the last line is no longer cut off too early.
closed 2024-07-15 by FASTSHIFT