lvgl/lvgl #5620: Unknown opcode error using LVGL 9.0.0 in Arduino IDE 2.3.0 and ESP 2.0.14

Superberti, 2024-02-07:
### LVGL version

v9.0.0

### What happened?

After including <lvgl.h> I get the following strange errors while compiling (from the assembler?):
Yo do not need to use any lvgl specific command, the header is enough to reproduce this.
All I do is to use TFT_eSPI, but I don't know if this is important. Without lvgl TFT_eSPI runs perfect.

**c:\users\rutsch\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\sys-include\machine\_default_types.h: Assembler messages:
c:\users\rutsch\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\sys-include\machine\_default_types.h:41: Error: unknown opcode or format name 'typedef'
[...lot of more lines with the same opcode error...]**

A few weeks ago I compiled successfully lvgl for the ESP32 and it worked out of the box. Here I'm out of ideas.

### How to reproduce?

```
#include <TFT_eSPI.h>  
#include <lvgl.h>
#include <SPI.h>

void setup(void) {
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
}

void loop() {
  //lv_timer_handler();
  delay(5);
}
```

kisvegabor, 2024-02-07:
@espzav @tore-espressif do you have any idea for this?

C47D, 2024-02-08:
From [here](https://github.com/lvgl/lvgl/blob/master/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino) the headers should be:

```cpp
#include <lvgl.h>

#if LV_USE_TFT_ESPI
#include <TFT_eSPI.h>
#endif
```

Superberti, 2024-02-08:
Yes, this was the trick ;-)

So to be complete when updating from 8.3.0 to 9.0.0:

1. Update your lv_conf.h from the new lv_conf_template.h (from the lvgl dir)
2. Set LV_USE_TFT_ESPI 1 in your lv_conf.h
3. Use the includes as shown above!

Many thanks for the fast response!!!

WebDust21, 2024-07-07:
> 1. Update your lv_conf.h from the new lv_conf_template.h (from the lvgl dir)

This here being the actual solution.

Trying to upgrade a project from LVGL 8.4.0 to LVGL 9.1.0--and ran into this exact problem.  Turns out that there were no #defines to disable the precompiled assembler "helium" and "arm" blending functions--which is where all the errors were coming from (as you aren't going to be able to run ARM code on an Tensilica Xtensa processor!)

https://github.com/lvgl/lvgl/blob/f8c26ea150189d26bcd0c0cd98d45bf615ed692f/lv_conf_template.h#L146-L150

I don't understand why these very specialized functions aren't "disabled by default"--but hey, at least they got me to update the project's entire "lv_conf.h"!

kisvegabor, 2024-07-10:
So you "just" needed to update lv_conf.h and it works now?

WebDust21, 2024-07-10:
> So you "just" needed to update lv_conf.h and it works now?

Yup, that's all it was.

vednildip, 2025-01-05:
following  error  with  new  version  9.2.2  "/esp-x32/2405/xtensa-esp-elf/include/stdint.h:71: Error: unknown opcode or format name 'typedef'"   &  "xtensa-esp-elf/include/machine/_default_types.h:77: Error: unknown opcode or format name 'typedef'"

Vladivarius, 2025-09-15:
I fixed this error by moving lv_conf.h to lvgl/src/