lvgl/lvgl #5624: Vendor/Platform specific default in Kconfig and/or lv_conf.h

kisvegabor, 2024-02-08:
### Introduce the problem

As it was shown in #5614 and #5391 adding vendor/Platform specific default configs can be useful.

Use cases:
- `LV_CONF_SKIP` in Kconfig: I think we can set it to `y` unconditionally as we can assume that if there is Kconfig there is no `lv_conf.h`
- `LV_USE_STDLIB_*`: Some might want to use Clib or custom STDLIB implementations
- `LV_USE_DRAW_SW_ASM` settings
- [Profiler config](https://docs.lvgl.io/master/overview/profiler.html) (Not in the config files now)
- `LV_USE_OS`
- `LV_DRAW_BUF_STRIDE_ALIGN` to match the requirements of GPUs by default
- `LV_USE_DRAW_<GPU>`: enable automatically?
- All `LV_ATTRIBUTE_*`s
- Maybe some drivers, e.g. `LV_USE_SDL`?
- Enable some platform specific libraries: e.g. `lodepng` vs `libpng`

### Proposal

As these settings can be hardware, SDK, OS and application specific too, it's hard to find how to make it flexible yet inututive.

It already came up several times to have 2 `lv_conf.h` files:
1. Target device specific (mainly for the configs I've mentioned above)
2. LVGL specific: pure LVGL settings

1) can be provided by vendors and we can also host some templates for some devices, and 2) could be used by the users to configure the application.

cc @tore-espressif @espzav @igrr @faxe1008 @MouriNaruto @nicusorcitu

MouriNaruto, 2024-02-08:
It would be good if LVGL could have separate platform-specific default configuration definition header files, and users can select and customize one of them via kconfig or lv_conf.h. Also, it will reduce the complexity of maintaining LVGL itself.

Kenji Mouri

IAMMX, 2024-02-08:
Kconfig and lv_conf.h, I think the official should only consider the most commonly used configuration. As for the special platform, shouldn't the user modify the two files according to their own platform? Why set default values based on specific platforms? Isn't the configuration file for users to configure themselves?

nicusorcitu, 2024-02-12:
@cristian-stoica

kisvegabor, 2024-02-12:
It's great that there is agreement that having 2 config files wold be beneficial. As there is the Chinese holiday now, let's wait a little with the final decision to be sue our Chinese contributors can also read this issue. 

Until that could you list from ESP side what features should be moved to this new platform specific config file?

I'm also wondering how will it work with Kconfig.  Shall we host some platform specific Kconfig files, and source them based on some conditions?

espzav, 2024-03-05:
Hi @kisvegabor, please, are you working on these changes now? When do you plan release version 9.1?

kisvegabor, 2024-03-13:
Not yet.
 v9.1 will be released next Tuesday. See https://github.com/lvgl/lvgl/issues/5843

kisvegabor, 2024-04-03:
Continue in #6010

closed 2024-04-03 by kisvegabor