lvgl/lvgl #5628: SDL API is limiting

kdschlosser, 2024-02-09:
### Introduce the problem

OK so with respect to handling of a display driver it is actually 2 separate parts. One part being the driver for the bus and the other part is the driver for the display. These are technically 2 things and are normally handled separately of each other. When dealing with this in C code it appears to be more of a seamless due to preprocessor macros that control what is being used. They are still 2 separate things. When dealing with this in a higher level runtime language having SDL written the way it is makes it impossible to alter things like the thread that gets run for the task handler updates and collecting the frame buffer from the lv_display_t structure. while the structure is returned when creating the display because the structure is a private thing there is no way to access the fields that hold the frame buffers.

The SDL driver should be written more like a bus driver and it  should not be running threads that call lv_tick_inc and lv_task_handler. It should not be creating the display in LVGL. The SDL driver should be responsible for handling the creation of the window and it should be responsible for passing the frame buffer data to that window only when it is told to do so.

The same things holds true for the keyboard driver and also the mouse driver. There should be a way to collect the needed information from those drivers either through manual calls functions of by registering a callback if the information is forwarded through an interrupt.

### Proposal

To write the SDL driver code so it doesn't package everything into something that makes it impossible to access specific parts and so it doesn't hard code duration for when things are done.

An example.

if you take a device like an Omega Onion which is an MCU that runs Linux it doesn't have the processing power of say a Raspberry Pi. It will run SDL2 but forcing it to run a thread that does updates every 5 milliseconds if going to overrun the processor. This is even more of an issue is the user has the refresh timer set as the default of 33 milliseconds. the CPU gets interrupted 6 times and no work gets done. That is a complete waste of resources.

Now because the SDL2 driver provides access to threads even tho there is no technical DMA availability for the frame buffer memory which makes double buffering a pointless thing to use we can actually provide a way to provide it using a thread that handles writing the buffer data.

The reason why this is causing an issue is I am working with MicroPython making frameworks for indev devices and displays and I am not able to do this with how the SDL drivers are currently set up. Having to code around it adding in equality operations with 2 different sets of code is going to make the code footprint really large and separating the code into 2 separate files that have different API's makes having a framework pointless because of the inconsistencies in the API. 

Now I understand the reason why it is done the way it is. It's for simplicity purposes but it causes problems if a user wants to develop on a PC and then simply swap out SDL for and MCU display. It is no longer something that is easy to do and it would require a large code change to be made in order to do that.

The same holds true for any of the drivers in the dev folder. Those drivers completely remove the concept of portability which is a unified API and only small things needed to be changed depending on the system it is running on. Now in C code this is less of an issue then it is with a high level programming language. 

I should be able to take the same GUI code that is written to run in MicroPython and run it on an MCU or run it in MicroPython that is compiled to run on Linux with only needing to change the names of the display and the bus that is being used. I shouldn't have to change large sections of code to make that possible.

For the purposes of MicroPython I could write the code needed to handle SDL and not use what is built into LVGL. That really doesn't make sense because that was the entire point of adding the code to LVGL. Pull the pieces apart and expose the structures to be able to access the different pieces. Let the user handle the creation of frame buffers if they want and let the user decide on how they want to handle the flushing of the frame buffer. bottling it all up inside of the source files where things are not able to be accessed really places limits that causes extra work.

kdschlosser, 2024-02-12:
These lines are problematic because they take over the indev and the user is not able to handle these things on their own. It is also off for it to do this because there are separate functions the user would call for making the drivers yet they are actually being made here.
https://github.com/lvgl/lvgl/blob/e6dc2f05d239b914617f19e1db019ebb35d83449/src/drivers/sdl/lv_sdl_window.c#L254
https://github.com/lvgl/lvgl/blob/e6dc2f05d239b914617f19e1db019ebb35d83449/src/drivers/sdl/lv_sdl_window.c#L255
https://github.com/lvgl/lvgl/blob/e6dc2f05d239b914617f19e1db019ebb35d83449/src/drivers/sdl/lv_sdl_window.c#L256

you are correct there no longer is a thread being used.
https://github.com/lvgl/lvgl/blob/e6dc2f05d239b914617f19e1db019ebb35d83449/src/drivers/sdl/lv_sdl_window.c#L88

Using the storage means that it is not able to be used upstream at all. I will explain more on this in a second.
https://github.com/lvgl/lvgl/blob/e6dc2f05d239b914617f19e1db019ebb35d83449/src/drivers/sdl/lv_sdl_window.c#L110

How the indev drivers are handled gets completely circumvented because of this.
https://github.com/lvgl/lvgl/blob/e6dc2f05d239b914617f19e1db019ebb35d83449/src/drivers/sdl/lv_sdl_window.c#L87C1-L87C75
The timer that is normally stored in the indev_t structure doesn't do anything. 

This completely removes the ability to use double buffering. 
https://github.com/lvgl/lvgl/blob/e6dc2f05d239b914617f19e1db019ebb35d83449/src/drivers/sdl/lv_sdl_window.c#L240C1-L240C34

Double buffering is able to be done because SDL provides access to threading. A thread is able to be created and it's sole responsibility would be do any updating to SDL for the purposes of writing the buffer data. Because LVGL already has mechanics built into it that prevent a buffer that is being written to the display from being written to by LVGL this is able to be done fairly easily.

The links below are to untested code but the idea behind them is sound.
https://github.com/kdschlosser/lvgl_micropython/blob/d39f656a429aaf3ea492136475b2f12e611aca70/ext_mod/lcd_bus/sdl_bus/sdl_bus.c#L166

https://github.com/kdschlosser/lvgl_micropython/blob/d39f656a429aaf3ea492136475b2f12e611aca70/ext_mod/lcd_bus/sdl_bus/sdl_bus.c#L177C9-L177C17

https://github.com/kdschlosser/lvgl_micropython/blob/d39f656a429aaf3ea492136475b2f12e611aca70/ext_mod/lcd_bus/sdl_bus/sdl_bus.c#L87

The last link is the function that gets called from the flush function to write the data to the display. One of the things you will notice is there is no lvgl display driver created or managed by the SDL "bus" driver. That is the responsibility of the code that is upstream.

This is the framework for display drivers. The API is the same across all display drivers no matter what kind of a data bus is used. 

https://github.com/kdschlosser/lvgl_micropython/blob/main/driver/frozen/display/display_driver_framework.py

Here is the display driver that uses the framework and is specific to SDL.
https://github.com/kdschlosser/lvgl_micropython/blob/main/driver/display/sdl_display.py

Going this route makes it easy to transition a project from running on linux to running on an MCU with an actual display. The user is able to do the following in their code.

```python
import sys
import lvgl as lv
import lcd_bus

if sys.platform == 'linux':
    data_bus = lcd_bus.SDLBUS(
        lcd_bus.SDLBus.WINDOW_BORDERLESS | lcd_bus.SDLBus.WINDOW_ALWAYS_ON_TOP
    )

    import sdl_display

    display = sdl_display.SDL(
        data_bus=data_bus,
        display_width=480,
        display_height=320,
    )

else:
    data_bus = lcd_bus.SPIBus(
        dc=2,
        host=1,
        sclk=5,
        freq=80000000,
        mosi=13,
        miso=12,
        cs=-1
    )

    import st7796

    display = st7796.ST7796(
        data_bus=data_bus,
        display_width=480,
        display_height=320,
        reset_pin=23,
        power_pin=24,
        backlight_pin=26,
        color_space=lv.COLOR_FORMAT.RGB565,
        rgb565_byte_swap=True
    )
```

the display that is created has the exact same API. They function in the exact same manner. This is code that is able to be used for development and also production because it is able to switch between the 2 and it does this at runtime.

kdschlosser, 2024-02-12:
OK so as far as setting the driver data goes. What I set in this place is the display driver that is created using the code above. This is done in the framework. The reason why this gets used in this manner is to link the python indev driver to the python display driver. so if the user changes the rotation in the display driver it is going to automatically change the rotation in the indev driver. If the user changed the display width and/or height this automatically gets populated into the indev driver that is attached to the display.

both the python indev driver and the python display driver have the functions that are seen in lvgl for accessing the various things for that display driver. The user has no need to call `lv_get_display_default()` to collect the display.

Unfortunately because the lv_display_t structure is not public I am not able to make a simple wrapper around the structure so it is not able to be passed to any LVGL functions. Same goes for any indev drivers. had those structures been made publicly available so the binding would generate the code that turns the structure into a class then I would be able to wrap the class so it would act exactly like the structure in LVGL.

kisvegabor, 2024-02-15:
Sorry for not responding to everything, but I would like to understand the basics first.

> Double buffering is able to be done because SDL provides access to threading. A thread is able to be created and it's sole responsibility would be do any updating to SDL for the purposes of writing [...]

Gotcha! And it's a good idea. Why don't we add it the the SDL driver then?

> if sys.platform == 'linux': [...]

Why is it better than just calling `lv.sdl_window_create(480, 320)`?

kdschlosser, 2024-02-17:
> Why is it better than just calling lv.sdl_window_create(480, 320)?

because the driver API is completely different. I have no way to collect things like the frame buffers because there is no function to do this and lv_display_t is private. 

when a user creates a display driver in LVGL that is not the SDL driver they are responsible for setting the frame buffers, and the color format. they are responsible for setting the flush function. All of this is done for the user inside of the SDL driver so the user doesn't have the ability to alter things like the color format because the size of the frame buffer is attached to the color format and the user doesn't have the ability to alter the size of the frame buffers. Setting the color format is a runtime thing not a compile time thing. but when using the SDL driver it becomes a compile time thing. 

It's all about having a consistent API regardless of what device the binding is running on. Running on Linux using SDL is going to make the API for the display different which would result in a lot of additional code the user would have to put in place if they wanted to develop on a desktop and then move the code to an STM32 or to an ESP32. If the API is able to be made the same then only a single test would need to be done like I have shown in the example I gave. There wouldn't need to be any special handling that would have to be coded in.

kdschlosser, 2024-02-17:
`sdl_display.SDL` and `st7796.ST7796` both inherit from the same base class and would have the same functionality.

closed 2024-03-13 by lvgl-bot