lvgl/lvgl #5631: Draw buf too small for new shape lv_draw_buf.c:235

phhoef, 2024-02-10:
### LVGL version

v9.0.0

### What happened?

When trying to change the text of a drawn label, I see `Draw buf too small for new shape lv_draw_buf.c:235` message in log.

VScode with PlatformIO on ESP32 S3 DevKit C with Arduino Framework.
Now, when I press my external switch I call the `change` method `NHD` class. I see the described log message.
I was logging the required size and the buffer size.
I added these two LV_LOG_INFO lines in `lv_draw_buf.c`  file at line 232.
```c
    LV_LOG_INFO("Draw Buffer size is %" LV_PRId32, draw_buf->data_size);
    LV_LOG_INFO("Required size is %" LV_PRId32, size);
    if(size > draw_buf->data_size) {
        LV_LOG_INFO("Draw buf too small for new shape");
        return NULL;
    }
```
```
[Info]  (12.331, +8991)  lv_draw_buf_reshape: Draw Buffer size is 15360 lv_draw_buf.c:232
[Info]  (12.331, +0)     lv_draw_buf_reshape: Required size is 3744 lv_draw_buf.c:233
[Info]  (12.390, +59)    lv_draw_buf_reshape: Draw Buffer size is 40 lv_draw_buf.c:232
[Info]  (12.478, +88)    lv_draw_buf_reshape: Required size is 64 lv_draw_buf.c:233
[Info]  (12.563, +85)    lv_draw_buf_reshape: Draw buf too small for new shape lv_draw_buf.c:235
[Info]  (12.663, +100)   lv_draw_buf_reshape: Draw Buffer size is 64 lv_draw_buf.c:232
[Info]  (12.752, +89)    lv_draw_buf_reshape: Required size is 88 lv_draw_buf.c:233
[Info]  (12.837, +85)    lv_draw_buf_reshape: Draw buf too small for new shape lv_draw_buf.c:235
[Info]  (12.937, +100)   lv_draw_buf_reshape: Draw Buffer size is 88 lv_draw_buf.c:232
[Info]  (13.026, +89)    lv_draw_buf_reshape: Required size is 88 lv_draw_buf.c:233
[Info]  (13.113, +87)    lv_draw_buf_reshape: Draw Buffer size is 88 lv_draw_buf.c:232
[Info]  (13.200, +87)    lv_draw_buf_reshape: Required size is 22 lv_draw_buf.c:233
[Info]  (13.286, +86)    lv_draw_buf_reshape: Draw Buffer size is 88 lv_draw_buf.c:232
[Info]  (13.374, +88)    lv_draw_buf_reshape: Required size is 64 lv_draw_buf.c:233
[Info]  (13.459, +85)    lv_draw_buf_reshape: Draw Buffer size is 88 lv_draw_buf.c:232
[Info]  (13.548, +89)    lv_draw_buf_reshape: Required size is 15 lv_draw_buf.c:233
[Info]  (13.633, +85)    lv_draw_buf_reshape: Draw Buffer size is 88 lv_draw_buf.c:232
[Info]  (13.722, +89)    lv_draw_buf_reshape: Required size is 56 lv_draw_buf.c:233
```

I also noticed, that I MUST call `lv_timer_handler` at the end of my `begin` method, otherwise it crashes and ESP32 resets.

### How to reproduce?

main.h
```c++
#ifndef MAIN_H
#define MAIN_H

#include <Arduino.h>
#include <OneButton.h>
#include "TFT/NHD/NHD.h"

#define UPPER_RIGHT_PIN 5

void setup();
void loop();
void click(void *param);

NHD nhd;
OneButton upperRight = OneButton(UPPER_RIGHT_PIN, false);

#endif
```

main.cpp
```c++
#include "main.h"

SET_LOOP_TASK_STACK_SIZE(20 * 1024);

void setup()
{
  Serial.begin(9600);

  nhd.begin();
  upperRight.attachClick(click);
}

void click()
{
  Serial.println("Click ...");
  nhd.change();
}

void loop()
{
  upperRight.tick();

  lv_timer_handler(); // I found out, that this needs to be done in main loop
  vTaskDelay(5 / portTICK_PERIOD_MS);
}
```

NHD.h
```c++
#ifndef NHD_H
#define NHD_H

#include <TFT_eSPI.h>
#include "../lib/lvgl/lvgl.h"

/*Set to your screen resolution*/
#define TFT_HOR_RES 320
#define TFT_VER_RES 240
#define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))

class NHD
{
protected:
    lv_display_t *display;
    lv_obj_t *label;
    lv_obj_t *label1;

public:
    void begin();
    void change();
};

#endif
```

NHD.cpp
```c++
#include "NHD.h"

void tick(void *ctx)
{
    while (1)
    {
        lv_tick_inc(1);
        vTaskDelay(1 / portTICK_PERIOD_MS);
    }
}

void NHD::begin()
{
    lv_init();

    uint8_t draw_buf[DRAW_BUF_SIZE];
    display = lv_tft_espi_create(TFT_HOR_RES, TFT_VER_RES, draw_buf, DRAW_BUF_SIZE);
    TFT_eSPI *tft = (TFT_eSPI *)lv_display_get_driver_data(display);
    tft->invertDisplay(0);

    xTaskCreate(tick, "tick", 3000, NULL, 1, NULL);
    label1 = lv_label_create(lv_scr_act());
    lv_label_set_text(label1, "Hallo!");
    lv_obj_align(label1, LV_ALIGN_CENTER, 0, 30);

    lv_timer_handler();
}

void NHD::change()
{
    if (!label)
    {
        label = lv_label_create(lv_scr_act());
        lv_label_set_text(label, "This is a extremely");
        lv_obj_align(label, LV_ALIGN_LEFT_MID, 5, 0);
    }
    else
    {
        lv_label_set_text(label, "other text");
    }
}
```

C47D, 2024-02-10:
Hi,

I would move the draw buffer out of the setup function and make it global, or a class member. Also create the tick thread after all the LVGL code. See `examples/arduino/LVGL_Arduino/LVGL_Arduino.ino`
Notice how the draw buffer is uint32_t and not uint8_t.

Sadly I don't have hardware to test it :/

phhoef, 2024-02-11:
Hi @C47D,

thanks for your thoughts.
I was using the download from the release asset and it seems that at least the example `LVLG_Arduino.ino` had some bugs. For example the buffer was `uint8_t` and it was also within the setup function ...
Now, I downloaded the latest code from the master branch and it got better, though not completely good 😉

Anyhow, I made some adaptions and it improved, but it still not working at all.
NHD.h
```c++
#ifndef NHD_H
#define NHD_H

#include <TFT_eSPI.h>
#include "../lib/lvgl/lvgl.h"

/*Set to your screen resolution*/
#define TFT_HOR_RES 320
#define TFT_VER_RES 240
#define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))

class NHD
{
protected:
    uint32_t draw_buf[DRAW_BUF_SIZE / 4];
    lv_display_t *display;
    lv_obj_t *label;
    lv_obj_t *label1;

public:
    void begin();
    void change();
};

#endif
```
Do you know, why the buffer size is divided by 4 `DRAW_BUF_SIZE / 4`?

NHD.cpp
```c++
#include "NHD.h"

void NHD::begin()
{
    lv_init();

    display = lv_tft_espi_create(TFT_HOR_RES, TFT_VER_RES, draw_buf, sizeof(draw_buf));
    TFT_eSPI *tft = (TFT_eSPI *)lv_display_get_driver_data(display);
    tft->invertDisplay(0);

    label1 = lv_label_create(lv_scr_act());
    lv_label_set_text(label1, "Hallo!");
    lv_obj_align(label1, LV_ALIGN_CENTER, 0, 30);

    lv_timer_handler();
    xTaskCreate(tick, "tick", 3000, NULL, 1, NULL);
}

void NHD::change()
{
    if (!label)
    {
        Serial.println("here");
        label = lv_label_create(lv_scr_act());
        lv_label_set_text(label, "This is a extremely long text, that needs scrolling to be visible at all");
        lv_label_set_long_mode(label, LV_LABEL_LONG_SCROLL_CIRCULAR);
        lv_obj_align(label, LV_ALIGN_LEFT_MID, 5, 0);
    }
    else
    {
        Serial.println("here2");
        lv_label_set_text(label, "other text");
    }
}
```
I've 
* changed the buffer type and size
* moved the task creation at the very end of the `begin` method.

Now on button press, the new label is at least drawn. 😄 
BUT, I cannot update the text of any label. Though, the log saying "Daw buff size too small`.

I tried adding `long_mode` for the 2nd label, but it does not scroll through the text.

For some weird reason, I've to call `` at the end of the `begin` method. Otherwise the screen looks like this
[image]

After button press it looks like 
[image]

In addition, I've to call
```c++
    TFT_eSPI *tft = (TFT_eSPI *)lv_display_get_driver_data(display);
    tft->invertDisplay(0);
```
Otherwise, the colors are inverted for some reason. Though, this is not directly related to `lvgl, as it also happens only with `TFT_eSPI`, but could this be a problem?

XuNeo, 2024-02-11:
Hi, 
I have removed the misleading log of buffer reshape failing. https://github.com/lvgl/lvgl/pull/5634
It is normal when drawing a label, that a new draw buffer will be created if reshape failed.

phhoef, 2024-02-13:
Things getting stranger.
I found out, that the display will update properly, when the `lv_timer_handler` method is called twice.
In addition to the call in the main loop, I created a new task and call it there every 20ms. Then, the display updates as expected.
All in all, there are now three position where I call this function
* main loop
* task
* after init sequence

This is my current code.
main.cpp
```c++
#include "main.h"

SET_LOOP_TASK_STACK_SIZE(20 * 1024);

void setup()
{
  Serial.begin(9600);

  nhd.begin();
  upperRight.attachClick(click);
}

void click()
{
  Serial.println("Click ...");
  nhd.change();
}

void loop()
{
  upperRight.tick();

  lv_timer_handler(); // I found out, that this needs to be done in main loop
  vTaskDelay(5 / portTICK_PERIOD_MS);
}
```

NHD.h
```c++
#ifndef NHD_H
#define NHD_H

#include <TFT_eSPI.h>
#include "../lib/lvgl/lvgl.h"

/*Set to your screen resolution*/
#define TFT_HOR_RES 320
#define TFT_VER_RES 240
#define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))

class NHD
{
protected:
    uint32_t draw_buf[DRAW_BUF_SIZE / 4];
    lv_display_t *display;
    lv_obj_t *label;
    lv_obj_t *label1;

public:
    void begin();
    void change();
};

#endif
```

NHD.cpp
```c++
static void flush_cb(lv_display_t *disp, const lv_area_t *area, uint8_t *px_map)
{
    TFT_eSPI *tft = (TFT_eSPI *)lv_display_get_driver_data(disp);

    uint32_t w = (area->x2 - area->x1 + 1);
    uint32_t h = (area->y2 - area->y1 + 1);

    tft->startWrite();
    tft->setAddrWindow(area->x1, area->y1, w, h);
    tft->pushColors((uint16_t *)px_map, w * h, true);
    tft->endWrite();

    lv_display_flush_ready(disp);
}

void timer(void *ctx)
{
    while (1)
    {
        lv_timer_handler();
        vTaskDelay(20 / portTICK_PERIOD_MS);
    }
}

void tick(void *ctx)
{
    while (1)
    {
        lv_tick_inc(1);
        vTaskDelay(1 / portTICK_PERIOD_MS);
    }
}

void log(lv_log_level_t level, const char *buf)
{
    ESP_LOGI("LVGL", buf);
}

void NHD::begin()
{
    lv_init();

    display = lv_display_create(TFT_HOR_RES, TFT_VER_RES);
    TFT_eSPI *tft = new TFT_eSPI(TFT_HOR_RES, TFT_VER_RES);
    tft->begin();        /* TFT init */
    tft->setRotation(1); /* Landscape orientation, flipped */
    tft->initDMA();
    lv_display_set_driver_data(display, (void *)tft);
    lv_display_set_flush_cb(display, flush_cb);
    lv_display_set_buffers(display, (void *)draw_buf, NULL, sizeof(draw_buf), LV_DISPLAY_RENDER_MODE_PARTIAL);

    label1 = lv_label_create(lv_scr_act());
    lv_label_set_text(label1, "Hallo!");
    lv_obj_align(label1, LV_ALIGN_CENTER, 0, 30);

    lv_timer_handler();
    xTaskCreate(tick, "tick", 3000, NULL, 1, NULL);
    xTaskCreate(timer, "handle", 4000, NULL, 1, NULL);
}

void NHD::change()
{
    if (!label)
    {
        Serial.println("here");
        label = lv_label_create(lv_scr_act());
        lv_label_set_text(label, "This is a extremely long text, that needs scrolling to be visible at all");
        lv_label_set_long_mode(label, LV_LABEL_LONG_SCROLL_CIRCULAR);
        lv_obj_align(label, LV_ALIGN_LEFT_MID, 5, 0);
        lv_obj_set_width(label, 100);
    }
    else
    {
        Serial.println("here2");
        lv_label_set_text(label, "other text");
    }
}
```

Does this make sense for anyone?
I cannot understand this behaviour at all...

C47D, 2024-02-13:
This is from the arduino example, the task handler must be called once and then the `lv_tick_inc`. I think you're missing calling the second function. This should be done in the tick thread as far as I know. 

```c
void loop()
{
    lv_task_handler(); /* let the GUI do its work */
    lv_tick_inc(5); /* tell LVGL how much time has passed */
    delay(5); /* let this time pass */
} 
```

C47D, 2024-02-15:
@kisvegabor Do we need to protect the LVGL calls with a mutex when they're done in a separated thread/task?

phhoef, 2024-02-16:
I am completely new to lvgl, so I have no insights.
But I wonder if this would be a problem of a mutex, why isn't it working when calling `lv_task_handler`and `lv_tick_inc` in the Arduino main loop? It's only working, when it's called twice in the main loop and in a separate task for some reason ...
And even then I think I found situations where the screen didn't update.

It feels unpredictable and I've no idea how to dig into it.
Do you want me to make some tests?

phhoef, 2024-02-23:
any idea, what's wrong with my code and why I see this strange behaviour? 😄

kisvegabor, 2024-02-24:
Hi,

Could you try it out without DMA?
```c
//   tft->initDMA();
```

phhoef, 2024-02-25:
Hallo @kisvegabor thanks for your reply.

DMA has unfortunately no influence. The behaviour is exactly the same with or without DMA.
I did two tests after commenting `tft->initDMA()`
1. I removed the two handler calls in the Arduino main loop
```cpp
  lv_timer_handler();
  lv_tick_inc(5);
```
and only did the calls in two tasks.
2. I removed the tasks and only called these handlers in Arduino main loop.

In both scenarios the behaviour is exactly the same.
There are a few parts of the screen which will not be completely initialised as shown on the screens in earlier comments.
In addition, it's then also not possible to change any widget ...

phhoef, 2024-03-02:
I found the problem: Other SPI connected devices caused the problem. I disconnected them, and it's working now. Thanks for your help!

closed 2024-03-02 by phhoef