lvgl/lvgl #5633: events no longer with in the micropython bnding

kdschlosser, 2024-02-11:
### LVGL version

V9.0.0

### What happened?

Because of making structures private the gen_py script is no longer able to locate the user_data field in lv_event_t and because it is not able to locate that field the callback function never gets made. This causes the script to generate code that attempts to convert the Python function into a C function pointer in which it fails.

Any structure that has anything to do with callbacks cannot be made private.

kisvegabor, 2024-02-12:
We hid `lv_display_t` and `lv_indev_t` in `*_private.h` files as well. In this case the binding uses `lv_..._get_user_data()`. Can we do the same for the events?

BTW, do you know why wasn't it discovered by the binding test?

kdschlosser, 2024-02-12:
> BTW, do you know why wasn't it discovered by the binding test?

Because there are no events that are generated. There is only animation callbacks being done in the test.

Me saying that it is because of things being made private is only partially correct. While lv_event_t is not a private structure it is using the same forward declaration mechanics as the structures that are private.

pycparser doesn't play "connect the dots" so to speak. when it reads the C code it does so quite literally.

How it reads the code is exactly how it is seen in the preprocessor output. It reads that output line by line. The preprocessor output if generated in the order in which header files are included. One of the first things that gets included is lv_types.h. Pycparser sees the forward declarations in lv_types.h and what are forward declarations? They are empty is what they are. It reads the typedef structs as structure declarations that have no fields and then then it reads that there is a typedef that points to the name of the structure but it never actually points to the structure itself. The structure has no fields so the code that runs in the gen_mpy script doesn't see any user_data field so it doesn't generate the needed code to make it work properly. This is due to the linear way that the gen_mpy script was written. It doesn't have the ability to look forward or backwards before generating code. It only looks to see if the name has already been used and if it has it moves to the next item. Because the code is output at the time it is added top it being used there is no easy way of checking and removing the forward declaration because the code for it has already been output for the forward declaration.

kdschlosser, 2024-02-12:
Now, I did fix it. It is not the most ideal way of going about it as it could have the potential to cause issues later on, or maybe it won't.... IDK.

I can tell you that the fix corrected more than just this one problem. It corrected a very large amount of missing buts and pieces from the binding. It added almost 2000 lines worth of missing pieces to the generated C code.

kdschlosser, 2024-02-12:
Here is a screen shot of the before and after comparison. If you look  at the vertical bars on either side this shows you where the differences are

[image]

the before is 37722 lines of generated code and the after is 39508 lines. 

Here are the before and after files.

[lv_mpy.zip](https://github.com/lvgl/lvgl/files/14251347/lv_mpy.zip)

kisvegabor, 2024-02-13:
Cool! Please send a Pull request to the binding repo.

kisvegabor, 2024-03-03:
What is the status of this issue?

kdschlosser, 2024-03-04:
The problem still exists. I am working on getting my port of the binding running with the changes I made to fix this issue. I want to make sure it doesn't cause any other problems. From what I can see it looks like it actually fixes some issues and doesn't create any problems.

I can submit a PR to the micropython binding if you like. Ultimately we need to trigger some events in the test case for LVGL to make sure this kind of a problem gets caught in the future. I should be able to use `lv.event_send(obj, lv.EVENT.CLICKED, None)` to push an event to an object that has a registered callback and that is all should need to be added to the test case is the callback and that function call then increment the tick and call the task handler. inside the callback it will set a global flag that can be checked to ensure it worked.

I will make the changes to the test as well. Lets see if it actually fails. It should, but for shits and giggles lets check anyhow.

kdschlosser, 2024-03-04:
OK i submitted a PR for the tests. it is #5787

kdschlosser, 2024-03-04:
Yup it's failing the tests. I will make the PR for the MicroPython port.

kdschlosser, 2024-03-04:
PR has been made to lv_binding_micropython

https://github.com/lvgl/lv_binding_micropython/pull/335

closed 2024-03-19 by kdschlosser

xlla, 2024-09-01:
I was encounter this issue again ,  exactly error message as  lvgl-micropython/lvgl_micropython#1 

I see many symbols begin with `_lv_` was changed to `lv_` , 

in misc/lv_types.h
<pre>
struct _lv_event_t;
typedef struct _lv_event_t lv_event_t;
</pre>
change to 

<pre>
typedef struct lv_event_t lv_event_t;
</pre>

will that cause the script lvgl-micropython/lib/lv_bindings/gen/gen_mpy.py failure to generate callback function pointer for lv?