lvgl/lvgl #5636: Label text doesn't wrap according to `max_width`

romdt, 2024-02-11:
### LVGL version

v9.0.0

### What happened?

Expected the text to wrap according to `lv_obj_set_style_max_width` constraint

[image]

### How to reproduce?

```c
lv_obj_t* text_new(lv_obj_t* parent, const char* txt)
{
    lv_obj_t* text = lv_label_create(parent);
    lv_label_set_text_static(text, txt);
    lv_obj_set_style_text_color(text, lv_color_white(), 0);
    lv_obj_set_style_bg_color(text, lv_color_black(), 0);
    lv_obj_set_style_pad_all(text, 10, 0);
    lv_obj_set_style_bg_opa(text, LV_OPA_COVER, 0);
    lv_obj_set_style_text_align(text, LV_TEXT_ALIGN_CENTER, 0);

    return text;
}

void text_doesnt_wrap_on_max_width(lv_obj_t* parent)
{
    lv_obj_t* root = lv_obj_create(lv_screen_active());
    lv_obj_set_size(root, lv_pct(100), lv_pct(100));
    lv_obj_set_flex_flow(root, LV_FLEX_FLOW_COLUMN);
    lv_obj_set_flex_align(root, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);
    lv_obj_set_style_pad_gap(root, 10, 0);

    // short text 
    lv_obj_t* one = text_new(root, "1111");

    // set manual width
    lv_obj_t* two = text_new(root, "2222222222222222");
    lv_obj_set_width(two, 100);

    // expected text to be wrapped, similar to `two` 
    lv_obj_t* three = text_new(root, "3333333333333333");
    lv_obj_set_style_max_width(three, 100, 0);
}
```

C47D, 2024-02-12:
Hi,

I can replicate the issue. Should the `3333333333333333` background also padded by 10 on all directions isn't? Because it isn't at the right and left. It seems to be missing six `3's`

kisvegabor, 2024-02-12:
Hi,

> What is the max_width property used for when it comes to labels?

It should have worked as you have expected. I've opened #5644 to fix this.

closed 2024-02-13 by kisvegabor