lvgl/lvgl #5642: RunThread stack size with FreeRTOS

Felix-devel, 2024-02-12:
### LVGL version

V9.0.0

### What happened?

The stack used in the file lv_draw_sw.c in the function lv_draw_sw_init is 8*1024. FreeROTS uses this value and multiplies it with the word size (which could be more than one byte). In my case it would use 32kbyte. Is this correct or should it be only 8kbytes.

Felix-devel, 2024-02-13:
Ok, but than [here ](https://github.com/lvgl/lvgl/blob/e6dc2f05d239b914617f19e1db019ebb35d83449/src/osal/lv_freertos.c#L76) it should be divided by sizeof(StackType_t) or sizeof(portSTACK_TYPE).

kisvegabor, 2024-02-13:
You are right! Could you send a PR?

Felix-devel, 2024-02-13:
It was my first PR. I hope I did it correct

closed 2024-02-14 by FASTSHIFT

kisvegabor, 2024-02-14:
You did, thank you! :heart:

PR #5647 (2024-02-14): fix(FreeRTOS): Stacksize calculation
fixes #5642

### Description of the feature or fix

Stack size for FreeRTOS Task was to high if StackType_t != byte

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https [... cut]