lvgl/lvgl #5645: Software rotation for the Linux framebuffer

Johennes, 2024-02-12:
### Introduce the problem

The `sw_rotate` parameter from v8 doesn't exist anymore on `master` and not all framebuffer kernel drivers appear to support hardware rotation. So this is a needed feature.

### Proposal

The docs say

> In lack of hardware display rotation support `lv_draw_sw_rotate` can be used to rotate the buffer in the `flush_cb`.

However, I'm struggling with how to use this function correctly and it doesn't appear to be used in any of the other drivers yet. Most importantly, the function seems to rotate the whole buffer but `flush_cb` is called with a subarea. Can somebody advise how to implement software rotation correctly?

Johennes, 2024-02-15:
What I got to work so far is draw from `color_p` into a temporary buffer and then use `lv_draw_sw_rotate` to rotate the pixels onto the framebuffer.

This only works when rotating by 180 degrees though. With 90 or 270 degrees, I'm hitting segfaults or all garbled output.

It also feels slightly inefficient to rotate the entire buffer and not just the rect that was drawn into on every flush.

```
--- a/src/drivers/display/fb/lv_linux_fbdev.c
+++ b/src/drivers/display/fb/lv_linux_fbdev.c
@@ -249,8 +251,11 @@ void lv_linux_fbdev_set_force_refresh(lv_display_t * disp, bool enabled)
  *   STATIC FUNCTIONS
  **********************/
 
+uint8_t * dest = NULL;
+
 static void flush_cb(lv_display_t * disp, const lv_area_t * area, uint8_t * color_p)
 {
+
     lv_linux_fb_t * dsc = lv_display_get_driver_data(disp);
 
     if(dsc->fbp == NULL ||
@@ -266,6 +271,14 @@ static void flush_cb(lv_display_t * disp, const lv_area_t * area, uint8_t * colo
     uint32_t fb_pos = color_pos + dsc->vinfo.yoffset * dsc->finfo.line_length;
 
     uint8_t * fbp = (uint8_t *)dsc->fbp;
+
+    int32_t hor_res = lv_display_get_horizontal_resolution(disp);
+    int32_t ver_res = lv_display_get_vertical_resolution(disp);
+    if (!dest) {
+        dest = malloc(hor_res * ver_res * px_size * sizeof(uint8_t));
+    }
+    fbp = dest;
+
     int32_t y;
     if(LV_LINUX_FBDEV_RENDER_MODE == LV_DISPLAY_RENDER_MODE_DIRECT) {
         for(y = area->y1; y <= area->y2; y++) {
@@ -282,6 +295,8 @@ static void flush_cb(lv_display_t * disp, const lv_area_t * area, uint8_t * colo
         }
     }
 
+    lv_draw_sw_rotate(dest, dsc->fbp, hor_res, ver_res, hor_res * px_size, ver_res * px_size, lv_display_get_rotation(disp), lv_display_get_color_format(disp));
+
     if(dsc->force_refresh) {
         dsc->vinfo.activate |= FB_ACTIVATE_NOW | FB_ACTIVATE_FORCE;
         if(ioctl(dsc->fbfd, FBIOPUT_VSCREENINFO, &(dsc->vinfo)) == -1) {
```

kisvegabor, 2024-02-16:
Hi,

It's great that you have opened this issue and we can talk about this topic. 

To be honest I haven't played with display rotation that much on PC yet, so there were a few things for me to understand (and fix).

Most importantly: what should happen when the display is rotated? (I've used the SDL driver) What I concluded is that the size of the window shouldn't change. So if you create a 320x240 window and rotate the display by 90° you still should have 320x240 window on the monitor, but the content should be rotated. It's more obvious with linux/fb where you can't set the resolution of the display. So it should look something like this:
[image]

It's important that `lv_display_get_horizontal/vertical_resolution()` returns the rotated values. That is it swaps hor_res and ver_res on 90° and 270° rotations. Therefore we need to use the pure resolutions from `disp->hor_res`, `disp->ver_res`.

Note 2) The area should be rotated too. It was like 10 minutes of error and trial to figure out what becomes what after the rotation. 

I've opened #5669 with my changes to the SDL driver and some fixes in LVGL in general. It supports only 90° now,  but if you say it's good I can finish it.

Johennes, 2024-02-16:
> Most importantly: what should happen when the display is rotated? (I've used the SDL driver) What I concluded is that the size of the window shouldn't change. So if you create a 320x240 window and [...]

That sounds sensible, yes. :+1: 

> Note 2) The area should be rotated too. It was like 10 minutes of error and trial to figure out what becomes what after the rotation.
>
> I've opened https://github.com/lvgl/lvgl/pull/5669 with my [...]

This looks great, thank you! I'll comment on the PR.

Johennes, 2024-02-18:
One thought actually: Could software rotation also be implemented generically at the rendering level instead of making it each driver's responsibility? That is, could we maybe just call `lv_draw_sw_rotate` before the flush callback is invoked and pass in the properly rotated bytes so that each display driver can stay agnostic of LVGL's internal rotation?

kisvegabor, 2024-02-19:
> Could software rotation also be implemented generically at the rendering level instead of making it each driver's responsibility?

I've commented about it [here](https://github.com/lvgl/lvgl/pull/5669#discussion_r1494987004).

Johennes, 2024-02-20:
Ok, got it. That makes sense. :+1:

Johennes, 2024-03-06:
This will be fixed with https://github.com/lvgl/lvgl/pull/5703.

kisvegabor, 2024-03-06:
Lets continue in the PR.

closed 2024-03-06 by kisvegabor