lvgl/lvgl #5648: Canvas set buffer with image

Chineko-Koizumi, 2024-02-13:
### LVGL version

v9.0.0

### What happened?

Image generated via https://lvgl.io/tools/imageconverter_v9 LVGL website .c file generator is in invalid format or function void lv_canvas_set_buffer(lv_obj_t * obj, void * buf, int32_t w, int32_t h, lv_color_format_t cf); does not support buffers generated via this website. Below photo of how it looks like. 
[image]

### How to reproduce?

```

//WiFiSignal4_map is array of uint8_t generated from 40x40 png image to c file as LV_COLOR_FORMAT_ARGB8888

//---------------------------------------------------------
    m_pCanvas = lv_canvas_create(lv_screen_active());
    lv_obj_set_pos(m_pCanvas, posX, posY);
    lv_canvas_set_buffer(m_pCanvas, reinterpret_cast<void*>(const_cast<uint8_t*>(WiFiSignal4_map)), 40, 40, LV_COLOR_FORMAT_ARGB8888 );
//---------------------------------------------------------    
    
    
```

FASTSHIFT, 2024-02-14:
Have you tried using the image widget?

```c
LV_IMG_DECLARE(WiFiSignal4);
lv_obj_t* img = lv_image_create(lv_screen_active());
lv_image_set_src(img, &WiFiSignal4);
```

Chineko-Koizumi, 2024-02-14:
@FASTSHIFT Thank you, it works as a lv_image so lv_canvas_set_buffer is bugged.

FASTSHIFT, 2024-02-15:
> @FASTSHIFT Thank you, it works as a lv_image so lv_canvas_set_buffer is bugged.

My side is working normally:
[image]

Please ensure that the width, height, and color formats are set correctly, or try to reproduce it on the [simulator](https://github.com/lvgl/lv_port_pc_eclipse), or provide complete reproduction code.

Chineko-Koizumi, 2024-02-15:
Here is reproduction code, already changed from canvas to img:
https://github.com/Chineko-Koizumi/ESP32_S3_multipurpose_device/blob/master/include/Display/DisplayCanvasSequence.h
https://github.com/Chineko-Koizumi/ESP32_S3_multipurpose_device/blob/master/src/Display/DisplayCanvasSequence.cpp line 14 and 45

My setup is ESP32 S3 and ili9341 TFT display connected via SPI

FASTSHIFT, 2024-02-17:
The color format generated by the image resource is `LV_COLOR_FORMAT_RGB565A8`:
```c
lv_image_dsc_t WiFiSignal1 = {
  {
    .magic = LV_IMAGE_HEADER_MAGIC,
    .cf = LV_COLOR_FORMAT_RGB565A8,
    .w = 40,
    .h = 40,
  },
  .data_size = 1600 * 3,
  .data = WiFiSignal1_map,
};
```

So the canvas needs to be set to the same color format:
```c
lv_canvas_set_buffer(m_pCanvas, reinterpret_cast<void*>(const_cast<uint8_t*>(WiFiSignal4_map)), 40, 40, LV_COLOR_FORMAT_RGB565A8);
```

It is recommended to use the image component to complete the display of image widget, so that after the modification of image resources, there is no need to update the application layer code, and there will be no consistency problems.

closed 2024-02-17 by Chineko-Koizumi