lvgl/lvgl #5649: lv_obj_align_to method BUG

mhtrbg, 2024-02-14:
### LVGL version

V9.0.0

### What happened?

Dear LVGL Development Team,

I am encountering a critical issue with the LVGL library version 9.0. Whenever I use the lv_obj_align_to method, my display crashes immediately. This issue occurs consistently only when this specific method is invoked. If I avoid using lv_obj_align_to(), everything else functions normally, including the execution of official demo applications.

### How to reproduce?

    lv_obj_t *obj1 = lv_obj_create(lv_scr_act());
    lv_obj_center(obj1);
    lv_obj_set_size(obj1, 50, 50);
    lv_obj_t *obj2 = lv_obj_create(lv_scr_act());
    lv_obj_set_size(obj2, 50, 50);
    lv_obj_align_to(obj1, obj2, LV_ALIGN_OUT_TOP_MID, 0, 0);

FASTSHIFT, 2024-02-14:
This code works well in the Simulator, have you tried replicating the problem in the [simulator](https://github.com/lvgl/lv_port_pc_eclipse)?
[image]

kisvegabor, 2024-02-15:
That code looks good to me too. 

> This issue occurs consistently only when this specific method is invoked.

That's great! What platform do you use? Do you have access to a debugger? If not please [enable LVGL's logging](https://docs.lvgl.io/master/porting/log.html) as described here with `TRACE` level.

closed 2024-03-09 by lvgl-bot