lvgl/lvgl #5650: 请问能否关闭lvgl控件的动画效果？

Szeroy, 2024-02-14:
### Problem to solve

我把lvgl用在显示墨水屏上，因为墨水屏的刷新率很慢，局部刷新也要300ms，因为控件会有动画效果，显得十分的卡顿，请问lvgl有没有提供选项把控件的动画效果关闭，直接默认显示最后的结果，比如选项框打勾，直接忽略掉中间的动画效果，显示最后的勾呢？

### Success criteria

无

### Solution outline

无

### Rabbit holes

无

### Testing

无

### Teaching

无

### Considerations

无

Szeroy, 2024-02-14:
我看到网上有说把LV_USE_ANIMATION写成0，但是我在使用的esp32官方移植好的版本上没看到这个宏定义

FASTSHIFT, 2024-02-14:
Set `LV_THEME_DEFAULT_TRANSITION_TIME` to 0 in lv_conf.h.

The default animation style of the control is adjusted through the anim_duration in style
And search for the anim_duration field, setting its default value to 0.
[image]

In addition, it is recommended to ask questions in English so that everyone can understand the question.

Szeroy, 2024-02-15:
谢谢，我在idf里面发现有(Top) → Component config → LVGL configuration → Themes中的(80)    Default transition time in [ms]  选项，应该把这个设成0，再把默认主题动画时间设置成0就好了吧，因为没有看到LV_THEME_DEFAULT_TRANSITION_TIME选项，已经被idf改成自己的宏定义了

closed 2024-02-19 by Szeroy