lvgl/lvgl #5657: LV_COLOR_DEPTH 8 don't work

razor5k, 2024-02-15:
### LVGL version

v9.0.0

### What happened?

I tried to use LVGL with a SSD 1322 display controller. This is a 4 Bit display. To save memory, LV_COLOR_DEPTH is set to 8 but the display shows weird patterns.
If I set LV_COLOR_DEPTH 16, and convert the rgb565 to grayscale (luma), the display works but I need the double buffer space.

### How to reproduce?

//##############################################################################
//# 16 bit works
//##############################################################################

...
lv_conf.h:
#define LV_COLOR_DEPTH 16
...

...

header of driver:
#define DISPLAY_BUFFER_SIZE_LVGL (DISPLAY_WIDTH * 10u * (LV_COLOR_DEPTH / 8u))

init method of driver:

        dev->disp_buf1 = (uint8_t*)malloc(DISPLAY_BUFFER_SIZE_LVGL);
        if(dev->disp_buf1 == NULL) return -1;

        dev->disp = lv_display_create(DISPLAY_WIDTH, DISPLAY_HEIGHT);
        lv_display_set_user_data(dev->disp, dev);
        //lv_display_set_buffers(dev->disp, dev->disp_buf1, dev->disp_buf2, DISPLAY_BUFFER_SIZE_LVGL), LV_DISPLAY_RENDER_MODE_PARTIAL);
        lv_display_set_buffers(dev->disp, dev->disp_buf1, NULL, DISPLAY_BUFFER_SIZE_LVGL, LV_DISPLAY_RENDER_MODE_PARTIAL);
        lv_display_set_flush_cb(dev->disp, oled_flush_cb);
...

static void oled_flush_cb(lv_disp_t * disp, const lv_area_t * area, uint8_t * pxmap)
{
        /*Truncate the area to the screen*/
        uint16_t* buf = (uint16_t*)pxmap;
        struct display_driver_device* dev = (struct display_driver_device*) lv_display_get_user_data(disp);
        uint16_t color;
        uint16_t gray;
        uint16_t grayscale;
        uint16_t r;
        uint16_t g;
        uint16_t b;

        //Write to frame buffer
        for(int32_t y = area->y1; y <= area->y2; y++) {
                for(int32_t x = area->x1; x <= area->x2; x++) {
                        //Luma grayscale (values are not float, so factors are multiplicated with 256)
                        color = *buf;
                        r = ((color >> 11) & 0x1f) * 77;
                        g = ((color >> 5) & 0x3f) * 151;
                        b = (color & 0x1f) * 28;
                        //We've to devide green by 2 because it has 6 bit but other colors have only 5 bit
                        //Gray means all colors have the same value.
                        gray = (r + (g >> 1) + b) >> 8;
                        //Green will be multiplicated by 2 (shift left 6 instead of 5)
                        grayscale = (gray << 11) + (gray << 6) + gray;
                        //make 16 bit grayscale to 4 bit by grayscale >> 12
                        display_driver_draw_pixel(dev, x, y, grayscale >> 12);
                        buf++;
                }
        }

        //flush frame buffer to display only once if all parts are flushed
        if(lv_display_flush_is_last(disp)) {
                display_driver_update_display(dev);
        }

        //Inform the graphics library that you're ready with the flushing
        lv_display_flush_ready(disp);
}

//##############################################################################
//# 8 bit doesn't work
//##############################################################################

...
lv_conf.h:
#define LV_COLOR_DEPTH 8
...

...

header of driver:
#define DISPLAY_BUFFER_SIZE_LVGL (DISPLAY_WIDTH * 10u * (LV_COLOR_DEPTH / 8u))

init method of driver:

        dev->disp_buf1 = (uint8_t*)malloc(DISPLAY_BUFFER_SIZE_LVGL);
        if(dev->disp_buf1 == NULL) return -1;

        dev->disp = lv_display_create(DISPLAY_WIDTH, DISPLAY_HEIGHT);
        lv_display_set_user_data(dev->disp, dev);
        //lv_display_set_buffers(dev->disp, dev->disp_buf1, dev->disp_buf2, DISPLAY_BUFFER_SIZE_LVGL), LV_DISPLAY_RENDER_MODE_PARTIAL);
        lv_display_set_buffers(dev->disp, dev->disp_buf1, NULL, DISPLAY_BUFFER_SIZE_LVGL, LV_DISPLAY_RENDER_MODE_PARTIAL);
        lv_display_set_flush_cb(dev->disp, oled_flush_cb);
        lv_display_set_color_format(dev->disp, LV_COLOR_FORMAT_L8); //Is L8 correct?
...

static void oled_flush_cb(lv_disp_t * disp, const lv_area_t * area, uint8_t * pxmap)
{
        /*Truncate the area to the screen*/
        uint8_t* buf = pxmap;

        struct display_driver_device* dev = (struct display_driver_device*) lv_display_get_user_data(disp);

        for(int32_t y = area->y1; y <= area->y2; y++) {
                for(int32_t x = area->x1; x <= area->x2; x++) {
                        //Convert 8 bpp to 4bpp by shifting right
                        display_driver_draw_pixel(dev, x, y, (*buf) >> 4);
                        buf++;
                }
        }

        //flush frame buffer to display only once if all parts are flushed
        if(lv_display_flush_is_last(disp)) {
                display_driver_update_display(dev);
        }

        //Inform the graphics library that you're ready with the flushing
        lv_display_flush_ready(disp);
}

MountainRipper, 2024-02-18:
same problem~~~A8 format is black

kisvegabor, 2024-02-19:
L8 and A8 rendering still needs to be added. We need it for many things. I'll take care of that this week!

llisauskas, 2024-02-29:
@kisvegabor on a bit (un)related note - and what about rgb332? Was it deprecated in v9? Can't find any info about it here.

kisvegabor, 2024-03-05:
> and what about rgb332? Was it deprecated in v9? Can't find any info about it here.

Yes, I should have mention it in the changelog. A note is added in #5796.

FYI, @zjanosy is working on the L8 rendering. It seems we can have a PR by Friday.

kisvegabor, 2024-03-09:
The PR is opened: https://github.com/lvgl/lvgl/pull/5800

I couldn't review it yet, but any feedback is welcome.

razor5k, 2024-03-11:
The changes from zjanosy works now with my display driver :) I use a SSD1322 display with 4 bit grayscale and the the flush function is the code snippet "8 bit doesn't work" described above which works now. The 8 bit grayscale is converted to 4 bit by shifting right 4 times. Because of the L8 implementation I can save half of my graphics memory, thank you very much for this very fast implementation.

kisvegabor, 2024-03-12:
Amazing :tada: 
Thank you for the feedback!

zjanosy, 2024-04-02:
Currently finalizing AL88 (monochrome with alpha channel) support.

kisvegabor, 2024-04-29:
https://github.com/lvgl/lvgl/pull/5800 is merged for L8 and AL88 support so I close this issue.

jareddlc, 2024-05-27:
@razor5k I'm wrote my own display driver for SSD1322, but i have an issue when using render mode partial, it appears incorrectly, but render mode full works fine. Do you think you could share the code for your flush?

jareddlc, 2024-05-28:
here is my flush command

```
void my_disp_flush(lv_display_t *disp, const lv_area_t *area, uint8_t *px_map) {
  // send commands to OLED
  oled.set_column_address(0x1C + (area->x1 / 4), 0x1C + (area->x2 / 4));
  oled.set_row_address(0x00 + area->y1, area->y2);
  oled.set_write_ram();

  for (int y = area->y1; y <= area->y2; y++) {
    for (int x = area->x1; x <= area->x2; x += 4) {
      for (uint8_t i = 0; i < 2; i++) {
        uint8_t a = *px_map >> 4;
        px_map++;
        uint8_t b = *px_map >> 4;
        px_map++;
        oled.send_sd1322_data(a << 4 | b);
      }
    }
  }

  // ready to flush
  lv_display_flush_ready(disp);
}
```

kisvegabor, 2024-05-28:
Please send a photo about the issue.

razor5k, 2024-05-28:
Hi @jareddlc,
it's important to align the area which should be displayed. This is needed because you can't read back pixels between a segment. So you need to align x1 to 0, 4, 8 etc. and x2 to 3, 7, 11 etc. You need to add a callback function which makes the alignment.

Here's an example of my glue code from lvgl to the display driver:
[lvgl_ssd1322.zip](https://github.com/lvgl/lvgl/files/15469776/lvgl_ssd1322.zip)

jareddlc, 2024-05-30:
@razor5k thank you, i will take a look at the provided code and reply back with my findings

@kisvegabor sorry, i'm not having an issue specifically with the color depth. Sorry to post my question here.

jareddlc, 2024-05-30:
Full render

[image]

Partial render

[image]

razor5k, 2024-05-31:
Because of (area->x1 / 4) and (area->x2 / 4)) only every 4th pixel can be addressed. If full rendering is used, the whole display will be refreshed which is always aligned to every 4th pixel.
Partial rendering needs the alignment because of the division by 4. Hence you need to register a callback function which triggers on LV_EVENT_INVALIDATE_AREA event. In this function the alignment must be done.

jareddlc, 2024-06-02:
@razor5k thank you! with your code i was able to get the partial rendering working!

razor5k, 2024-06-03:
@jareddlc I'm glad to hear it works 👍 Partial rendering is much faster if you render small animations.

kisvegabor, 2024-06-05:
As the issue seems to be resolved, I close it.

closed 2024-06-05 by kisvegabor

jareddlc, 2024-09-08:
@razor5k I've made a repo with the code for the SSD1322 for use with lvgl here: https://github.com/jareddlc/SSD1322

I would like to make this so that it can be directly added as a component for the espidf

kisvegabor, 2024-09-12:
I think you should open an issue for this in the ESP-IDF repo.