lvgl/lvgl #5665: Permanently align/anchor 2 objects (w/o parent-child relationship)

valeriosetti, 2024-02-16:
### Problem to solve

If a child is aligned to its parent with `lv_obj_align()`, then if the parent is moved or resized the child is moved accordingly. However if the 2 objects are non parent-child and they are aligned/anchored with `lv_obj_align_to()`, then if one of the 2 moves[^1] then the other one remains unchanged.

This is clearly [reported in the official documentation](https://docs.lvgl.io/8.3/overview/coords.html#align), but it would be nice to fix it.

[^1]: for example during an animation.

### Success criteria

When 2 objects are anchored each other with `lv_obj_align_to()` then if one of the two is moved the other must follow it.

### Solution outline

No idea for now, sorry.

### Rabbit holes

Unfortunately I never contributed to LVGL yet, but I guess that the difference between parent-child changes and non parent-child ones is that in the first case we have to worry about one possible change (i.e. the parent is changed so the child must be updated) whereas in the latter case there is a double relationship (i.e. if any of the 2 is moved, then the other one should be updated).
Moreover in the 2nd case attention should be paid to the fact that this might create a loop, i.e. I move object 1, so I need to update object 2, but hey object 2 is anchored to object 1 so I need to update it, and so forth.

### Testing

Empty window with 2 objects anchored together + an animation moving one of the two. The 2nd object should just follow the 1st one.

### Teaching

Remove this sentence from the documentation:

> Note that, unlike with lv_obj_align(), lv_obj_align_to() can not realign the object if its coordinates or the reference object's coordinates change.

kisvegabor, 2024-02-19:
I agree that it's a real problem that we should think about.

We can simplify a little bit what needs to be updated and when:
1. The parent size changes: update all children
2. The parent position changes: update all children
3. The child size changes: update only that child
4. The child position changes: do nothing.

It's an 1 to N relationship, that is a parent can have any number of children. Storing this kind of parent is possible in each widget (it's just a single pointer). We can even have a style property like `align_parent`. It's NULL by default, meaning the real parent. 

We also need to manage deletions. So if a parent is deleted, it needs to remove itself from all children's pointers. However if `align_parent` is a style property it's not trivial from which style shall we remove the parent. Maybe it doesn't need to be a style property. :slightly_smiling_face: 

Storing all a reference to all children in each parent doesn't sound that good. 

We should also consider how it cooperates with other layouts. Let's say we have a container with a grid layout, we still might want to add a label above the container using `align_to`. However the children cannot be part of a grid or flex layout, as they were overwrite its position. 

I have some performance concerns too. Let's say the parent is a list item in a long list and the list can be scrolled. On scroll we need to check all list items' and all their children if they have an anchored widget which needs to be scrolled.

Sorry, for mentioning only problems, however these are the main concerns why it is not implemented yet. I hope that we can figure out something together.

valeriosetti, 2024-03-12:
Since I cannot remove the `stale` label I'll add my comment. I'm investigating on this on my spare time, but unfortunately is not as much as I would like it to be, so it might take some time before I get to some working solution.

closed 2024-06-20 by kisvegabor