lvgl/lvgl #5668: Crash with LV_DISPLAY_RENDER_MODE_PARTIAL When Creating OSD UI on LVGL 9.0.0

JLJimmyH, 2024-02-16:
### LVGL version

v9.0.0

### What happened?

**Version:** LVGL 9.0.0

**Description:** I'm trying to create an OSD UI with LVGL to overlay on top of my other screens. I've been using LV_DISPLAY_RENDER_MODE_PARTIAL to render my display with the following settings:

```
g_pixel_buffer_size = hor_res * 10 * sizeof(lv_color32_t);
g_pixel_buffer = lv_malloc(g_pixel_buffer_size);
...
lv_obj_set_style_bg_opa(lv_scr_act(), LV_OPA_TRANSP, 0);
lv_obj_set_style_bg_opa(lv_layer_bottom(), LV_OPA_TRANSP, 0);
lv_display_set_color_format(g_disp, LV_COLOR_FORMAT_NATIVE_WITH_ALPHA);
...
lv_display_set_flush_cb(g_disp, flush_cb);
lv_display_set_buffers(g_disp, g_pixel_buffer, NULL, g_pixel_buffer_size, LV_DISPLAY_RENDER_MODE_PARTIAL);
```
However, I've encountered a crash as soon as the program starts. Through testing, I found that normal rendering occurs under the following circumstances:

1. Switching to LV_DISPLAY_RENDER_MODE_DIRECT.
2. Setting LV_DISPLAY_RENDER_MODE_PARTIAL with g_pixel_buffer_size = hor_res * ver_res * sizeof(uint32_t)
3. Using lv_disp_set_color_format(g_disp, LV_COLOR_FORMAT_NATIVE); however, this results in a solid background, which I cannot use as an OSD to overlay on other screens.

Due to the performance constraints of embedded devices, we cannot use DIRECT MODE or set g_pixel_buffer_size to the size of the entire screen...

If there is any mistake in my settings, please let me know. Thank you.

JLJimmyH, 2024-02-19:
Update for my test:

I encountered an issue with incorrect buffer data received within the flush_cb function when using transparent configurations and the LV_COLOR_FORMAT_NATIVE_WITH_ALPHA setting.

Observations:
Both test using #define LV_COLOR_DEPTH 32 and LV_DISPLAY_RENDER_MODE_PARTIAL.

Same configuration, with transparency:

Active screen and bottom layer background opacities set to transparent (LV_OPA_TRANSP).
Color format set to LV_COLOR_FORMAT_NATIVE_WITH_ALPHA.
Result: Incorrect buffer received in flush_cb
[image]

Same configuration, without transparency:
( I think the color format is LV_COLOR_FORMAT_NATIVE )
Transparency settings commented out.
Result: Correct buffer received in flush_cb.
[image]

Request for assistance:

I'd appreciate any insights or suggestions to address this issue.

kisvegabor, 2024-02-20:
You need to enable `LV_COLOR_SCREEN_TRANSP` in `lv_conf.h`

JLJimmyH, 2024-02-21:
Thank you. I noticed in the v8.3 branch that LV_COLOR_SCREEN_TRANSP is available, but I couldn't find this flag in the lv_conf.h of version 9.0. It appears it may have been removed or changed in the latest version.

kisvegabor, 2024-02-22:
In v9 you can simple set `lv_display_set_color_format(display, LV_COLOR_FORMAT_ARGB8888)`.

JLJimmyH, 2024-02-22:
Yes I already set this.
```lv_display_set_color_format(display, LV_COLOR_FORMAT_ARGB8888);```
it's work in display LV_DISPLAY_RENDER_MODE_DIRECT
However when i using LV_DISPLAY_RENDER_MODE_PARTIAL.
it get the wrong buffer in flush_cb

kisvegabor, 2024-02-24:
I've really found a bug with partial rendering in ARGB8888. See the fix here: #5741

JLJimmyH, 2024-02-26:
Thank you so much @kisvegabor it's work now.
I'll close the issue

closed 2024-02-26 by JLJimmyH

kisvegabor, 2024-02-27:
Amazing!