lvgl/lvgl #5674: will lvgl v10 add page manager Component?

FangXS-github, 2024-02-17:
### Introduce the problem

I see that everyone has their own set of management interface methods when using lvgl, which seems to be inconsistent. There are also some differences in performance and user experience. Will LVGL add a page management component as part of LVGL in future versions?

### Proposal

For example, the pagemanager of [FASTSHIFT: X-TRACK](https://github.com/FASTSHIFT/X-TRACK)?

kisvegabor, 2024-02-19:
Hi,

I'm not entirely sure what do you mean by page manger. Could elaborate on that?

FangXS-github, 2024-02-20:
> Hi,
> 
> I'm not entirely sure what do you mean by page manger. Could elaborate on that?

The page manager is used to manage the lifecycle of each page that needs to be displayed. includes Multi level page switching，Cross page switching，Page transition animation，You can specify whether to delete the current page when switching pages.....

FangXS-github, 2024-02-20:
> Hi,
> 
> I'm not entirely sure what do you mean by page manger. Could elaborate on that?

[FASTSHIFT: X-TRACK](https://github.com/FASTSHIFT/X-TRACK)used stack as the data structure of page manager, The feature of stack first in, last out is very suitable for page switching, page management is an important module in LVGL, responsible for managing and displaying various elements on the page.

kisvegabor, 2024-02-20:
If [fragment](https://docs.lvgl.io/master/others/fragment.html) something that you are looking for?

closed 2024-02-21 by FangXS-github