lvgl/lvgl #5678: LVGL9 lv_obj_remove_flag() issue

KenVanHoeylandt, 2024-02-18:
### LVGL version

v9.0.0

### What happened?

I'm upgrading LVGL from `v8.3` to `v9.0`.

There seems to be a behaviour change when calling `lv_obj_remove_flag()` (formerly `lv_obj_clear_flag()`).
Let's take a look at this code:

```c
lv_obj_set_style_img_recolor(image, lv_color_white(), 0);
lv_obj_set_style_img_recolor_opa(image, 255, 0);
lv_img_set_src(image, icon->image);
lv_obj_remove_flag(image, LV_OBJ_FLAG_HIDDEN); // <-- problematic
```
The main intention of that problematic line of code is that I want to show/hide widgets.
The code above is called from a custom widget's event callback for `LV_EVENT_DRAW_MAIN`.

Before v9, this was not a problem, but now with v9 it triggers an assert with the following callstack:
- ` LV_ASSERT_MSG(!disp->rendering_in_progress, "Invalidate area is not allowed during rendering.");`
- `_lv_inv_area`
- `lv_obj_invalidate_area`
- `lv_obj_invalidate`

There seems to be no documentation on how to change existing code:
https://github.com/lvgl/lvgl/blob/master/docs/CHANGELOG.rst

So it seems that either some documentation is missing on this behaviour change, or there is an unintended change.

XuNeo, 2024-02-18:
The assertion means you are modify widget properties during rendering. It will generate new dirty areas thus it's forbidden.

Could you check where are you calling those functions? Are you calling it in DRAW_EVENT?
> 
```c
lv_obj_set_style_img_recolor(image, lv_color_white(), 0);
lv_obj_set_style_img_recolor_opa(image, 255, 0);
lv_img_set_src(image, icon->image);
lv_obj_remove_flag(image, LV_OBJ_FLAG_HIDDEN); // <-- problematic
```

KenVanHoeylandt, 2024-02-18:
Yep, as I mentioned:
> The code above is called from a custom widget's event callback for LV_EVENT_DRAW_MAIN.

I understand why it's a problem and I have a solution, but I kinda expected it to be in the changelog / migration guide.

kisvegabor, 2024-02-19:
It was not by design that it was working in v8, only pure luck :slightly_smiling_face: 
So I don't think it needs be highlighted in the changelog as it was never a recommended to add any modification during drawing.

C47D, 2024-02-19:
@kisvegabor maybe we can add that explicitly 

> Updating widgets during rendering is not supported.

KenVanHoeylandt, 2024-02-20:
> It was not by design that it was working in v8, only pure luck 🙂 So I don't think it needs be highlighted in the changelog as it was never a recommended to add any modification during drawing.

Thanks!

closed 2024-02-20 by KenVanHoeylandt

kisvegabor, 2024-02-22:
> Updating widgets properties during rendering is not supported.

Where can be a good place for that?

kisvegabor, 2025-01-30:
@vwheeler63 Adding a line of warning there would be helpful for sure.

vwheeler63, 2025-01-30:
> [@vwheeler63](https://github.com/vwheeler63) Adding a line of warning there would be helpful for sure.

Done in PR #7685.  Doc modification shown in the [Base Widget > Events > Drawing Events section](http://crystal-clear-research.com/for_gabor/demo/docs.lvgl.io/master/details/base-widget/event.html#drawing-events).

PR #7685 (2025-02-07): fix(docs):  add warning not to modify Widgets during draw events.
Resolves #5678

See generated result in [Base Widget > Events > Drawing Events section](http://crystal-clear-research.com/for_gabor/demo/docs.lvgl.io/master/details/base-widget/event.html#drawing-events).

cc:  @kisvegabor 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.  Done.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.  n/a
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.  n/a
- If you added new options  [... cut]