lvgl/lvgl #5680: Usage of stdint, stddef, ...

MiSimon, 2024-02-19:
### Introduce the problem

In some header and source files there are includes for stdint, stddef, ... . In some environments these headers do not exist. 

It is possible to produce simple versions of these headers that include everything needed for lvgl. But since the includes happen in the headers and not only inside source files, these headers have to be provided with lvgl.h and the other headers.

### Proposal

I suggest to create lv_xxx headers for them (like lv_string.h) and use these in the header and source files. Inside this headers a define like LV_USE_OS can be used to either include the existing headers or to include specific headers.

An example for lv_stdint.h:

```
#if LV_USE_OS == LV_OS_CUSTOM
  #include LV_OS_CUSTOM_STDINT_INCLUDE
#else
  #include <stdint.h>
#endif
```

With defines in lv_conf.h:

```
#if LV_USE_OS == LV_OS_CUSTOM
    #define LV_OS_CUSTOM_STDINT_INCLUDE <my_stdint.h>
    #define LV_OS_CUSTOM_STDDEF_INCLUDE <my_stddef.h>
    #define LV_OS_CUSTOM_STDBOOL_INCLUDE <my_stdbool.h>
#endif
```

kisvegabor, 2024-02-19:
I see the problem. Just out of curiosity, do you develop an UEFI bios?

MiSimon, 2024-02-19:
Yes indeed, I use LVGL with UEFI for a couple of years now. 
Till now i used it in a company project with other ported libraries, to make porting easier there is a UEFI version of the standard library. 
Now i want to use it in a private project where i do not need/want an UEFI version of the standard library to keep it simple. The goal would be a driver like the windows/x11/... drivers integrated in LVGL.

MouriNaruto, 2024-02-20:
I also hope to have that feature.

In the current stage, I have made a minimal C runtime (My public but old implementation at https://github.com/ProjectMile/Mile.Uefi/tree/main/Mile.Uefi.RuntimeLibrary/Include) for LVGL to workaround that issue, lol.

Kenji Mouri

kisvegabor, 2024-02-22:
@MouriNaruto 
Would this on its own solve your issue too?
```
#if LV_USE_OS == LV_OS_CUSTOM
    #define LV_OS_CUSTOM_STDINT_INCLUDE <my_stdint.h>
    #define LV_OS_CUSTOM_STDDEF_INCLUDE <my_stddef.h>
    #define LV_OS_CUSTOM_STDBOOL_INCLUDE <my_stdbool.h>
#endif
```

MouriNaruto, 2024-02-22:
@kisvegabor 

I think the style should be OK.

[image]
I think these headers need to be used according to my experience for using LVGL in UEFI. Maybe will add more header file custom macros.

Kenji Mouri

MiSimon, 2024-02-26:
Shall the lv_.* includes be used in every file in the src directory?

I think most of the drivers and draw methods are platform specific and may use stdint,.. directly. Whereas widgets, libs and the other core files should be platform independent and use only lv_.* headers and lv_xxx wrapper functions (lv_strlen, lv_malloc, ...).

MouriNaruto, 2024-02-26:
> Whereas widgets, libs and the other core files should be platform independent and use only lv_.* headers and lv_xxx wrapper functions (lv_strlen, lv_malloc, ...).

I agree with that.

Kenji Mouri

kisvegabor, 2024-02-27:
> Shall the lv_.* includes be used in every file in the src directory?

In the [libs](https://github.com/lvgl/lvgl/tree/master/src/libs) we have integrated into LVGL we can easily change the defines too.

> I also agree with that. 

Do you have time/interest to send a  Pull request with the discussed changes?

MouriNaruto, 2024-02-27:
> Do you have time/interest to send a Pull request with the discussed changes?

I'm sorry for that because I’m busy on maintaining other open-source projects (for example, NanaZip) created by me in the current stage. (Many people asked me when the new NanaZip 3.0 release will be came out. I felt stressed. I have delayed that for at least one year, I don't think I should delay that anymore, lol.)

I don't feel shame with that words because I finally pushed Windows backend to LVGL mainline. I have finished the promise that LVGL v9 will have a new Windows backend before. I need some break to process other projects I created or maintained. 

Of course, the improvement and bugfixes work for LVGL Windows backend will not be affected because I always think maintaining open-source projects is a type of giving serious promise to users.

In recent days, I have added Windows dynamic and static library versions of LVGL to the LVGL Visual Studio repository. Some Windows desktop developers tried to use LVGL to develop installers for their Windows desktop apps because LVGL is lightweight. I also need to find some insights and do some experiments to help me find the meaning of maintaining for LVGL. For example, I will try to experiment that using LVGL v9 in NanaZip Classic, which is for classic Windows that don't have XAML Islands.

Kenji Mouri

MiSimon, 2024-02-28:
I created the pull request turned out it was not that much work. 
I removed includes for string and stdlib from the headers and added them to the platform specific files if they needed them.

kisvegabor, 2024-03-03:
Thank you for opening the PR. Let's continue there.

kisvegabor, 2024-03-03:
@MouriNaruto First of all congratulations to NanaZip. It seems very popular. It must be difficult to juggle between the maintenance of two projects. 

If LVGL can be of any help for NanaZip, just let me know either here or in private at gabor@lvgl.io.