lvgl/lvgl #5709: Arduino compiler error when using Canvas - designator order for field lv_image_header_t

cuci90, 2024-02-21:
### LVGL version

v9.0.1

### What happened?

I am using the Arduino IDE to compile the Canvas example on the documentation:

[Rectangle on canvas](https://docs.lvgl.io/master/widgets/canvas.html#draw-a-rectangle-to-the-canvas)

But the compiler throughs this error:
`error: designator order for field 'lv_image_header_t::cf' does not match declaration order in 'lv_image_header_t'`

### How to reproduce?

````
#include <lvgl.h>
#define CANVAS_WIDTH  280
#define CANVAS_HEIGHT 120
lv_layer_t layer;
lv_obj_t * canvas;

void setup() {
  // put your setup code here, to run once:
lv_setup_canvas();
}

void lv_setup_canvas(){
    /*Create a buffer for the canvas*/
    LV_DRAW_BUF_DEFINE(draw_buf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_COLOR_FORMAT_ARGB8888);

    /*Create a canvas and initialize its palette*/
    canvas = lv_canvas_create(lv_screen_active());
    lv_canvas_set_draw_buf(canvas, &draw_buf);

    lv_canvas_fill_bg(canvas, lv_color_hex3(0xb8eb), LV_OPA_COVER);
    lv_obj_set_align(canvas,LV_ALIGN_BOTTOM_MID);

    lv_canvas_init_layer(canvas, &layer);
}

void loop() {
  // put your main code here, to run repeatedly:

}

XuNeo, 2024-02-23:
Could you try using below macro instead in lv_draw_buf.h?
```c
#define LV_DRAW_BUF_DEFINE(name, _w, _h, _cf) \
    static uint8_t buf_##name[_LV_DRAW_BUF_SIZE(_w, _h, _cf)]; \
    static lv_draw_buf_t name = { \
        .header = { \
                    .magic = LV_IMAGE_HEADER_MAGIC, \
                    .cf = (_cf), \
                    .flags = LV_IMAGE_FLAGS_MODIFIABLE, \
                    .w = (_w), \
                    .h = (_h), \
                    .stride = _LV_DRAW_BUF_STRIDE(_w, _cf), \
                    .reserved_2 = 0, \
                }, \
        .data_size = sizeof(buf_##name), \
        .data = buf_##name, \
        .unaligned_data = buf_##name, \
    }

```

cuci90, 2024-02-24:
Compiling is working with your code now and I had to put` LV_COLOR_FORMAT_ARGB8888` directly after `draw_buf` to work. Thank you very much @XuNeo.
But unfortunately the canvas is not shown on the screen. Did I miss something?

```

#include <lvgl.h>
#include <SPI.h>
#if LV_USE_TFT_ESPI
#include <TFT_eSPI.h>
#endif

#define CANVAS_WIDTH  50
#define CANVAS_HEIGHT 50
SPIClass mySpi = SPIClass(VSPI);

lv_layer_t layer;
lv_obj_t * canvas;

#define TFT_HOR_RES   320
#define TFT_VER_RES   240

#define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 )
uint32_t draw_buf[DRAW_BUF_SIZE / 4];
void setup() {
  // put your setup code here, to run once:
    Serial.begin( 115200 );
       lv_init();

    lv_display_t * disp;

    disp = lv_tft_espi_create(TFT_HOR_RES, TFT_VER_RES, draw_buf, sizeof(draw_buf));

    lv_setup_canvas();

}

void lv_setup_canvas(){

      /*Create a buffer for the canvas*/
    LV_DRAW_BUF_DEFINE(draw_buf, LV_COLOR_FORMAT_ARGB8888 ,CANVAS_WIDTH, CANVAS_HEIGHT );

    /*Create a canvas and initialize its palette*/
    lv_obj_t * canvas = lv_canvas_create(lv_screen_active());
    lv_canvas_set_draw_buf(canvas, &draw_buf);

    lv_canvas_fill_bg(canvas, lv_color_hex3(0xccc), LV_OPA_COVER);
    lv_obj_center(canvas);

    lv_layer_t layer;
    lv_canvas_init_layer(canvas, &layer);

    lv_draw_rect_dsc_t dsc;
    lv_draw_rect_dsc_init(&dsc);
    dsc.bg_color = lv_palette_main(LV_PALETTE_RED);
    dsc.border_color = lv_palette_main(LV_PALETTE_BLUE);
    dsc.border_width = 3;
    dsc.outline_color = lv_palette_main(LV_PALETTE_GREEN);
    dsc.outline_width = 2;
    dsc.outline_pad = 2;
    dsc.outline_opa = LV_OPA_50;
    dsc.radius = 5;
    dsc.border_width = 3;

    lv_area_t coords = {10, 10, 40, 30};

    lv_draw_rect(&layer, &dsc, &coords);

    lv_canvas_finish_layer(canvas, &layer);
}

void loop() {
  // put your main code here, to run repeatedly:
     lv_task_handler(); /* let the GUI do its work */
    lv_tick_inc(5); /* tell LVGL how much time has passed */
    delay(5); /* let this time pass */

}

void my_disp_flush( lv_display_t *disp, const lv_area_t *area, uint8_t * px_map)
{

    lv_disp_flush_ready(disp);
}

```

XuNeo, 2024-02-26:
I'm not sure but is the function loop() called periodically? You may try adding a while(1) in loop() to keep calling 
```c
     lv_task_handler(); /* let the GUI do its work */
    lv_tick_inc(5); /* tell LVGL how much time has passed */
    delay(5); /* let this time pass *
```

cuci90, 2024-02-27:
When I use
` lv_draw_buf_create(CANVAS_WIDTH, CANVAS_HEIGHT,LV_COLOR_FORMAT_ARGB8888,0);  `  
instead of
 `LV_DRAW_BUF_DEFINE(draw_buf, LV_COLOR_FORMAT_ARGB8888 ,CANVAS_WIDTH, CANVAS_HEIGHT );`  

it is working. So it must have something to do with the way the buffer is initializalized.
On the simulator both initialization methods are working.

closed 2024-03-02 by cuci90