lvgl/lvgl #5717: SDL Keyboard lv_indev_get_key returns 0

Ka1mes, 2024-02-22:
### LVGL version

v9.0.1-dev

### What happened?

I am trying to read keyboard inputs on Windows using SDL2. The sdl_keyboard_read callback function of the SDL keyboard seems to be returning the correct keys. I am also able to use the SDL mouse without any problems.
In the code snippet below I am expecting the program to print the last key pressed on the keyboard but it will only print 0.

### How to reproduce?

#include "lvgl.h"
#include <stdio.h>

lv_indev_t * keyboard;

int main(int argc, char *argv[])
{
        lv_init();       
        lv_sdl_window_create(1024,650);      
        lv_obj_t * main = lv_obj_create(lv_screen_active());

        keyboard = lv_sdl_keyboard_create();
        

        while (1)   
        {
                lv_task_handler();
                printf("last key: %d\n",lv_indev_get_key(keyboard));
        }
        
        lv_deinit();
        return 0;
}

kisvegabor, 2024-02-24:
I've just checked it out, and when processing the keypad it return very early when there is no focused widgets to forward the keys to. See [here](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L691-L695).

However I see no issue with moving [these lines](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L699-L704) before the early returns. 

Could you test if with these modification it works well on your end too?

Ka1mes, 2024-02-26:
Works like a charm on my end as well. Thank you for the quick fix

closed 2024-02-28 by Ka1mes

kisvegabor, 2024-03-05:
I've opened https://github.com/lvgl/lvgl/pull/5795 to fix it.

PR #5795 (2024-03-12): fix(indev): always save the last key
fixes #5717

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](ht [... cut]